rauc/rauc #734: Official Dockerfile

hongkongkiwi, 2021-05-28:
It would be nice to host an official Dockerfile in the repo (this can be linked to dockerhub for automatic builds).... here's what I'm using for reference:
```
FROM ubuntu:18.04
RUN export DEBIAN_FRONTEND=noninteractive && \
    export PKG_CONFIG_PATH=/usr/lib/x86_64-linux-gnu/pkgconfig && \
    echo "Installing packages (this may take a while)..." && \
    apt-get update -qq && \
    apt-get install -qq -y --no-install-recommends \
       build-essential automake libtool libdbus-1-dev libglib2.0-dev \
       libssl-dev libjson-glib-dev libjson-glib-dev \
       libcurl4-openssl-dev libssl-dev git ca-certificates && \
    echo "Cloning RAUC ..." && \
    git clone "https://github.com/rauc/rauc" "/build/rauc" && \
    echo "Configuring RAUC ..." && \
    cd "/build/rauc" && \
    ./autogen.sh && \
    ./configure && \
    echo "Building RAUC ..." && \
    make -j`nproc` && \
    echo "Installing RAUC ..." && \
    make install && \
    rauc --version
 CMD "/usr/bin/rauc"
 ```

ejoerns, 2021-05-28:
What we have is a testing container that we use in our CI:  https://hub.docker.com/r/rauc/testing
But that's only the build environment, not RAUC itself.

How would you use that in your setup? Or more precise, why is a Debian install or a Linux-Buildsystem (like Yocto/OE) integration insufficient for your use case?

Is there a specific reason you need to use ubuntu 18.04 instead of a more recent one?

hongkongkiwi, 2021-05-28:
The first reason, is if I want to build a package on my computer, I'm running OSX, so I need a docker in order to create bundles. I wish that you can put on the roadmap for a standalone tool I can compile for OSX just for creating bundles (but I understand this is not really high priority).

The second reason is that we use buildroot docker containers for building and it saves time if we don't have to build all the host dependencies as we can assume that they are installed in the container already, why recompile rauc if it's already compiled and installed into the container?

The third is that we have other non-buildroot images that we bundle using RAUC as well and these have nothing to do with buildroot, the build systems for these other systems are in docker containers and run as part of CI processes.

hongkongkiwi, 2021-05-28:
For completeness I'll share my other Dockerfile, this can be used to build rauc on one image and then copy the RAUC file to another container which is more lightweight:
```
FROM ubuntu:18.04 AS rauc-builder
RUN export DEBIAN_FRONTEND=noninteractive && \
    export PKG_CONFIG_PATH=/usr/lib/x86_64-linux-gnu/pkgconfig && \
    echo "Installing packages (this may take a while)..." && \
    apt-get update -qq && \
    apt-get install -qq -y --no-install-recommends \
       build-essential automake libtool libdbus-1-dev libglib2.0-dev \
       libssl-dev libjson-glib-dev libjson-glib-dev \
       libcurl4-openssl-dev git ca-certificates && \
    echo "Cloning RAUC ..." && \
    git clone "https://github.com/rauc/rauc" "/build/rauc" && \
    echo "Configuring RAUC ..." && \
    cd "/build/rauc" && \
    ./autogen.sh && \
    ./configure && \
    echo "Building RAUC ..." && \
    make -j`nproc` && \
    echo "Built RAUC version:" && \
    ./rauc --version

FROM ubuntu:18.04 AS final
COPY --from=rauc-builder /build/rauc/rauc /usr/bin/rauc
RUN export DEBIAN_FRONTEND=noninteractive && \
    echo "Installing packages (this may take a while)..." && \
    apt-get update -qq && \
    apt-get install -qq -y --no-install-recommends \
      libdbus-1-3 libglib2.0 libssl1.1 \
      libjson-glib-1.0-0 libjson-glib-1.0-common \
      libcurl4 && \
    chmod +x "/usr/bin/rauc" && \
    echo "Installed RAUC version:" && \
    rauc --version
 ```

ejoerns, 2021-05-28:
> The first reason, is if I want to build a package on my computer, I'm running OSX, so I need a docker in order to create bundles. I wish that you can put on the roadmap for a standalone tool I can compile for OSX just for creating bundles (but I understand this is not really high priority).
> 

It is always a good rule to use the same OS your are actually developing for imho ;)
Saves a lot of pain.

> The second reason is that we use buildroot docker containers for building and it saves time if we don't have to build all the host dependencies as we can assume that they are installed in the container already, why recompile rauc if it's already compiled and installed into the container?

Relying on host tools being provided by the host instead of building them explicitly always adds many moving parts to the build and can make reproducibility quite challenging/impossible. Thus, yes, if you do not build them then it is a good idea to have them in a defined state by knowing which container revision was used for the build..

I'd however limit host tools provided by the host to very standard ones. Saving build time is maybe something one can solve with other approaches.

> The third is that we have other non-buildroot images that we bundle using RAUC as well and these have nothing to do with buildroot, the build systems for these other systems are in docker containers and run as part of CI processes.

My understanding of a BSP would be to provide a comprehensive and redproducible state of the entire system. Having too many loosely coupled artifacts can make it hard to figure out what is actually running in the field. May I ask what kind of components you add to you bundle? FPGA bitstreams? external module firmware?

> For completeness I'll share my other Dockerfile, this can be used to build rauc on one image and then copy the RAUC file to another container that is more lightweight:

Do you really need to build rauc? When using ubuntu-20.04 as container base, you could simply do an apt-get install rauc there.
Then you can drop all your -dev packages.

hongkongkiwi, 2021-05-28:
> It is always a good rule to use the same OS your are actually developing for imho ;)
> Saves a lot of pain.

Well, it's only creating a bundle, so it should be fairly OS agnostic I would hope. For example, it's okay to zip or tar up a file on OSX and distribute to any platform as zip/tar is agnostic and available on all platforms. I think the problem at the moment is that the RAUC bundle mechanism is bundled into the main RAUC update logic so it would a lot of work (and a bit pointless) to port the whole thing to OSX.... it sure would be handy to have a bundle creation and verification tool though ;)

For reference, in our case, some of the firmware is actually built on OSX (via GCC) , so it would be nice to have a RAUC bundler tool to run as part of that build process. 

> Relying on host tools being provided by the host instead of building them explicitly always adds many moving parts to the build and can make reproducibility quite challenging/impossible. Thus, yes, [...]

Well, I think in the case of Docker, the host system is identical every time and exactly as you need. I think it's something that is really mature now and wasn't around when buildroot was first created. Every time I fire up my docker instance, it's going to be identical and in a way it is more reproducible than a random host buildroot is installed on to. For example, even though buildroot tries it's best, it still relies on some stuff being on the host system to build things. That's why buildroot is really only supported on ubuntu.

From my perspective I would rather have a version of RAUC that I specifically choose and control rather than whatever is bundled with buildroot, plus buildroot is a bit of a beast at times with many moving parts, so to have the tools specifically installed means it's probably less portable, but it doesn't need to be since the docker container is a fixed known reproducible quantity.

Really though, this is just personal preference :-)

> I'd however limit host tools provided by the host to very standard ones. Saving build time is maybe something one can solve with other approaches.

I think this is a good approach for non-containerised systems, but really it's just a choice. I always prefer containerisation as you can even build on Windows (if you want), your not constrained as much and you don't need to rely on all the other requirements for buildroot to run.

I've shot myself in the foot here really because I'm asking for an OSX version, but can easily use the containerised version. I do like some tools to quickly use without having to fire up a docker, for example, I don't fire up a docker to use grep, since it works well on OSX natively. So having' that option would be great.

> My understanding of a BSP would be to provide a comprehensive and redproducible state of the entire system. Having too many loosely coupled artifacts can make it hard to figure out what is actually [...]

There are about 3-4 linux partitions with mixed usages (e.g. squashfs, ext4 etc), and several different flash types (emmc/spi nor) along with several processors, something like STMicro and Atmel with custom bootloaders. So the idea being one RAUC package could contain all update components with custom scripts to flash them potentially, or each part could also be packaged.

> Do you really need to build rauc? When using ubuntu-20.04 as container base, you could simply do an apt-get install rauc there.
> Then you can drop all your -dev packages.

Yea, your right, I'd prefer not to build RAUC actually, but I want full control over which version it is, so I will add a build argument for docker which passes the version. So then we can clone that specific git tag.

hongkongkiwi, 2021-05-31:
Just to answer a previous quesation, I was thinking it would be good to host a tool specifically for bundling. The reason not to use the ci image is because there's a lot of extra stuff that is unnecessary installed in the docker if you just want to create & verify bundles.

ejoerns, 2021-06-03:
> Well, it's only creating a bundle, so it should be fairly OS agnostic I would hope.

Ok, I was assuming that you have the entire development process in OS X.

> I think the problem at the moment is that the RAUC bundle mechanism is bundled into the main RAUC update logic so it would a lot of work (and a bit pointless) to port the whole thing to OSX.... it sure would be handy to have a bundle creation and verification tool though ;)

Yes, it's in the same code base to not duplicate functionality required both for bundling and for signing, yes. It could be more decoupled than it is currently, but probably not as much as you would need it. ;)

> Well, I think in the case of Docker, the host system is identical every time and exactly as you need. I think it's something that is really mature now and wasn't around when buildroot was first [...]

Your probably right for buildroot (which I personally do not have so much experience with), other ones like Yocto did a lot of work in building a bunch of host tools, including a croos toolchain and even wrap this with a pre-built host C library environment, called 'uninative'. So this is far more decoupled. But it still can make sense to use a docker build environment additionally, yes.

> From my perspective I would rather have a version of RAUC that I specifically choose and control rather than whatever is bundled with buildroot, plus buildroot is a bit of a beast at times with many [...]

Buildroot by default bundles all releases of RAUC, but will only have a specific version for each release, yes. Requiring a specific version of RAUC sounds a bit contrary to what you said about seeing it like 'zip' or 'tar' where you probably don't care about the specific version?

But if you do not want to use the build system to generate the bundle, then using buildroot on OS X only for providing a rauc host tool is of course unnecessary overhead.

> There are about 3-4 linux partitions with mixed usages (e.g. squashfs, ext4 etc), and several different flash types (emmc/spi nor) along with several processors, something like STMicro and Atmel [...]

This sounds like a rather complex system. If you do not use a BSP, I bet you have another mechanism for control which exact combination of artifacts is installed on the target?

What we normally do when using a BSP is either building all this additional tools with the BSP tooling, or, at least create recipes for the distinct bitstream versions, etc. that download the (pre-built) artifacts in a well-known and hash-controlled state and put this all together in a bundle finally. Then a single git revision (the one of the BSP) describes the entire system. This is the standard way to do it from BSP perspective, but of course not the only way.

> Yea, your right, I'd prefer not to build RAUC actually, but I want full control over which version it is, so I will add a build argument for docker which passes the version. So then we can clone that specific git tag.

So what you actually would want here is a docker tag per rauc version, isn't it?

> The problem with ubuntu is that you are at the mercy of whatever is in the package repos. If built standalone ubuntu packages were offered here, I would probably use those instead of building.

I would like to avoid having to maintain a second source for packages (like PPA) if we can go the standard way.

The mercy part here is of course that the package has to pass the OS maintainers build tests, and OS release stages, etc. This is a much slower process than building the latest and greatest with docker of course. The original source of the ubuntu packages btw. is debian testing. The most recent version you will get from debian [unstable](https://packages.debian.org/sid/admin/rauc) repo, but of course, built against the libs from unstable.

> Just to answer a previous quesation, I was thinking it would be good to host a tool specifically for bundling. The reason not to use the ci image is because there's a lot of extra stuff that is unnecessary installed in the docker if you just want to create & verify bundles.

Yes, this was also more like 'this is the only thing we have currently'. I'd agree that you don't need all these dependencies for just building.

The reason why we do not provide a host tool docker container so far is that it is not in our scope (yet) currently, as we focus more on BSP usage and development on Linux platforms.
Also, building a bundle requires access to files, keys, pks#11-urls, etc. that are outside the container and need to be added as volumes etc. first. I am unsure how handy this is for only providing a 'tool'.
From my point of view Docker is fine for providing defined environments and for stand-alone components or components that need to interact with other containers, but less for simple tools that just need to be 'packaged' and that you normally want to use on the host's shell. But feel free to convince me otherwise. ;)

jluebbe, 2022-08-19:
@hongkongkiwi How do you use this container in practice? How do you give the container access to the necessary directories and keys?

mikehaller, 2023-07-27:
Maybe I share my opinion here as well: I am currently looking at getting Eclipse Leda (which uses RAUC) to run on Red Hat In-Vehicle Operating System 8RHIVOS aka CentOS AutoSD). Our Leda vehicle edge stack is basically a couple of containers to add automotive functionality, such as accessing Vehicle Signals, providing an abstraction / well-defined API to it etc. As update mechanism, we trigger RAUC via an MQTT message remotely.

RHIVOS has podman preinstalled, so for development and testing purposes it would be great to just install our containers (including rauc) and put them on the official RHIVOS distro images.

For people who develop on top of this stack, it's an easy setup.
For a production setup, we would be building RAUC into the base images, but for now, it's not necessary as we can start using the provided AutoSD nightlies and just install everything we need as containers on top.

I know it's more like a convenient feature, just saying it would be handy :-D

Since RAUC is not providing an official Docker container image, i'm now either going to publish my own, or i'm just leaving RAUC out for now.