rauc/rauc #735: Persistent data storage not configured properly

MitchGaines, 2021-05-28:
This is touched upon a little bit in the documentation, but I haven't found any definitive answers in the docs for how to have persistent data storage between OTA updates.

My `system.conf` is as follows:
```
[system]
compatible=RaspberryPi4
bootloader=uboot

[keyring]
path=/etc/rauc/ca.cert.pem
 
[slot.rootfs.0]
device=/dev/mmcblk0p2
type=ext4
bootname=A

[slot.rootfs.1]
device=/dev/mmcblk0p3
type=ext4
bootname=B

[slot.appfs.0]
parent=rootfs.0
device=/dev/mmcblk0p4
type=ext4

[slot.appfs.1]
parent=rootfs.1
device=/dev/mmcblk0p4
type=ext4
```

My wks file to define my partitioning scheme is as follows:
```
part /boot --source bootimg-partition --ondisk mmcblk0 --fstype=vfat --label boot --active --align 4096 --size 100
part / --source rootfs --ondisk mmcblk0 --fstype=ext4 --label rootfs_A --align 4096 
part / --source rootfs --ondisk mmcblk0 --fstype=ext4 --label rootfs_B --align 4096
part /home --source rootfs --rootfs-dir=${IMAGE_ROOTFS}/home --ondisk mmcblk0 --fstype=ext4 --label homefs --align 1024 --size 5000 --fsoptions "x-systemd.growfs"
```

I wish to make the partition mounted at `/home` be the partition containing the persistent data. However, whenever I generate a new rauc bundle using `bitbake update-bundle` and subsequently `rauc install some-bundle.raucb`, it wipes my partition.

I am using the meta-rauc and meta-rauc-community/meta-rauc-raspberrypi yocto layers and have confirmed that the above configurations are being created properly. Can anyone please tell me from a configuration standpoint what I am doing incorrectly?

Thanks for the help!

ejoerns, 2021-06-02:
@MitchGaines The first question to ask would be if you actually need redundant data partitions in your use case. If that is not the case, I would simply have a single one and mount it for both the A and the B systems (and not list it in the system.conf at all)

If you need to have them redundant, then they should not be touched by RAUC unless you have an image for it in the RAUC bundle. Is that the case? (If in doubt, `rauc info` will tell)
What you would normally need is some kind of migration from one to the other data partition during the update. This will be a rsync in the simplest case but could also be more complex if your data format needs to be migrated.
A simple data copying from within RAUC you could do for example by using an [install hook](https://rauc.readthedocs.io/en/latest/using.html#install-hooks) that does the rsysnc or cp between the currently active partition and the target partition (`RAUC_SLOT_DEVICE`).

MitchGaines, 2021-06-03:
Ahh, this makes sense. I guess my approach to this problem was totally wrong. Thanks for explaining that @ejoerns. Appreciate your time :smile:

closed 2021-06-03 by MitchGaines