rauc/rauc #736: AWS KMS for storing verity signing key

hongkongkiwi, 2021-05-28:
I would like to use AWS KMS to store the verity signing key for RAUC.

The reason for this is because with AWS KMS, you don't ever get access to the private key,, so this way it's more secure than having a private key on your filesystem and there's a central place to keep it.

As a future option, I think it would be really great to have plugin options for the signing part, I could imagine this plus similar offerings from Azure and Google developed as part of the bundle flow.

However, in the meantime, I've found an openssl plugin which seems to be able to. do the job. To use this plugin, it requires we specify the openssl engine type and some custom options.
[https://github.com/nakedible/openssl-engine-kms](https://github.com/nakedible/openssl-engine-kms)

Can this currently be done with RAUC? If not, is it something that can be exposed relatively easily? As I know openssl is already used for signing.

For the above mentioned plugin, we could offer a certificate file with the public key and the actual signing operation is used like this:
```
openssl pkeyutl -engine kms -sign -keyform engine \
    -inkey arn:aws:kms:eu-west-1:111122223333:key/deadbeef-dead-dead-dead-deaddeafbeef \
    -in digest.bin -out sig.bin -pkeyopt digest:sha256
```

hongkongkiwi, 2021-05-28:
I've just found this, do you think this would work directly via the pkcs11 support?
[https://github.com/JackOfMostTrades/aws-kms-pkcs11](https://github.com/JackOfMostTrades/aws-kms-pkcs11)

jluebbe, 2021-05-28:
I've not tried this PKCS#11 implementation, but it looks like the simplest approach. For key access in RAUC, we support either plain files or PKCS#11, as this is the defacto standard for accessing HSMs. Also, it's relatively easy to build adapters to other backends as PKCS#11 modules.

Please take a look at https://rauc.readthedocs.io/en/latest/advanced.html#pkcs-11-support. If your PKCS#11 module does not support requesting the certificate, you can still use a plain file for that and use a ``pkcs11:`` URL for the key.

hongkongkiwi, 2021-05-29:
Thanks for the reply, this is encouraging, I'm going to do some testing today on the PKCS#11 KMS implementation I linked, I'll post an update here if successful.

For my case, I'm looking to use AWS KMS which is almost free. The PKCS#11 implemented provided by Amazon is for their CloudHSM product which is about $400-500 USD/month (which sadly isn't very clear until you look into it deeply).  It's a good product, but a heavy initial investment. They don't offer such functionalities except via custom API for KMS product.

hongkongkiwi, 2021-05-29:
For reference, I've also found this openssl plugin:
[https://github.com/nakedible/openssl-engine-kms](https://github.com/nakedible/openssl-engine-kms)

This would require passing the engine parameter and a few other custom parameters. Is there support for passing custom openssl engines?

hongkongkiwi, 2021-05-30:
@jluebbe I have a followup question here. For the above linked PKCS11 plugin, it seems like I need to pass the plugin path to pkcs tools (e.g. osslsigncode).

For example,  in osslsigncode I would use the following:
`osslsigncode sign -h sha256 -pkcs11engine /usr/lib/x86_64-linux-gnu/engines-1.1/pkcs11.so -pkcs11module /usr/lib/x86_64-linux-gnu/pkcs11/aws_kms_pkcs11.so -certs /etc/aws-kms-pkcs11/kms-test.crt -key pkcs11: -in 'test.txt' -out 'test.signed'`

For rauc, I understand from the docs I could use `rauc --key 'pkcs11:' --cert 'kms-test.crt'` but how can I pass the pkcs11engine and pkcs11module arguments to point to my pkcs11 plugin?

hongkongkiwi, 2021-05-30:
EDIT: looks like I missed this environment variable **RAUC_PKCS11_MODULE**. I think this can be closed.

closed 2021-05-30 by hongkongkiwi

jluebbe, 2021-05-31:
The engine ist OpenSSL-specific and used automatically by RAUC if you use a 'pkcs11:' URI. The module is configured using RAUC_PKCS11_MODULE (or system-wide via p11kit).

Were you able to sign a build with this setup?

hongkongkiwi, 2021-06-03:
Yes, this works perfectly, could I suggest you list this project in the official documentation ? It's a pretty cool tie in!

Author has added a note on RAUC support in the [README](https://github.com/JackOfMostTrades/aws-kms-pkcs11).

`RAUC_PKCS11_MODULE=/usr/lib/x86_64-linux-gnu/pkcs11/aws_kms_pkcs11.so rauc bundle --cert=mycert.pem --key='pkcs11:' input_dir/ my_bundle.raucb`

jluebbe, 2021-06-03:
Thanks, I've added a hint to the docs: https://github.com/rauc/rauc/commit/74307f470934c8246fb36941e38e91cb483648e1