rauc/rauc #737: Silence failed to resolve realpath error

hongkongkiwi, 2021-05-31:
When creating a bundle I get this error:
`rauc-Message: 04:13:16.008: Failed to resolve realpath for '/dev/disk/by-uuid/a4af6bed-7716-4281-8865-b8ac4b283339'`

It seems irrelevant to my situation as I'm only using the bundle command not the other functionality. Is it possible to silence this error in this case? I currently don't have the --debug flag passed.

I'm running rauc with this command line:
`rauc --cert=/rauc-keys/dev.crt --key=/rauc-keys/dev.key --debug bundle /tmp/rauc-SwIFA8IWDd8BTqrun /images/output.raucb`

jluebbe, 2021-05-31:
During initialization, RAUC tries to find the device for the rootfs, which isn't actually needed when creating bundles. We should split the initialization, so this code path is not used on the host.

hongkongkiwi, 2021-06-08:
Looks like this has been fixed in the latest master code version
```
rauc-Message: 15:36:00.837: Debug log domains: 'rauc'
(rauc:240): rauc-DEBUG: 15:36:00.838: bundle start
(rauc:240): rauc-DEBUG: 15:36:00.838: system config not found, using default values
(rauc:240): rauc-DEBUG: 15:36:00.838: input directory: /tmp/rauc-VM31QxA0Q5Rb6ZYpQ
(rauc:240): rauc-DEBUG: 15:36:00.838: output bundle: /test.raucb
(rauc:240): rauc-DEBUG: 15:36:01.119: Payload size: 2584576 bytes.
Creating bundle in 'verity' format
(rauc:240): rauc-DEBUG: 15:36:01.119: Hash creation SHA256, data blocks 631.
(rauc:240): rauc-DEBUG: 15:36:01.119: Using 2 hash levels.
(rauc:240): rauc-DEBUG: 15:36:01.119: Data size: 2584576 bytes.
(rauc:240): rauc-DEBUG: 15:36:01.119: Hashed size: 2609152 bytes.
rauc-Message: 15:36:03.652: No keyring given, skipping signature verification
(rauc:240): rauc-DEBUG: 15:36:03.652: Signature offset: 2609152 bytes.
(rauc:240): rauc-DEBUG: 15:36:03.652: Bundle size: 2610826 bytes.
```

PR #1059 (2023-02-17): Skip target-related context setup if no config is required
Do not invoke any target-related context setup steps if no config is required.
This is safe since config mode is 'required' for all target-related subcommands (like 'service') that need to have specific context available (such as bootname, etc.).

With this we gain two major benefits:

* Not invoking pointless procfs, sysfs and mount inspections when running as a host tool

* Removing the misleading 'info'-Messages like:
  > rauc-Message: 13:53:56.765: Failed to resolve realpath for '/dev/mapper/Root-root'

  e.g. when running rauc 'ho [... cut]