rauc/rauc #739: Failed to create bundle: cannot sign inconsistent manifest: Missing size for verity bundle

hongkongkiwi, 2021-06-08:
Hi, when attempting to sign a bundle (using latest master branch build) I'm getting this error:
`Failed to create bundle: cannot sign inconsistent manifest: Missing size for verity bundle`

I'm using the following to create the bundle:
`rauc --cert=/keys/dev-rauc-kms.crt   --key=pkcs11:token=mytoken   --debug bundle /tmp/rauc  /tmp/test.raucb`

For this test, I'm just using a file that's 127 bytes (it's not a real image, just a text file my /etc/hosts file).

My bundle dir looks like this:
```
ls -lh /tmp/rauc
total 8.0K
-rw-r--r-- 1 root root 172 Jun  8 15:29 manifest.raucm
-rw-r--r-- 1 root root 176 Jun  8 15:22 rootfs.img
```

My manifest looks like this:
```
[update]
compatible=test123

[bundle]
format=verity

[image.rootfs]
sha256=f0ecbb34e69b6fda41be5c7cd7d37c3a6bbeb3bb3295a7b8fd6d7645d184d9d3
size=176
filename=rootfs.img
```

My original bundle did not contain the size or sha256 lines, they were added automatically by RAUC.

Debug mode does not seem to yield anything more informative:
```
rauc-Message: 15:29:21.325: Debug log domains: 'rauc'
(rauc:206): rauc-DEBUG: 15:29:21.326: bundle start
(rauc:206): rauc-DEBUG: 15:29:21.326: system config not found, using default values
(rauc:206): rauc-DEBUG: 15:29:21.326: input directory: /tmp/rauc
(rauc:206): rauc-DEBUG: 15:29:21.326: output bundle: /test.raucb
(rauc:206): rauc-DEBUG: 15:29:21.352: Payload size: 4096 bytes.
Creating bundle in 'verity' format
(rauc:206): rauc-DEBUG: 15:29:21.352: Hash creation SHA256, data blocks 1.
(rauc:206): rauc-DEBUG: 15:29:21.352: Using 0 hash levels.
(rauc:206): rauc-DEBUG: 15:29:21.352: Data size: 4096 bytes.
(rauc:206): rauc-DEBUG: 15:29:21.352: Hashed size: 4096 bytes.
Failed to create bundle: cannot sign inconsistent manifest: Missing size for verity bundle
```

Any idea what the problem could be?

hongkongkiwi, 2021-06-08:
Ah, I found there is an issue signing small files. It might be good to do a size check on the bundle to give a more informative error message when encountering this issue.

What is the minimum size for files in the bundle?

jluebbe, 2021-06-08:
Without looking at the code, you should avoid this corner case when your compressed payload is larger than 8k. Please use something more realistic for testing. :)

ejoerns, 2021-06-08:
Yep, this is caused by this code: https://github.com/rauc/rauc/blob/master/src/bundle.c#L540

Maybe it is worth adding a sanity check there.

ejoerns, 2021-06-08:
Ok, to be precise, the sanity check is already there, but should probably be

``` diff
-g_assert(combined_size*4096 >= (off_t)offset);
+g_assert(combined_size*4096 > (off_t)offset);
```

as otherwise `combined_size*4096` can be equal to `offset` and subtracting the latter from the first will result in a `verity_size` of 0.

hongkongkiwi, 2021-06-08:
Thanks for the info, I will ensure my image is 8k or larger!

hongkongkiwi, 2021-06-09:
Oh that's a good idea, thanks @jluebbe. I will close this now that I know the issue and limit. Thanks!

PR #740 (2021-06-25): src/bundle: fix error handling for small squashfs sizes
When having a squashfs that is only 4096 bytes in size, we cannot
generate a hash table.

To catch these issues, an assert was places in the code.
However, the assert must check for '>' instead of '>=' as otherwise it
would allow 'combined_size*4096' equal to 'offset' which results in a
'verity_size' of 0.

As the size of 'offset' actually depends on user input (i.e. bundle
content size), we should not use g_assert() as primary check instance
here, but add a GError check that allows to emit a user-readable error.

Fixes #739
closed 2021-06-09 by hongkongkiwi