rauc/rauc #741: casync-http invoked oom-killer

oberbaum, 2021-06-14:
```
rauc-Message: 13:32:34.354: Debug log domains: 'rauc'

(rauc:349): rauc-DEBUG: 13:32:34.378: install started
(rauc:349): rauc-DEBUG: 13:32:34.389: input bundle: http://192.168.2.28:8000/casync-bundle.raucb
(rauc:349): rauc-DEBUG: 13:32:34.423: Trying to contact rauc service
installing
  0% Installing
  0% Determining slot states
 20% Determining slot states done.
 20% Checking bundle
 20% Verifying signature
 40% Verifying signature done.
 40% Checking bundle done.
 40% Loading manifest file
 60% Loading manifest file done.
 60% Determining target install group
 80% Determining target install group done.
 80% Updating slots
 80% Checking slot rootfs.1
 90% Checking slot rootfs.1 done.
 90% Copying image to rootfs.1
EXT4-fs (mmcblk0p3): mounted filesystem with ordered data mode. Opts: (null)
cfg80211: failed to load regulatory.db
imx-sdma 20ec000.sdma: external firmware not found, using ROM firmware
coda 2040000.vpu: Using fallback firmware vpu/vpu_fw_imx6d.bin
coda 2040000.vpu: Firmware code revision: 570363
coda 2040000.vpu: Initialized CODA960.
coda 2040000.vpu: Firmware version: 3.1.1
coda 2040000.vpu: encoder registered as video0
coda 2040000.vpu: decoder registered as video1
casync-http invoked oom-killer: gfp_mask=0x500cc2(GFP_HIGHUSER|__GFP_ACCOUNT), order=0, oom_score_adj=0
CPU: 1 PID: 369 Comm: casync-http Not tainted 5.4.40 #1
Hardware name: Freescale i.MX6 Quad/DualLite (Device Tree)
[<801101a8>] (unwind_backtrace) from [<8010b674>] (show_stack+0x10/0x14)
[<8010b674>] (show_stack) from [<80a080dc>] (dump_stack+0x84/0x98)
[<80a080dc>] (dump_stack) from [<801ca2ec>] (dump_header+0x54/0x22c)
[<801ca2ec>] (dump_header) from [<801c9920>] (oom_kill_process+0x34c/0x35c)
[<801c9920>] (oom_kill_process) from [<801c9f18>] (out_of_memory+0xe8/0x3f4)
[<801c9f18>] (out_of_memory) from [<80203198>] (__alloc_pages_nodemask+0xa8c/0xda4)
[<80203198>] (__alloc_pages_nodemask) from [<802237a8>] (pipe_write+0x1e4/0x490)
[<802237a8>] (pipe_write) from [<80218768>] (__vfs_write+0x144/0x1c8)
[<80218768>] (__vfs_write) from [<8021b10c>] (vfs_write+0xb0/0x17c)
[<8021b10c>] (vfs_write) from [<8021b370>] (ksys_write+0x64/0xec)
[<8021b370>] (ksys_write) from [<80101000>] (ret_fast_syscall+0x0/0x54)
Exception stack(0x870a7fa8 to 0x870a7ff0)
7fa0:                   00c2e1e0 00000000 00000001 767e9008 00006457 00000000
7fc0: 00c2e1e0 00000000 76fde4d0 00000004 00c2e328 00434bd4 7ea22be8 00000000
7fe0: 00000004 7ea22ad0 76c9ffef 76c2bd16
Mem-Info:
active_anon:23210 inactive_anon:6293 isolated_anon:0
 active_file:427 inactive_file:1456 isolated_file:8
 unevictable:2 dirty:0 writeback:7 unstable:0
 slab_reclaimable:3224 slab_unreclaimable:7852
 mapped:1210 shmem:23097 pagetables:290 bounce:0
 free:79913 free_pcp:2 free_cma:74424
Node 0 active_anon:92840kB inactive_anon:25172kB active_file:1708kB inactive_file:6364kB unevictable:8kB isolated(anon):0kB isolated(file):32kB mapped:5236kB dirty:0kB writeback:28kB shmem:92388kB writeback_tmp:0kB unstable:0kB all_unreclaimable? no
Normal free:318700kB min:1672kB low:2088kB high:2504kB active_anon:92868kB inactive_anon:25172kB active_file:1748kB inactive_file:6676kB unevictable:8kB writepending:40kB present:524288kB managed:503748kB mlocked:0kB kernel_stack:928kB pagetables:1160kB bounce:0kB free_pcp:44kB local_pcp:0kB free_cma:297696kB
lowmem_reserve[]: 0 0 0
Normal: 614*4kB (UC) 923*8kB (UMEC) 515*16kB (MEC) 129*32kB (UMEC) 53*64kB (MEC) 15*128kB (EC) 0*256kB 0*512kB 0*1024kB 6*2048kB (EC) 2*4096kB (C) 1*8192kB (E) 0*16384kB 8*32768kB (C) = 318336kB
25334 total pagecache pages
0 pages in swap cache
Swap cache stats: add 0, delete 0, find 0/0
Free swap  = 0kB
Total swap = 0kB
131072 pages RAM
0 pages HighMem/MovableOnly
5135 pages reserved
81920 pages cma reserved
Tasks state (memory values in pages):
[... 270 lines cut ...]
[    254]     0   254      563       14    12288        0             0 syslogd
[    255]   996   255      840      199    12288        0          -900 dbus-daemon
[    256]   994   256      917       51    10240        0             0 avahi-daemon
[    257]     0   257      888       19    12288        0             0 agetty
[    260]     0   260     1489      116    14336        0             0 ofonod
[    263]     0   263     1803       96    12288        0             0 sh
[    275]     0   275     1304      343    14336        0             0 systemd-logind
[    278]   998   278     1613       88    12288        0             0 ntpd
[    282]     0   282     9942      250    28672        0             0 rauc
[    349]     0   349     7637      177    26624        0             0 rauc
[    368]     0   368     2790      699    20480        0             0 casync
[    369]     0   369     2754      456    18432        0             0 casync-http
[    377]     0   377      660      138    10240        0             0 xinit
[    380]     0   380     2058      369    14336        0             0 Xorg
oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),task=casync,pid=368,uid=0
Out of memory: Killed process 368 (casync) total-vm:11160kB, anon-rss:2132kB, file-rss:664kB, shmem-rss:0kB, UID:0 pgtables:20kB oom_score_adj:0
oom_reaper: reaped process 368 (casync), now anon-rss:0kB, file-rss:0kB, shmem-rss:0kB
100% Copying image to rootfs.1 failed.
100% Updating slots failed.
100% Installing failed.
LastError: Installation error: Failed updating slot rootfs.1: failed to run casync extract: Child process killed by signal 9
Installing `http://192.168.2.28:8000/casync-bundle.raucb` failed

```
The board itself has 512mb ram and we use yocto as distribution. Is there a way to reduce the amount of memory used?

jluebbe, 2021-06-14:
Could you check where you have placed casync download cache? I suspect that you have a tmpfs which is using a large portion of your RAM. casync prefetches chunks, so this cache can grow quite large.

If you're using .caidx (the .tar equivalent), you could use https://github.com/florolf/casync-nano, which doesn't use a cache.

oberbaum, 2021-06-14:
Yes, I use tmpfs, but i dont know how to check where rauc/casync caches the download. Is there a config for casync?

jluebbe, 2021-06-15:
It might be from the environment ($TMPDIR) or `/var/tmp`. Check with `df` while casync is running. The OOM output doesn't list any process which uses a large amount of memory, so I suspect that it's used by a tmpfs.

`.cacnk` is the extension of the chunk files. The index has either `.caidx` for archive mode or `.chibx` for image mode. You can check with `rauc info`.

oberbaum, 2021-06-15:
```
Filesystem     1K-blocks   Used Available Use% Mounted on
/dev/root        1965904 984148    860516  54% /
devtmpfs           87520      0     87520   0% /dev
tmpfs             251872      0    251872   0% /dev/shm
tmpfs             251872   8924    242948   4% /run
tmpfs             251872      0    251872   0% /sys/fs/cgroup
tmpfs             251872    364    251508   1% /tmp
tmpfs             251872 121976    129896  49% /var/volatile
/dev/mmcblk0p1     26554   8004     18550  31% /boot
/dev/mmcblk0p4   1046512      4   1046508   1% /opt
```
It seems that the cache is somewhere in /var/volatile. Does the parameter `--cache` i mentioned above help me in this case? It would be nice if i could use the stock casync instead of a fork.

I use rauc 1.2.

jluebbe, 2021-06-15:
Yes, this looks like casync using /var/volatile, perhaps there is a symlink from /var/tmp? To set a different tmp directory, you can set TMPDIR in the environment of your rauc server (which will get passed to casync) or use the `tmppath` option in the `[casync]` section of the system.conf  (see: https://rauc.readthedocs.io/en/latest/reference.html#system-configuration-file).

casync-nano is not a fork, but a reimplementation of the subset needed for image installation on a target with rauc. Development of casync itself has stalled a bit: https://github.com/systemd/casync/commits/master

oberbaum, 2021-06-15:
I tried to set the tmppath, but there is no change in the behavior. Are you sure this works with rauc 1.2?

jluebbe, 2021-06-15:
At least the commit 446214528c8d6bb502bb81cddc9b2325cc31efb1 is included in v1.2. Nevertheless, please update to the current release. We don't support old versions.

oberbaum, 2021-06-15:
Ok, i updated to version 1.5.1 and edit the system.conf with:

```
[casync]
tmppath=/home/root
```
but now i have permission errors:
```
rauc[280]: Failed to run synchronizer: Operation not permitted
```
I found  #598 and patched the bundle class manually and with the yocto built from the casync bundle i am able to update :)

Thanks for the great support <3

jluebbe, 2021-07-22:
This seems solved. If you have limited temporary space, the HTTP streaming draft PR https://github.com/rauc/rauc/pull/755 might be interesting for you (and we'd be happy about test feedback).

closed 2021-07-22 by jluebbe