rauc/rauc #743: Rauc Install - Failed mounting bundle: failed to mount bundle: No such device

MitchGaines, 2021-06-16:
Hi there,

I'm taking a stab at extending meta-rauc-community support to the OrangePi. Anyway, I am experiencing the issue outlined below that I _think_ I understand, but am not really sure how to proceed on and was wondering if someone here could point me towards a potential resolution. I'm sure its a problem in my system config, but I haven't been able to nail down exactly what that config is. Any help is appreciated

I am running on Rauc v1.5.1 on Yocto dunfell. Here is my rauc status: 

```
root@orange-pi-zero:~# rauc status
=== System Info ===
Compatible:  OrangePiZero
Variant:     
Booted from: rootfs.0 (A)

=== Bootloader ===
Activated: rootfs.0 (A)

=== Slot States ===
o [rootfs.1] (/dev/mmcblk0p3, ext4, inactive)
        bootname: B
        boot status: good

x [rootfs.0] (/dev/mmcblk0p2, ext4, booted)
        bootname: A
        mounted: /
        boot status: good
```

During `rauc install`, I am getting the following error: 

```
root@orange-pi-zero:~# rauc install -d update-bundle-orange-pi-zero.raucb 
rauc-Message: 10:47:40.694: Debug log domains: 'rauc'
(rauc:306): rauc-DEBUG: 10:47:40.710: install started
(rauc:306): rauc-DEBUG: 10:47:40.710: input bundle: /home/root/update-bundle-orange-pi-zero.raucb
(rauc:306): rauc-DEBUG: 10:47:40.727: Trying to contact rauc service
installing
  0% Installing
  0% Determining slot states
 20% Determining slot states done.
 20% Checking bundle
 20% Verifying signature
 40% Verifying signature done.
 40% Checking bundle done.
100% Installing failed.
LastError: Failed mounting bundle: failed to mount bundle: No such device
Installing `/home/root/update-bundle-orange-pi-zero.raucb` failed
```

Here is my journal output:

```
Sep 20 10:44:04 orange-pi-zero systemd[1]: Starting Rauc Update Service...
Sep 20 10:44:04 orange-pi-zero rauc[168]: Config option 'statusfile=<path>/per-slot' unset, falling back to per-slot status
Sep 20 10:44:04 orange-pi-zero rauc[168]: Using per-slot statusfile
Sep 20 10:44:04 orange-pi-zero systemd[1]: Started Rauc Update Service.
Sep 20 10:44:04 orange-pi-zero rauc[168]: mounting slot /dev/mmcblk0p3
Sep 20 10:44:04 orange-pi-zero rauc[168]: Failed to load status file /run/rauc/rootfs.1/slot.raucs: No such file or directory
Sep 20 10:44:04 orange-pi-zero rauc[168]: Failed to load status file /slot.raucs: No such file or directory
Sep 20 10:44:04 orange-pi-zero rauc[168]: rauc mark: marked slot rootfs.0 as good
Sep 20 10:47:40 orange-pi-zero rauc[168]: input bundle: /home/root/update-bundle-orange-pi-zero.raucb
Sep 20 10:47:40 orange-pi-zero rauc[168]: Active slot bootname: A
Sep 20 10:47:40 orange-pi-zero rauc[168]: installing /home/root/update-bundle-orange-pi-zero.raucb: started
Sep 20 10:47:40 orange-pi-zero rauc[168]: installing /home/root/update-bundle-orange-pi-zero.raucb: Checking and mounting bundle...
Sep 20 10:47:40 orange-pi-zero rauc[168]: Reading bundle: /home/root/update-bundle-orange-pi-zero.raucb
Sep 20 10:47:40 orange-pi-zero rauc[168]: Verifying bundle signature... 
Sep 20 10:47:45 orange-pi-zero rauc[168]: Verified detached signature by 'O = Test Org, CN = Test Org Development-1'
Sep 20 10:47:45 orange-pi-zero rauc[168]: Mounting bundle '/home/root/update-bundle-orange-pi-zero.raucb' to '/run/rauc/bundle'
Sep 20 10:47:45 orange-pi-zero rauc[168]: Configured loop device '/dev/loop0' for 158969856 bytes
Sep 20 10:47:45 orange-pi-zero rauc[168]: Failed mounting bundle: failed to mount bundle: No such device
Sep 20 10:47:45 orange-pi-zero rauc[168]: installing /home/root/update-bundle-orange-pi-zero.raucb: Failed mounting bundle: failed to mount bundle: No such device
Sep 20 10:47:45 orange-pi-zero rauc[168]: installing /home/root/update-bundle-orange-pi-zero.raucb: finished
Sep 20 10:47:45 orange-pi-zero rauc[168]: installing `/home/root/update-bundle-orange-pi-zero.raucb` failed: 1
```
Below is an excerpt from the `strace` that may or may not shed some light into this. Let me know if you'd like to see more:

```
write(1, "  0% Installing\n", 16  0% Installing
)       = 16
write(1, "  0% Determining slot states\n", 29  0% Determining slot states
) = 29
write(1, " 20% Determining slot states don"..., 35 20% Determining slot states done.
) = 35
write(1, " 20% Checking bundle\n", 21 20% Checking bundle
)  = 21
poll([{fd=3, events=POLLIN}], 1, 0)     = 1 ([{fd=3, revents=POLLIN}])
read(3, "\1\0\0\0\0\0\0\0", 16)         = 8
write(1, " 20% Verifying signature\n", 25 20% Verifying signature
) = 25
poll([{fd=3, events=POLLIN}], 1, -1)    = 1 ([{fd=3, revents=POLLIN}])
read(3, "\1\0\0\0\0\0\0\0", 16)         = 8
write(1, " 40% Verifying signature done.\n", 31 40% Verifying signature done.
) = 31
poll([{fd=3, events=POLLIN}], 1, -1)    = 1 ([{fd=3, revents=POLLIN}])
read(3, "\1\0\0\0\0\0\0\0", 16)         = 8
write(1, " 40% Checking bundle done.\n", 27 40% Checking bundle done.
) = 27
poll([{fd=3, events=POLLIN}], 1, -1)    = 1 ([{fd=3, revents=POLLIN}])
read(3, "\2\0\0\0\0\0\0\0", 16)         = 8
futex(0x51af68, FUTEX_WAKE_PRIVATE, 1)  = 1
write(1, "100% Installing failed.\n", 24100% Installing failed.
) = 24
futex(0xb6db0494, FUTEX_WAKE_PRIVATE, 1) = 0
ioctl(1, TCGETS, {B115200 opost isig icanon echo ...}) = 0
write(1, "\33[JLastError: Failed mounting bu"..., 77LastError: Failed mounting bundle: failed to mount bundle: No such device
) = 77
write(3, "\1\0\0\0\0\0\0\0", 8)         = 8
futex(0x51af74, FUTEX_WAKE_PRIVATE, 2147483647) = 0
write(2, "Installing `/home/root/update-bu"..., 66Installing `/home/root/update-bundle-orange-pi-zero.raucb` failed
) = 66
write(6, "\1\0\0\0\0\0\0\0", 8)         = 8
futex(0x52c248, FUTEX_WAKE_PRIVATE, 1)  = 1
futex(0x52c0ac, FUTEX_WAKE_PRIVATE, 1)  = 1
futex(0x52803c, FUTEX_WAKE_PRIVATE, 1)  = 1
futex(0xb5000010, FUTEX_WAIT_PRIVATE, 2, NULL) = -1 EAGAIN (Resource temporarily unavailable)
futex(0xb5000010, FUTEX_WAKE_PRIVATE, 1) = 0
write(6, "\1\0\0\0\0\0\0\0", 8)         = 8
futex(0x52803c, FUTEX_WAKE_PRIVATE, 1)  = 1
futex(0x52803c, FUTEX_WAKE_PRIVATE, 1)  = 1
exit_group(1)                           = ?
+++ exited with 1 +++
```

Anyway, I feel like what is failing is the mounting of the bundle to the directory `/run/rauc/bundle`. So, I guess, any idea what might cause this to occur? 

Thanks :smile:

jluebbe, 2021-06-16:
Is your log complete? There should be a mount() system call.

Do you have squashfs support in your kernel (`grep squashfs /proc/filesystems`)?

MitchGaines, 2021-06-16:
Yep - those are complete logs. Considering `grep squashfs /proc/filesystems` didn't return anything, I'm going to assume that's the issue here. My kernel _should_ support it... I'm running Linux Mainline v5.4.111. Sounds like I'm going to need to do some digging to figure out what I need to do to get it enabled. Thanks!

ejoerns, 2021-06-16:
@MitchGaines maybe the documentation prevents you from having to dig to deep? https://rauc.readthedocs.io/en/latest/integration.html#kernel-configuration

MitchGaines, 2021-06-16:
Thanks for the quick help on this @jluebbe and @ejoerns. So, I was doing it the way you specified in the docs you posted, but of course I had it appending to the linux-sunxi kernel bbappend as opposed to the linux-mainline bbappend so it was not configured on my system. Once I got that fixed, OTA updates worked like a charm.

You all are the best. Thanks again for the help. :raised_hands:

closed 2021-06-16 by MitchGaines