rauc/rauc #745: Custom: rauc bundle custom installation.

linuxanonimous84, 2021-06-24:
Hello,
I'm working with rauc 1.0 and I'm creating a bundle that it contains a barebox image and a rootfs.image.
The manifest.raucm calls to the hook (pre-)install (I've tried booth options) but reading the documentation I don't get it how i Can customize the installation in order to let install for the rootfs image but not the barebox, for example. I can print some variables like RAUC_SLOT_NAME or CLASS i can read information from the linux enviroment, but i donät see how can i disable this specific slot installation depending on some conditions.

Someone can guide me?
hook.sh
```
#!/bin/sh
case "$1" in
    slot-pre-install)
        echo "pre device is $RAUC_SLOT_DEVICE"
        echo "SLOT NAME $RAUC_SLOT_NAME"
         disable here installation of bootloader.
        ;;

    slot-install)
        echo "install device is $RAUC_SLOT_DEVICE"
        ;;
    slot-post-install)
        echo "pre device is $RAUC_SLOT_DEVICE"
        ;;
    *)
        echo "Hook arg is $1"
        exit 1
        ;;
esac
```
manifest:
```
[update]
compatible=********
version=1.0

[hooks]
filename=hook.sh

[image.rootfs]
filename=systemimage.ext4

[image.bootloader]
filename=barebox.img
hooks=pre-install; or install;
```

Thanks for your help

jluebbe, 2021-06-24:
Why do you need to use version 1.0? It's 2.5 years old by now.

There is no feature to skip installation of slots from hooks (yet). If you use the install hook, you could perfrom the image writing your self (or skip).

linuxanonimous84, 2021-06-24:
Thanks for your response,

I don't decide the version to use it so I cannot change it in short term but I'll push it for it.
So if i understand well you then, the only option that i have is to call hook=install on an specific slot and perform by myself the installation?
But then i have to disable in some way the automatic installation, isn't?

ejoerns, 2021-06-24:
The normal way to skip installation is to not mention the slot in the manifest.

What would be the reason for conditionally not updating some slots? I assume that you need to make your decision on the target, right?

linuxanonimous84, 2021-06-24:
Yes.
Because we want to keep always in the last version the bootloader. 
Depending on the context we have to install old bundles, so that can lead to incompatibilities because we can assure backward compatibility but not forward.
Ex.
```
Bootloader --> v2.0 
Rescue partition --> v2.0
Active partition v2.0
```
if we install bundel v1.0
then we have 
```
Bootloader --> v1.0 
Rescue partition --> v2.0
Active partition v1.0
```
So it can lead to not be able to boot from rescue partition.

jluebbe, 2021-06-24:
> So if i understand well you then, the only option that i have is to call hook=install on an specific slot and perform by myself the installation?

Yes.

> But then i have to disable in some way the automatic installation, isn't?

If you set hook=install, the default handler is not run.

linuxanonimous84, 2021-06-24:
> If you set hook=install, the default handler is not run.

That it means doing the manual installation for all the slots? or only from the one where it is called the install hok?

ejoerns, 2021-06-24:
> > If you set hook=install, the default handler is not run.
> 
> That it means doing the manual installation for all the slots? or only from the one where it is called the install hok?

No, the install hook affects only the installation of the slot that it is set for. All slots that have no hook set will be installed by the default mechanism.

linuxanonimous84, 2021-07-07:
Thanks, It was quite useful your help and I have been able to create a custom image for my rauc bundle.
I'll proceed to close the topic.

Have a great day! :+1:

closed 2021-07-07 by linuxanonimous84