rauc/rauc #746: Failed to load dm table: Invalid argument

om26er, 2021-06-29:
after enabling bundle type verity, I cannot install our update bundles
```
root@raspberrypi4-simple-things:~# rauc install rpi4-bundle-raspberrypi4-simple-things-20210629001821.raucb 
installing
  0% Installing
  0% Determining slot states
 20% Determining slot states done.
 20% Checking bundle
 20% Verifying signature
 40% Verifying signature done.
 40% Checking bundle done.
 40% Checking manifest contents
 60% Checking manifest contents done.
100% Installing failed.
LastError: Failed mounting bundle: Failed to load dm table: Invalid argument
Installing `/root/rpi4-bundle-raspberrypi4-simple-things-20210629001821.raucb` failed
```

jluebbe, 2021-06-29:
Please check that you have enable the required features: https://rauc.readthedocs.io/en/latest/integration.html#bundle-format-migration

jluebbe, 2021-07-22:
Is this still a problem for you? If so, please check https://github.com/rauc/rauc/issues/746#issuecomment-870313547 and reopen if needed.

closed 2021-07-22 by jluebbe

hongkongkiwi, 2021-11-18:
Same problem here, I was missing verity support in the kernel. It wasn't super clear from the error message, but after I searched the issues and found this I noticed I had forgot to include it. 

So for others who come, it's absolutely related to kernel dependencies.

ejoerns, 2021-11-18:
I wonder if we can easily add any check that tells the user to enable verity support on his target?

Extending the above-mentioned error message seems inappropriate to me as this can be caused by other reasons, too.

A safe way would be to do what `dmsetup targets` does. I can see all tools using `libdevmapper` for this, but it seems to boil down to a `DM_LIST_VERSIONS` ioct. Not sure if that's the only way and sufficient benefit for implementing it?

jluebbe, 2021-11-18:
`dmsetup targets` won't show targets from modules that have not been loaded yet. This same applies to the hash algorithms. So a check would need to try to configure a target or needs explicitly try to load the relevant modules and then look at `dmsetup targets`.

We have documented it in https://github.com/rauc/rauc/#target-prerequisites and https://rauc.readthedocs.io/en/latest/integration.html#kernel-configuration. If people don't read these pages, other things are likely misconfigured for their use-case as well (and even harder to auto-detect). I'm not sure we really help here.