rauc/rauc #748: Optional Desync Support

om26er, 2021-06-29:
Desync is fast and is actively maintained unlike casync. I have been playing with it for the last week and it seems that it's "concurrent" parameter on `desync extract` command is very fruitful especially when downloading chunks from S3.

Casync it seems downloads one chunk at a time and that results in very slow bundle install times.

It also doesn't do client side delta calculations, which means that the `caibx` of the seed block device would have to be provided separately, this in itself saves quite a bit of time depending on the size of the rootfs.

Desync can't be used as a drop in replace because
1. It doesn't support this parameter https://github.com/rauc/rauc/blob/a67c565cf7cac60b73ad4539c03d631aa41fb7b8/src/update_handler.c#L258
2. Doesn't directly take a block device with `--seed` parameter and have to do something like https://github.com/folbricht/desync/issues/178#issuecomment-869004096

What would be the "correct" approach to use desync with RAUC ?

om26er, 2021-06-30:
I think one "clean" approach would be to extend the `rauc install` command line to take an optional "casync-args" argument where we could define 
```
rauc install --casync-args="--seed=/home/om26er/imageA.caibx" https://something.com/casync-bundle.raucb
```

The could could then check if a "seed" is provided explicitly it should only use that.

om26er, 2021-06-30:
a `--casync-args=` option can also be very useful to provide desync parameters for concurrent downloads (`-n`)

jluebbe, 2021-06-30:
As the user should not need to know which image (A/B) is active or being installed to, specifying the location of the previously installed `.caibx` on the command line is not very useful.
RAUC has the necessary information during the initial installation, so it would be simplest to store the `.caibx` next to the central slot status on a shared storage. Then it could be used automatically for the next installation.

For configuring generial casync options, a `system.conf` option seems appropriate.

ejoerns, 2021-06-30:
I'd agree on what @jluebbe said as this matches my initial thoughts on this topic.

Maybe this would be the point where we should introduce some sort of generic 'rauc data dir' (instead of having only the very specific slot status file).

There are more things about to come in the future probably that need a distinct non-redundant storage location. Thus having a folder for this where we can put in all the management data for RAUC would be quite useful imho.

jluebbe, 2022-08-19:
Since https://github.com/rauc/rauc/pull/925, we now have a shared data directory where the `.caibx` for each slot could be stored (see https://github.com/rauc/rauc/pull/817#issuecomment-1158692257).