rauc/rauc #763: check_bundle_access() failing with "unable to find mounted device for bundle"

ian-abbott, 2021-08-17:
I am upgrading to rauc 1.5.1 installed by PTXdist 2021.08.1 (previously using rauc 1.0 installed by PTXdist 2019.06.0).

Running the command `rauc info /tmp/update.rauc` produces the following output:
```
rauc-Message: 12:28:57.908: Config option 'statusfile=<path>/per-slot' unset, falling back to per-slot status
rauc-Message: 12:28:57.923: Using per-slot statusfile
rauc-Message: 12:28:57.925: Reading bundle: /tmp/update.raucb
rauc-Message: 12:28:57.930: Verifying bundle signature... 
unable to find mounted device for bundle
```

The "/tmp" directory containing the "update.raucb" file is mounted as:
```
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,relatime,mode=1777)
```

The `stat -f /tmp/update.raucb` command indicates that the filesystem type is "ramfs":
```
  File: "/tmp/update.raucb"
    ID: 0        Namelen: 255     Type: ramfs
Block size: 4096      
Blocks: Total: 0          Free: 0          Available: 0
Inodes: Total: 0          Free: 0
```

After some debugging, I get to the part of `check_bundle_access()` that is checking the list of mount points returned by `g_unix_mounts_get(NULL)` to find the underlying device. However, the `dev_path` values returned by the calls to `g_unix_mount_get_device_path(m)` do not seem to be something that can be passed to `stat()`. All the calls to `stat(dev_path, &dev_stat)` are failing. The `dev_path` values for the list of mount points on my system are: "ubi0:root2", "devtmpfs", "none", "ubi0:data", "ubi0:data", "ubi0:root2", "overlay", "sysfs", "tmpfs", "tmpfs", "tmpfs", "tmpfs", "tmpfs", "devpts". I guess the `dev_path` for the "/tmp" mount is one of those "tmpfs" strings. It is not surprising that `stat(dev_path, &dev_stat)` fails on these `dev_path` strings!

I'm guessing the code should be calling `g_unix_mount_get_mount_path(m)` instead of `g_unix_mount_get_device_path(m)`? There could be more than one mount path for the device containing the "update.raucb" file and they might not all be owned by root, so perhaps the `break;` statement after setting `mount_checked = TRUE;` should be removed so that other matching mount paths can be checked too.

ian-abbott, 2021-08-17:
On second thoughts, replacing `g_unix_mount_get_device_path(m)` with `g_unix_mount_get_mount_path(m)` (and comparing the `st_dev` values to find the match) would only be useful for checking the ownership of the mount point, rather than the ownership of the underlying block device.

Perhaps the correct solution is to set `mount_checked = TRUE;` for filesystems of type `RAMFS_MAGIC`?

jluebbe, 2021-08-17:
`bundle_statfs.f_type` should be RAMFS_MAGIC in this case. We should probably handle RAMFS_MAGIC the same as TMPFS_MAGIC, as there is no underlying device.
Could you check if moving the RAMFS_MAGIC case down to where `mount_checked` is set to `TRUE`, fixes this for you?

Also, do you know you you see 'ramfs' for your /tmp mount?

ian-abbott, 2021-08-17:
@jluebbe Yes, moving the `RAMFS_MAGIC` case handling fixes it for me. I created a pull request #764.

I have determined that I see 'ramfs' for the /tmp mount because my kernel has `CONFIG_SHMEM` and `CONFIG_TMPFS` disabled, so the kernel replaces 'tmpfs' with 'ramfs'. The description for `CONFIG_SHMEM` includes "Disabling this option replaces shmem and tmpfs with the much simpler ramfs code, which may be appropriate on small systems without swap."

jluebbe, 2021-08-17:
Thanks! This was fixed by 5ebb3d7fb9fe4e75f2ff7adfd8dc86e849cad9c2 in PR #764.

closed 2021-08-17 by jluebbe