rauc/rauc #768: Is RAUC manages Interrupted downloads

vis-sin, 2021-09-01:
Hi,
Need info if RAUC manages to resume download if the last download was interrupted due to network unavailability 
Checking code very briefly I realized it uses curl but nowhere I was able to find curl is used to start from the last offset.
I am not sure of this. If somebody can clarify this will be a great help.

Thanks
Eagerly waiting for your Responses

jluebbe, 2021-09-01:
I've moved this question to the rauc repository, as it's not related to meta-rauc.

As the integrated download mode is mostly used for casync-based bundles (which are very small), it doesn't handle resuming downloads. With streaming (https://github.com/rauc/rauc/pull/755) the individual block requests are retried.

jluebbe, 2022-04-22:
We don't have plans to add resuming to the old download code for casync bundles. HTTP streaming will be part of the upcoming v1.7 release. For longer interruptions, future incremental modes may be helpful (https://github.com/rauc/rauc/pull/882).

Please reopen with more details if those alternatives are not appropriate for your use-case.

closed 2022-04-22 by jluebbe