rauc/rauc #773: casync seed mount of rootfs needs to detect mount options

j-licht, 2021-09-16:
Hi,

I'm currently playing around with casync support in rauc and two redundant rootfs slots.
Rauc mounts the booted slot to be used as seed for casync.
Usually my rootfs is mounted ro and rauc tries to mount it rw.
This fails with "device already mounted".
I think the problem here is, that this remount works only if the ReadWrite Option is the same.
A solution could be to detect the current mount options of the rootfs.

ejoerns, 2022-04-11:
Note that this mount behavior was addressed with https://github.com/util-linux/util-linux/commit/11b916cdabd9f57a4cddf10fd743af2165930dcd that now instead gives something like:

    mount: /run/rauc/rootfs.0: WARNING: source write-protected, mounted read-only

However, choosing the right mount options when invoking 'mount' should still be the proper solution.

jluebbe, 2023-03-02:
The possible fix is outlined in https://github.com/rauc/rauc/pull/774#issuecomment-1263316459.