rauc/rauc #776: utils: Possible memory leaks when using key_file_consume_string()

LeSpocky, 2021-09-28:
The function `key_file_consume_string()` from *src/utils.c* calls `g_key_file_get_string()` which says in its documentation:

> The caller of the method takes ownership of the data, and is responsible for freeing it.

As far as I understand `key_file_consume_string()` should free that memory then, or if it just returns that `gchar*` the caller should free it.

However I see usages of `key_file_consume_string()` in *src/config_file.c* where that's not the case, especially if the function is just used as parameter of another function like `r_resolve_path()`.

I tried improving the documentation for `key_file_consume_string()` with #775, but every use of that function should be checked for possible memory leaks.

ejoerns, 2021-09-29:
> The function `key_file_consume_string()` from _src/utils.c_ calls `g_key_file_get_string()` which says in its documentation:
>
>
> As far as I understand `key_file_consume_string()` should free that [...]

Yes.

> However I see usages of `key_file_consume_string()` in _src/config_file.c_ where that's not the case, especially if the function is just used as parameter of another function like `r_resolve_path()`.
>

As far as I see, these are the only problematic cases. Other allocated strings in config_file.c should be freed by `free_config()`.

> I tried improving the documentation for `key_file_consume_string()` with #775, but every use of that function should be checked for possible memory leaks.

Most string manipulating or parsing functions return a newly allocated string, thus this is nothing exclusive to this function.

ejoerns, 2022-06-23:
This has been fixed in #880 (commit https://github.com/rauc/rauc/commit/297215f322a47ea280a0c6baf44563b22d4092af)

closed 2022-06-23 by ejoerns