rauc/rauc #806: New tagged release?

hongkongkiwi, 2021-11-10:
Is it possible to release a new tagged version? The last v1.5.1 was 11 months ago and there's a number of improvements since then. I'm hoping to use tagged releases for my buildroot package.

Right now I'm building from master (or a git commit) and using rauc --version doesn't show up properly.
```
# rauc --version
rauc UNKNOWN
```

There are also some extra config options such as boot-attempts that rauc 1.5.1 complains about but was added in a later release.

ejoerns, 2021-11-10:
> Is it possible to release a new tagged version? The last v1.5.1 was 11 months ago and there's a number of improvements since then. I'm hoping to use tagged releases for my buildroot package.

I'd really appreciate a new release too, but so far the plan was that the next tagged release (v1.6) contains support for streaming and not only 'minor' bug fixes and enhancements.
Streaming support however was delayed at various points due to time resource constraints. I actually hope we will manage to merge it quite soon. If not, we should think about doing a new release before. But we also still have some other PRs that should be part of the [1.6 milestone](https://github.com/rauc/rauc/milestone/11).

> Right now I'm building from master (or a git commit) and using rauc --version doesn't show up properly.
> 
> ```
> # rauc --version
> rauc UNKNOWN
> ```

The version information can only be available if you use a release tarball or build from an actual git repository. If you use a normal archive without git information (like a github zip download), there is no version information contained of course.

> There are also some extra config options such as boot-attempts that rauc 1.5.1 complains about but was added in a later release.
> 
> As an aside, if there are no plans for a new tagged release, is there a file I can update which will make rauc --version show something useful?

Of course, there are plans for a new tagged release, for constraints, see above.

Creating an archive with version information from a git checkout can easily be done by running:

    make dist

Manually injecting a version information is possible by creating the file `.tarball-version` like

    echo "1.5.1.155" > .tarball-version

and then calling

     ./autogen.sh
     ./configure
     make

hongkongkiwi, 2021-11-17:
> The version information can only be available if you use a release tarball or build from an actual git repository

To clarify on this, I am building from the actual git repository in a dockerfile and the version information still doesn't appear.

`make dist` does not appear to be valid either.

ejoerns, 2021-11-17:
> > The version information can only be available if you use a release tarball or build from an actual git repository
> To clarify on this, I am building from the actual git repository in a dockerfile and the version information still doesn't appear.

Interesting. Could you provide what

    git describe

tells in this repo? And what a manual invocation of

    ./build-aux/git-version-gen .tarball-version

emits?

> However the trick with `echo "1.5.1.155" > .tarball-version` works fine! +1

Sounds good.

> `make dist` does not appear to be a valid command (?).

It should be. What does it print that you assume it is an invalid command?

`dist` should be a valid make target.

Try

    grep -Rn "^dist.*:" Makefile

to see it.

ejoerns, 2022-01-27:
Good news :tada: : RAUC [v1.6](https://github.com/rauc/rauc/milestone/11) is in the pipeline and very close to being released.

We would appreciate if some of you have the time to do a quick test with latest master in your existing projects.

jluebbe, 2022-02-09:
We've just released v1.6, closing this issue.

closed 2022-02-09 by jluebbe