rauc/rauc #807: Multiple variants in DTB seems not handled properly

hongkongkiwi, 2021-11-10:
Hi, in my DTS/DTB I have the following:
`compatible = "company1,company1-soc","company2,ourproduct";`

This works fine in linux and shows as a NULL character between in linux, e.g:
```
# cat /proc/device-tree/compatible
company1,company1-soc<NULL>company2,ourproduct
```
So, I might use this in bash to display with a space
`cat /proc/device-tree/compatible | xargs -0`

However, when using `rauc status`, it ignores everything after the first null.
```
# rauc status
Variant:     company1,company1-soc
......
```

So, I would expect to be able to see both, perhaps replacing null with a special char or something else? Lets say _ then in my bundle variant I can use "[slot.myslot.company1,company1-soc_company2,ourproduct".

As a workaround, I can use variant-file for this, but I prefer if I could use the dtb directly. 

I am not sure how common this use case is, but if it's planned not to fix, I think it would be good to add this into the docs somewhere to say the expected behaviour if the dtb is setup like this.

jluebbe, 2021-11-10:
As the most specific compatible comes first, you can list multiple variants in the manifest (and refer to the same file from several variants). Documenting this better would be useful, though.

hongkongkiwi, 2021-11-16:
I see, that makes sense, thanks.

closed 2021-11-16 by hongkongkiwi