rauc/rauc #808: Refinement of the current progress handling

Urist-McGit, 2021-11-11:
As discussed in the Matrix channel the last couple of days, some key problems of the current way progress is handled in RAUC were identified:

- the distribution of the different progress step could be improved. The most prominent example is the `Copying image to rootfs` step, that resides in the last 10% and takes up most of the installation time.
- there is no per slot installation progress
- there is no way to get more fine grained information on long running steps, like a slow download or a long installation. The user might be interested in more frequent progress updates in these cases

To fix these issues the current steps need to be taken:

1. redistribute the steps to match the real effort better
2. add per slot installation progress
3. implement a new API to get more detailed progress information

Related issues: #670 #727

Urist-McGit, 2021-11-11:
@ejoerns @jluebbe - I took the freedom and summarized the discussion on the Matrix channel here. Feel free to add more information or correct any mistakes I made ;)

ejoerns, 2022-08-19:
> 2. add per slot installation progress

This is partly addressed by https://github.com/rauc/rauc/pull/811

ejoerns, 2023-03-09:
> * redistribute the steps to match the real effort better

This is now addressed by #1091

ejoerns, 2023-03-21:
Except for `3.`, I guess all points were addressed now and are merged into `master`.

I am not sure if there is currently a need for more than what we have now.
Any opinions?

jluebbe, 2023-06-23:
Closing this as done. If there is still something missing, please reopen.

closed 2023-06-23 by jluebbe