rauc/rauc #809: RAUC unable to run on Ingenic T31

swidget1, 2021-11-11:
I am trying to use RAUC on an Ingenic T31N processor. I managed to generate the certificates, create the rauc executable, create the bundle and copy all of it to the NOR flash of the T31. There are 2 problems:

1. I cannot install RACU on to the T31. There is no documentation that describes this procedure. It shows how to install RACU on the host system, but not the target system. How do I install RAUC on the target system using an SD card? No internet connection.

2. When I try to run RAUC I get the gibberish. How do I run RAUC on the T31?

[root@Swidget-uc1_1:rauc]# ./rauc service -d
./rauc: line 1:
▒▒e▒mdAp▒▒▒▒d▒▒▒[lV▒1▒▒▒▒^▒▒▒▒▒▒ʡ▒9▒1%▒: not found
./rauc: line 1: can't create ▒▒@x▒
@: Invalid argument
./rauc: line 1:ELF: not found
./rauc: line 1: %@@@▒ȭȭ▒▒▒▒▒▒▒PPPHHPtP▒P%,8}8▒8▒p888: not found
./rauc: line 4: ▒▒▒w&▒#'
▒▒      JT#[▒$$: not found
./rauc: line 4: ▒%▒g▒C▒▒▒%d$▒▒: not found
./rauc: line 4: ▒
                 ▒: not found
./rauc: line 4: : not found
./rauc: line 4: ▒}▒@K: not found
*: not found 4: %▒D
./rauc: line 5: B▒: not found
./rauc: line 6: ▒: not found
./rauc: line 5: ▒
▒1: not found
./rauc: line 10:
                 du-: not found
./rauc: line 5: ▒▒: not found
./rauc: line 10: 3PJ▒#▒
                       B!▒#▒▒: not found
./rauc: line 11: ▒7s▒: not found
▒: not found 12: ▒B
./rauc: line 1: syntax error: unexpected "&"
[1]+  Done(2)                    ./rauc service
[root@Swidget-uc1_1:rauc]# ./rauc: line 11: ▒▒▒▒
                                                ▒▒▒: not found
▒%▒%▒▒▒#line 10: /"▒G`""k▒
▒▒zE▒O▒ ▒▒#▒#▒▒o▒: not found
PuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuT
TY
-sh: PuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTY: not found_

ejoerns, 2021-11-11:
> * I cannot install RACU on to the T31. There is no documentation that describes this procedure. It shows how to install RACU on the host system, but not the target system. How do I install RAUC on the target system using an SD card? By the way, there is currently no internet connection. I will add this later, for now I just want to do an upgrade from the SD card.

What do you mean by installing? How do you get other software into the target's rootfs? I would propose using the same mechanism if possible.
Did you cross-compile *RAUC* correctly for your target?
For embedded devices, it is highly recommended to use an embedded Linux build system like Yocto, Buildroot, PTXdist, ELBE, etc.
There is no reason to need an internet connection for RAUC. It actually depends where your bundles come from.

> * When I try to run RAUC I get gibberish as shown below. How do I run RAUC on the T31?

Looks like you broke the binary or attempt to execute something not meant to be executed or similar.

ejoerns, 2021-11-11:
As I cannot see any RAUC-related issue here, yet, I'll convert this to a 'Discussion'.

closed 2021-11-11 by ejoerns