rauc/rauc #819: CWE-377: Insecure tempfile creation for remote downloads

jktjkt, 2022-01-04:
I'm on rauc 1.4, but the same code is in current `master`. It appears that all remote files always end up in `$TMP/_download.raucb`, which is a problem because it tends to be a path that's often shared and writable by multiple users. A good fix would be to use unpredictable path name below the temp directory, e.g. via `mkstemp`. A quick-and-dirty workaround would be configuring the systemd unit file so that it has a private `/tmp`.

jktjkt, 2022-01-04:
I have not checked the code whether it's feasible for an attacker to swap the file for something malicious (perhaps to bypass signature verification?). What I had in mind was more like an attacker leaving a symlink in place before the update process starts, and then RAUC (running as root) accidentally overwriting some important system file.

jluebbe, 2022-01-05:
> I have not checked the code whether it's feasible for an attacker to swap the file for something malicious (perhaps to bypass signature verification?).

I meant that bypassing signature verification should be mitigated in 1.5.

> What I had in mind was more like an attacker leaving a symlink in place before the update process starts, and then RAUC (running as root) accidentally overwriting some important system file.

Ah, yes. That looks like it would be possible. I'll make a PR.

closed 2022-01-26 by ejoerns