rauc/rauc #830: RAUC and dm-crypt

fabianbuettner, 2022-01-27:
Hi,

I am currently playing around with a system which is running on a plain dm-crypt volume mapped ext4 rootfs.

boot partitions are currently configured as slot type vfat and root partitions are configured as slot type raw.

Is there any best practice on how to handle this use case? I tried to use a dm-crypt volume as rootfs slot but then of course the squashFS compression is not able to compress anything anymore ;)

regards, 
Fabian

jluebbe, 2022-01-27:
The usual approach is to use `/dev/mapper/<devicename>` as the device for the slot, so RAUC only sees the unencrypted device and content.

ejoerns, 2022-01-27:
Anyone volunteer who would like to add an entry for dm-crypt in the FAQ similar to this one? :) https://rauc.readthedocs.io/en/latest/faq.html#can-i-use-rauc-with-a-dm-verity-protected-partition

fabianbuettner, 2022-01-28:
@jluebbe Thanks, I will try that
@ejoerns I can try to write a short text. Where should I submit it?

Is this the right place to ask questions? I still have some more questions ;)

ejoerns, 2022-01-28:
> @jluebbe Thanks, I will try that 
> @ejoerns I can try to write a short text. Where should I submit it?

@fabianbuettner GitHub PRs are our favorite ;)

> Is this the right place to ask questions? I still have some more questions ;)

Depends a bit on the type of question, but in doubt: yes. Alternatives are [#rauc](https://app.element.io/#/room/#rauc:matrix.org) irc/matrix chat, or the 'Discussions' [Q&A](https://github.com/rauc/rauc/discussions/categories/q-a) section here.

fabianbuettner, 2022-01-28:
@ejoerns so another question that I had was:
Let's assume I have a A/B setup with one redundant data partition(which I want to manage via slot hook handler). If I read the documentation correctly, in that case I can not use a global status file, right?

ejoerns, 2022-01-28:
@fabianbuettner the global slot status file needs to be stored at some non-redundant location which remains persistent over updates. Thus, if your normal data partition is redundant, maybe you define another non-redundant partition that you use for RAUC information only?
Potentially, this could also be on another storage (like NOR), it just has to be a mountable file system so that you can make it visible for RAUC in your rootfs.

fabianbuettner, 2022-01-31:
@ejoerns yes, I understand that the global slot status file needs to be stored at some non-redundant location which remains persistent over updates but according to documentation a global status file is not mandatory: "When installing an image, RAUC can write the images hash together with some status information to a central **or** per-slot status file (refer statusfile option)."

ejoerns, 2022-01-31:
The global status file is not mandatory, but will probably become more important in the future for handling state between installations. Thus if possible, prepare a storage location for it. But yes, you will be able to run RAUC without or with a per-slot status file currently.

fabianbuettner, 2022-01-31:
Thank you, I will consider this in my system design.

I still have one last question about redundant data partitions.
As far as I understand it, it probably does not make sense to set the "parent" property for the data partition because in that case on each update also the data partition would be required to be updated.
Therefore(without the "parent" property), it could be possible to update the (currently inactive) data partition independently from the currently active rootfs slot. Is this how it is supposed to work? or should the data partition be switched in sync with the boot/rootfs partitions?

I understand that there is probably no one solution which fits it all, but I am a bit lost about where to do the data migration because I could not find an example for this use case.

fabianbuettner, 2022-01-31:
> The global status file is not mandatory, but will probably become more important in the future for handling state between installations. Thus if possible, prepare a storage location for it. But yes, you will be able to run RAUC without or with a per-slot status file currently.

Was the per-slot status file something which was added

fabianbuettner, 2022-02-01:
The changelog for v1.6 mentions "Allowed omitting the image filename when using the install slot hook."
I assume using that version I could configure the redundant data partitions with the "parent" property, because it would allow me to omitt the filename property when generating bundles and could just migrate the data from old slot to new slot

closed 2022-02-01 by ejoerns

fabianbuettner, 2022-02-03:
I am currently on v1.2 by the way.

Looks like it works a little bit different than I had understood from reading the documentation :)
I was able to associate my redundant data slots with their respective boot slots and in the bundle manifest, I can just add the tar.xz files for the boot and rootfs slots. No need to specify the data slot at all!
I also enabled "install-same=true" for the data slot.

Using the "install" hook for the rootfs and data slots, I was able to cover all my use cases.

RAUC works really well so far!

ejoerns, 2022-02-03:
> The changelog for v1.6 mentions "Allowed omitting the image filename when using the install slot hook." I assume using that version I could configure the redundant data partitions with the "parent" property, because it would allow me to omitt the filename property when generating bundles and could just migrate the data from old slot to new slot in the install hook.

It was exactly written for this use case, yes. :smirk: 

> Another thing I observed was: When using `/dev/mapper/<devicename>` as device for the rootfs slots, I am unable to put the `cryptsetup open` command in the pre-install or install hook because RAUC returns an error if the device does not exist.

Yes, cryptsetup should be handled by the system during startup and not by RAUC.

> I am currently on v1.2 by the way.

Why? ;) If you under development, I would highly recommend starting with recent software.
Software ages enough in the field. And you also should directly start with the 'verity' bundle format and make your system ready for this.

(refer Part C here https://github.com/rauc/rauc/security/advisories/GHSA-cgf3-h62j-w9vv)

> Looks like it works a little bit different than I had understood from reading the documentation :) I was able to associate my redundant data slots with their respective boot slots and in the bundle [...]

That sounds like the right way to do it, yes.
Data copying/migration however is always quite use-case specific, but if you do not need a complex reinterpretation of the data (which should then be handled by the application), a simple copy is sufficient for most cases.

On drawback of this approach is that you end up with old data upon a fallback of course.

> Using the "install" hook for the rootfs and data slots, I was able to cover all my use cases.
> 
> RAUC works really well so far!

Good to hear :)

fabianbuettner, 2022-02-03:
> Yes, cryptsetup should be handled by the system during startup and not by RAUC.

Maybe it's a bit unusual setup, but we need to be able to prepare the dm-crypt plain volumes used for rootfs and data partitions during the update. The key used for the dm-crypt volumes will be stored as encrypted blob on the boot slot. During an update I need to decrypt that key using the SoC's internal fuses and then open the dm-crypt volumes using that key. That way, the key used for the encrypted partitions could be changed by us. That's the reason why I can not put  `/dev/mapper/rootfs.1` in the `/etc/rauc/system.conf` file because if I call `rauc install` that device is not yet present. It's a little bit of a chicken-egg problem.

> Why? ;) If you under development, I would highly recommend starting with recent software.
Software ages enough in the field. And you also should directly start with the 'verity' bundle format and make your system ready for this.

It's because we started development with Yocto version called zeus and that's why I added meta-rauc's zeus branch to our project.

The only problem left is that I would also like to be able to update the currently active data slot using a `install` hook.
In that case, rauc will update the status of the currently unused data slot. Unfortunately, I could not find a solution to that problem, yet.

ejoerns, 2022-02-03:
> Maybe it's a bit unusual setup, but we need to be able to prepare the dm-crypt plain volumes used for rootfs and data partitions during the update. The key used for the dm-crypt volumes will be [...]

Maybe this could work with the pre-install *handler*?
https://rauc.readthedocs.io/en/latest/using.html#system-based-customization-handlers

> It's because we started development with Yocto version called zeus and that's why I added meta-rauc's zeus branch to our project.

zeus is EOL since 2020. One really would not want to develop on this. I'd suggest going at least to dunfell (LTS) or deven better to latest (honister).
 
> The only problem left is that I would also like to be able to update the currently active data slot using a `install` hook. In that case, RAUC will update the status of the currently unused data slot. Unfortunately, I could not find a solution to that problem, yet.

RAUC is not designed to alter the current system. However, your have a process running as root and scripting capailities. Thus pretty sure there is a way to hack this.

fabianbuettner, 2022-02-03:
> Maybe this could work with the pre-install handler?
https://rauc.readthedocs.io/en/latest/using.html#system-based-customization-handlers

Ok, i will try that out in my next step :)

> zeus is EOL since 2020. One really would not want to develop on this. I'd suggest going at least to dunfell (LTS) or deven better to latest (honister).

I agree, but we have only very limited man power and so far we did not get time to update anything(we have some legacy Delphi dependencies which are difficult to get running). Maybe I can update only the RAUC layer to a version which has the verity bundle format?

> RAUC is not designed to alter the current system. However, your have a process running as root and scripting capailities. Thus pretty sure there is a way to hack this.

Yes, that I have understood. If it would be possible to configure a slot property so that no slot status will be written by RAUC for such a manually managed slot that would be sufficient. I had thought that `install-same=true` would help me to achieve that but I just saw that I misinterpreted the documentation.

ejoerns, 2022-02-04:
> I agree, but we have only very limited man power and so far we did not get time to update anything(we have some legacy Delphi dependencies which are difficult to get running).

Delphi? Ouh. :smirk: 

There are some good embedded Linux consultants out there to solve this issue if time is the only very limited resource :)

> > Maybe I can update only the RAUC layer to a version which has the verity bundle format?
>

dunfell has RAUC v1.5.1 and is the release after zeus. It is not actively tested against zeus but could be worth a try.

> Yes, that I have understood. If it would be possible to configure a slot property so that no slot status will be written by RAUC for such a manually managed slot that would be sufficient. I had [...]

Maybe this is actually something to think about how to handle.

fabianbuettner, 2022-02-04:
> Delphi? Ouh. smirk

yes :smile: 

> There are some good embedded Linux consultants out there to solve this issue if time is the only very limited resource :)

Of course money is very limited, too :sweat_smile: 

> > > Maybe I can update only the RAUC layer to a version which has the verity bundle format?
> 
> dunfell has RAUC v1.5.1 and is the release after zeus. It is not actively tested against zeus but could be worth a try.

Ok, thank you, I am going to try that out.

> Maybe this is actually something to think about how to handle.

:+1:

fabianbuettner, 2022-02-04:
Do you know if RAUC v1.6 will be available in dunfell? I think the "omit filename if there is an install hook feature" would be really nice to have because it makes it possible to write a cleaner install hook script :smile:

fabianbuettner, 2022-02-04:
I switched to the dunfell branch and a first update with v1.5.1 seems to work:

```
rauc status --detailed
=== System Info ===
Compatible:  foo
Variant:
Booted from: boot.1 (B)

=== Bootloader ===
Activated: boot.1 (B)

=== Slot States ===
  [bootloader.0] (/dev/mmcblk1, boot-emmc, inactive)
      slot status:
          bundle:
              compatible=(null)

o [boot.0] (/dev/mmcblk1p2, vfat, inactive)
        bootname: A
        boot status: good
      slot status:
          bundle:
              compatible=(null)
    [resources.0] (/dev/mapper/resources.0, ext4, inactive)
      slot status:
          bundle:
              compatible=(null)
    [rootfs.0] (/dev/mapper/rootfs.0, ext4, inactive)
      slot status:
          bundle:
              compatible=(null)

x [boot.1] (/dev/mmcblk1p5, vfat, booted)
        bootname: B
        boot status: good
      slot status:
          bundle:
              compatible=foo
              version=1.0
              description=Test description
              build=0.1.2-123456
          checksum:
              sha256=580c43b0b4a6e79897fd8a1712cf44347edf9d5d3087d64d025dff43690d5556
              size=7064528
          installed:
              timestamp=2022-02-04T15:41:20Z
              count=2
          activated:
              timestamp=2022-02-04T15:43:56Z
              count=2
          status=ok
    [rootfs.1] (/dev/mapper/rootfs.1, ext4, active)
        mounted: /
      slot status:
          bundle:
              compatible=foo
              version=1.0
              description=Test description
              build=0.1.2-123456
          checksum:
              sha256=22d5e5241c8f03720e453bc30bab7e01bf0fcef228898d699b19e6e944234c70
              size=113758756
          installed:
              timestamp=2022-02-04T15:43:56Z
              count=2
          status=ok
    [resources.1] (/dev/mapper/resources.1, ext4, active)
        mounted: /resources
      slot status:
          bundle:
              compatible=(null)
```

Do you know why for slot `boot.0` the slots `resources.0` and `rootfs.0` are in a different order compared to `boot.1` with `rootfs.1` and `resources.1`?

ejoerns, 2022-02-06:
> I switched to the dunfell branch and a first update with v1.5.1 seems to work:

Sounds good!

> Do you know why for slot `boot.0` the slots `resources.0` and `rootfs.0` are in a different order compared to `boot.1` with `rootfs.1` and `resources.1`?

Slot order is currently dictated by the glib collections implementations and might be just random.
Alternatives would be to order them by alphabetic slot name or order in system.conf for displaying. But that is not implemented, yet.

fabianbuettner, 2022-02-07:
I see, it's a little bit confusing because the behavior seems not to be deterministic? for slot B it looks always correct and for slot A always incorrect.

I noticed another thing I can not explain :sweat_smile: 
I am not sure if those messages were there on v1.2, but during boot I get the following log messages:

```
-- Logs begin at Mon 2022-02-07 17:42:40 UTC, end at Mon 2022-02-07 17:59:19 UTC. --
systemd[1]: Starting Rauc Update Service...
rauc[227]: rauc-Message: 17:43:06.154: Using central status file /firmware_update/central-status.raucs
systemd[1]: Started Rauc Update Service.
rauc[227]: rauc-Message: 17:43:06.821: Failed to resolve realpath for '/dev/mapper/resources.1'
rauc[227]: rauc-Message: 17:43:07.199: Failed to resolve realpath for '/dev/mapper/resources.1'
rauc[227]: rauc-Message: 17:43:07.474: rauc mark: marked slot boot.0 as good
rauc[227]: rauc-Message: 17:47:25.624: Failed to resolve realpath for '/dev/mapper/resources.1'
rauc[227]: rauc-Message: 17:58:43.674: Failed to resolve realpath for '/dev/mapper/resources.1'
```

I am currently on slot A and this is my `/proc/cmdline`:
`enablelog=yes dm-crypt=yes enterinitramfs=yes rauc.slot=A`

As you can see `rauc.slot=A` is passed via `/proc/cmdline`. And in `get_cmdline_bootname` we have
```
bootname = regex_match("rauc\\.slot=(\\S+)", contents);
        if (bootname)
                return bootname;
```
so I think the function should be left here and the "Failed to resolve realpath" messages should never be printed?

ejoerns, 2022-02-08:
@fabianbuettner This should be caused by what I described in #846, but this does not seem to be the final solution, yet. Maybe we could decrease the debug level or rework/rethink the logic there.
But the messages are not critical in any way (just polluting the log a bit).

ejoerns, 2022-02-08:
You might probably also want to check why the realpath to the mapper device cannot be resolved in your case.

PR #835 (2022-02-01): Add dm-crypt section to faq documentation page
Closes #830