rauc/rauc #833: updating slot which are not part of A?B using RAUC

Avni22, 2022-01-28:
Hi,

I am updating the "appfs slot".

After installation is done by RAUC (Using  " rauc install "  command).

appfs.slot is updated successfully but the issue is that my rootfs(from where I have run rauc install bundle) is getting corrupted after installation of rauc bundle.

Also after rauc update that RAUC service is not available to check slot status. I am attaching logs below 

On the same side, I am able to update the rootfs without any error. But facing issue while updating the appfs slot only. 

I am attaching my configuration files, 

rauc status
----------------------
rauc status: error retrieving slot status via D-Bus: error calling D-Bus method "GetSlotStatus": The name de.pengutronix.rauc was not provided by any .service files 

system.config
--------------------
[system]
compatible=IMX8QXP
bootloader=uboot
statusfile=/home/root/central_slot.raucs

[keyring]
path=ca.cert.pem

[handlers]
pre-install=/usr/lib/rauc/preinstall.sh
post-install=/usr/lib/rauc/postinstall.sh

[slot.appfs.0]
device=/dev/mmcblk0p5
type=ext4
#parent=rootfs.0

#SYSTEM SLOT - A 
[slot.rootfs.0]
device=/dev/mmcblk0p2
type=ext4
bootname=A

#SYSTEM SLOT - B 
[slot.rootfs.1]
device=/dev/mmcblk0p3
type=ext4
bootname=B

central_slot.raucs    (Before Update)
--------------------------------------------------
[slot.rootfs.0]
bundle.compatible=IMX8QXP
status=ok

[slot.rootfs.1]
bundle.compatible=IMX8QXP
status=ok

[slot.appfs.0]
bundle.compatible=IMX8QXP
status=ok

central_slot.raucs    (After Update)
--------------------------------------------------
[slot.appfs.0]
bundle.compatible=IMX8QXP
bundle.version=13.157
status=ok
sha256=e0bbc98749564f34917faebb7b74be08e9bef310f8303a6128854b70726051e4
size=216
installed.timestamp=2021-03-24T10:26:03Z
installed.count=1

appfs slot
--------------------
logs while updating the
root@imx8qxp:~# rauc install appfs.raucb
installing
0% Installing
0% Determining slot states
20% Determining slot states done.
20% Checking bundle
20% Verifying signature
40% Verifying signature done.
40% Checking bundle done.
40% Checking manifest contents
60% Checking manifest contents done.
60% Determining target install group
80% Determining target install group done.
80% Updating slots
80% Checking slot appfs.0
90% Checking slot appfs.0 done.
90% Copying image to appfs.0
[ 51.705526] EXT4-fs (mmcblk0p5): mounted filesystem with ordered data mode. Opts: (null)
100% Copying image to appfs.0 done.
100% Updating slots done.
[ 61.008531] EXT4-fs (mmcblk0p5): mounted filesystem with ordered data mode. Opts: (null)
100% Installing done.
Installing `/home/root/appfs.raucb` succeeded

ejoerns, 2022-01-28:
Looks like the status file is being overwritten.

How is `/home/root` mounted in your setup? Or could you paste the output of the `mount` command here?

Avni22, 2022-01-28:
output of : `findmnt`

I am using seperate partition to hold status file. That partition are mounted after power on to `home/root`. 

[image]

ejoerns, 2022-01-28:
> I am using seperate partition to hold status file. That partition are mounted after power on to `home/root`.

But in your output, the mount point is `/home/root/data` while the status file is located at `/home/root/`.

Avni22, 2022-01-28:
Hi

Yes 

> > I am using seperate partition to hold status file. That partition are mounted after power on to `home/root`.
> 
> But in your output, the mount point is `/home/root/data` while the status file is located at `/home/root/`.

Yes thats also right. I am using init.sh script to put that data on home/root 

root@imx8qxp:~# ls
central_slot.raucs data

ejoerns, 2022-01-28:
Not sure if I understand why.

How do you ensure that you preserve the slot status over reboots? Your log does not look as if this would work, anyway.

If you have central_slot.raucs in /home/root and /home/root is part of your A or B partition, the file will be overridden on each update. Which is not how it's meant to be used. It should be in a fix non-redundant partition instead.

To actually get an idea why rauc fails in your case, the output of the RAUC service would also be helpful.

Avni22, 2022-01-28:
> Not sure if I understand why.
>
> How do you ensure that you preserve the slot status over reboots? Your log does not look as if this would work, anyway.
>
> If you have central_slot.raucs in /home/root [...]

I am using pri-install and post-install script for taking care of status file before reboot and as it is in seperat partition.

Also while updating with rootfs i am not facing any issue.  

status file after update that i have attached above is without reboot after appfs update from rootfs. ​

status file log before reboot and after updating the appfs using "rauc install bundle.raucb"
```
[slot.appfs.0]
bundle.compatible=IMX8QXP
bundle.version=13.157
status=ok
sha256=e0bbc98749564f34917faebb7b74be08e9bef310f8303a6128854b70726051e4
size=216
installed.timestamp=2021-03-24T10:26:03Z
installed.count=1
```

I think status file should contain information of all the slot after updating any slot.

jluebbe, 2022-01-28:
> I am using pri-install and post-install script for taking care of status file before reboot and as it is in seperat partition.

That can't work as the central slot status file is only written after the installation has completed (so after any post-install script). It *must* be a simple file on a shared data partition.