rauc/rauc #862: GRUB Bootloader update for slot selection

TimoK93, 2022-02-18:
Hello guys,

first of all: Thanks for your great work! 

I successfully installed and tested RAUC in combination with the HawkBit updater following the instructions and examples in the docs. There is one open issue I have, which I was not able to understand from the docs/examples. It is about the slot selection in the GRUB bootloader for a system with redundant rootfs partitions.

### My setup
I have an system with GRUB bootloader and two redundant slots  **roofs.0** and **rootfs1** for the operation system. Lets call them **A** and **B** as in the docs.  This allows the update of the OS system.

### Problem
How/where to add scripts to set GRUB environment variables to switch the partitions?

In the [docs](https://rauc.readthedocs.io/en/latest/examples.html) in 4.1.3 there is this script which sets *A_OK*, *B_OK*, *A_TRY*, *B_TRY* to grubenv and modifies the menuentries in GRUB: 

````shell
default=0
timeout=3

set ORDER="A B"
set A_OK=0
set B_OK=0
set A_TRY=0
set B_TRY=0
load_env

# select bootable slot
for SLOT in $ORDER; do
    if [ "$SLOT" == "A" ]; then
        INDEX=0
        OK=$A_OK
        TRY=$A_TRY
        A_TRY=1
    fi
    if [ "$SLOT" == "B" ]; then
        INDEX=1
        OK=$B_OK
        TRY=$B_TRY
        B_TRY=1
    fi
    if [ "$OK" -eq 1 -a "$TRY" -eq 0 ]; then
        default=$INDEX
        break
    fi
done

# reset booted flags
if [ "$default" -eq 0 ]; then
    if [ "$A_OK" -eq 1 -a "$A_TRY" -eq 1 ]; then
        A_TRY=0
    fi
    if [ "$B_OK" -eq 1 -a "$B_TRY" -eq 1 ]; then
        B_TRY=0
    fi
fi

save_env A_TRY B_TRY

CMDLINE="panic=60 quiet"

menuentry "Slot A (OK=$A_OK TRY=$A_TRY)" {
    linux (hd0,2)/kernel root=/dev/sda2 $CMDLINE rauc.slot=A
}

menuentry "Slot B (OK=$B_OK TRY=$B_TRY)" {
    linux (hd0,3)/kernel root=/dev/sda3 $CMDLINE rauc.slot=B
}
````

If i do understand right, it is not making sense to add this script raw to the **/boot/grub/grubenv** file. But where shall I add this script and when should it be executed?

Could this be added more precisely to the docs?

Many thanks in advance for every help!

Best,
Timo

ejoerns, 2022-02-18:
> Hello guys,
> 
> first of all: Thanks for your great work!

:blush: 

> 
> I successfully installed and tested RAUC in combination with the HawkBit updater following the instructions and examples in the docs. There is one open issue I have, which I was not able to understand from the docs/examples. It is about the slot selection in the GRUB bootloader for a system with redundant rootfs partitions.
> 
> [...]
> 
> If i do understand right, it is not making sense to add this script raw to the **/boot/grub/grubenv** file. But where shall I add this script and when should it be executed?

I would place the grub.cfg next to the bootloader, i.e. normally in your EFI partition.

The grub built-in cfg then should look like

```
search.file ($cmdpath)/EFI/BOOT/grub.cfg root                                                                                                                                                                                                             
set prefix=($root)/EFI/BOOT 
```

Note, if using Yocto, this is already done by grub-efi recipe.
Then an exemplary integration is our community layer:

https://github.com/rauc/meta-rauc-community/tree/master/meta-rauc-qemux86/recipes-bsp/grub

If you plan to make use of atomic bootloader updates [1], then also note that you should not place the env file next to the bootloader but in an independent non-redundant location instead.

The grub.cfg should then use something like

```
load_env --file=(hd0,2)/grubenv
```

[1] https://rauc.readthedocs.io/en/latest/advanced.html#update-boot-partition-in-gpt

> 
> Could this be added more precisely to the docs?

Yes, could :smirk: 

Best regards, Enrico

> Many thanks in advance for every help!
> 
> Best, Timo

TimoK93, 2022-02-28:
Thanks for your response! I will implement it in the next two weeks and close the issue after success :)

PR #931 (2022-06-27): docs/integration: add a hint to specifiy the file used to store GRUB env
Fixes #862
commit efc7c788 (2022-06-27): docs/integration: add a hint to specifiy the file used to store GRUB env

Fixes #862
closed 2022-06-27 by jluebbe