rauc/rauc #864: Failed to load dm table: No such file or directory

erickson-ureta, 2022-02-23:
Hi,

I'm currently investigating on how to migrate upgrade bundle formats for my project from `plain` to `verity`. When I do a `rauc install`, I get this:
```
installing
  0% Installing
  0% Determining slot states
 20% Determining slot states done.
 20% Checking bundle
 20% Verifying signature
 40% Verifying signature done.
 40% Checking bundle done.
 40% Checking manifest contents
 60% Checking manifest contents done.
100% Installing failed.
LastError: Failed mounting bundle: Failed to load dm table: No such file or directory
```
I'm looking into what that error means (something to do with something called device mapper) but I don't have much knowledge as to what they are and how exactly they work (and I never needed to until now, since upgrades with the plain format simply worked for me). Can anyone point me into the right direction? What does a correct "dm table" look like, and how do I go about creating it?

a3f, 2022-02-23:
See the section on [6.13. Bundle Format Migration](https://rauc.readthedocs.io/en/latest/integration.html#bundle-format-migration) and check that you have the relevant kernel configuration options enabled.

erickson-ureta, 2022-02-23:
> See the section on [6.13. Bundle Format Migration](https://rauc.readthedocs.io/en/latest/integration.html#bundle-format-migration) and check that you have the relevant kernel configuration options enabled.

I already have those flags enabled while building my bundle through Yocto. Here's me grepping for the needed flags on a gunzipped copy of `/proc/config.gz`:
```
# grep -nriw "CONFIG_MD" ./config 
1508:CONFIG_MD=y
# grep -nriw "CONFIG_BLK_DEV_DM" ./config 
1512:CONFIG_BLK_DEV_DM=y
# grep -nriw "CONFIG_DM_VERITY" ./config 
1536:CONFIG_DM_VERITY=y
```

jluebbe, 2022-02-24:
Hmm, it seems that list is incomplete. Please also check that you have:
```
CONFIG_BLK_DEV_LOOP=y
CONFIG_SQUASHFS=y
CONFIG_CRYPTO_SHA256=y
```

erickson-ureta, 2022-02-24:
> Hmm, it seems that list is incomplete. Please also check that you have:
> 
> ```
> CONFIG_BLK_DEV_LOOP=y
> CONFIG_SQUASHFS=y
> CONFIG_CRYPTO_SHA256=y
> ```

Ah, looks like I was missing the `CONFIG_CRYPTO_SHA56` option. Everything went fine after that. 

Thanks for the help! Closing issue.

closed 2022-02-24 by erickson-ureta