rauc/rauc #870: Machine readable output is mixed with debug messages

RyuzakiKK, 2022-03-02:
## Steps to reproduce

```
$ G_MESSAGES_DEBUG=all rauc status --output-format=json 2> /dev/null
$ G_MESSAGES_DEBUG=all rauc status --output-format=json > /dev/null
```

## Expected result

In one of the two executions we should get the debug messages while in the other the machine parsable JSON output.

## Actual result

The debug messages are printed in stdout, mixed with the machine parsable output.
Currently there is no way to redirect just the json output somewhere else, e.g. a file.

## Possible solutions

`stdout` could be reserved to user and machine readable output, while using `stderr` for all the other debug/warning/error messages.
Alternatively there could be a launch option like `--output-to-file` that redirects the output, and not the debug messages, in a text file.

ejoerns, 2022-03-17:
This is a result of the current standard behavior of the glib logging that we use.

> stderr is used for levels G_LOG_LEVEL_ERROR, G_LOG_LEVEL_CRITICAL, G_LOG_LEVEL_WARNING and G_LOG_LEVEL_MESSAGE. stdout is used for the rest, unless stderr was requested by g_log_writer_default_set_use_stderr().

Do you need the debug messages when invoking `rauc status`? I would assume they do not tell to much interesting about what is happening as most magic is happening in the service.

Bastian-Krause, 2022-03-17:
See https://gitlab.gnome.org/GNOME/glib/-/issues/2087 for a similar discussion.

ejoerns, 2022-03-18:
Yes, the thing I instantly found in the glib documentation was `g_log_writer_default_set_use_stderr ()` which apparently is also a result of the mentioned discussion. But using this is would increase the required glib version to 2.68 which is not even part of Debian bullseye.

An alternative would be to also implement this on our own first of all.

The question would be, if `G_LOG_LEVEL_MESSAGE` already ended up in stderr, do we have any reason that we want to have only `G_LOG_LEVEL_DEBUG` ending up in stdout instead?

RyuzakiKK, 2022-03-25:
One easy fix that doesn't cut off support for older glib would be to use something like:
```c
#if GLIB_CHECK_VERSION (2, 68, 0)
g_log_writer_default_set_use_stderr()
#endif
```

But as you noted, not everyone has a new enough glib.
Most notably Debian 11 that has glib 2.66 and Ubuntu 20.04 with glib 2.64.

At least the good news is that Ubuntu 21.10 and the upcoming LTS Ubuntu 22.04, both have glib >= 2.68.0

smcv, 2022-04-01:
I'm the GLib contributor who added `g_log_writer_default_set_use_stderr()` and `g_log_writer_default_would_drop()`. @RyuzakiKK asked me for an opinion on this.

I think conditionally using `g_log_writer_default_set_use_stderr (TRUE)` in a `GLIB_CHECK_VERSION` guard is a good 90% solution that will fix this for Debian >= 12, Ubuntu >= 21.10, CentOS Stream >= 9 and all rolling-release distributions. It's not perfect, but it will solve the issue for the future with a very small maintenance burden.

Resolving this for GLib 2.66 (Debian 11, Ubuntu 20.04) is harder, because `g_log_writer_default_would_drop()` is *also* new in GLib 2.68, so you can't benefit from it in older versions. There are basically two ways it could be done:

1. Save the original stdout file handle or file descriptor, overwrite it with a copy of the original stderr so that all log messages come out of the original stderr, and use the saved copy of the original stdout to write machine-readable output (only)
    * Doing this completely reliably requires Unix-specific goo with `dup2()`: I wrote [_srt_divert_stdout_to_stderr()](https://gitlab.steamos.cloud/steamrt/steam-runtime-tools/-/blob/v0.20220315.0/steam-runtime-tools/utils.c#L794) to do this in one of my other projects
    * This is the only thing that can be relied on if you call into third-party libraries that might also write arbitrary noise to stdout
2. Write your own `GLogFunc` and `GLogWriterFunc` that always write to stderr
    * For GLib < 2.68 support, this would require reimplementing the equivalent of `g_log_writer_default_would_drop()`, https://gitlab.gnome.org/GNOME/glib/-/blob/glib-2-72/glib/gmessages.c#L2648, which is tedious but not difficult

If it was up to me, I think I'd just use `g_log_writer_default_set_use_stderr (TRUE)` in a `GLIB_CHECK_VERSION` guard as @RyuzakiKK suggested - unless I knew that the project was highly reliant on machine-readable stdout or was going to call into third-party libraries that might be less careful, in which case I'd do the `dup2()` trick, similar to the `_srt_divert_stdout_to_stderr()` that I linked to above.

ejoerns, 2022-04-01:
@smcv Thank you for your valuable input on this. :+1:  I'd also tend to say that *if* we intend to change this, then using the 2.68 API method is actually the way to go.

closed 2022-04-07 by ejoerns