rauc/rauc #903: Failed to resolve realpath for 'ubi0:root1'

mschwan-phytec, 2022-04-28:
When executing `rauc status`, the following debug message is printed alongside the status on a system using UBIFS:
```
root@hostname:~# rauc status
rauc-Message: 09:24:49.256: Failed to resolve realpath for 'ubi0:root1'
...
```
Looking at the source code in `src/context.c` for `get_cmdline_bootname()` it looks like this does not affect the normal operation of RAUC. But stdlib's `realpath()` does not seem to support [UBI volume paths](https://www.kernel.org/doc/html/latest/filesystems/ubifs.html#quick-usage-instructions) with this format: `ubidevice:volumename` Which produces the above message.

As `r_realpath()` fails but `get_cmdline_bootname()` still returns a "correct" device path for the current boot slot, I wonder if there needs to be an additional detection for UBI volumes of said format in that function?

ejoerns, 2022-06-23:
The call of `realpath()` you see there happens when trying to determine slot states in `determine_slot_state()`.

It is called on the path of the device configured for the slot (in system.conf) to compare it against the bootname.
This was introduced in https://github.com/rauc/rauc/commit/fce9505d566c3ce4f6fbd9dac96dc9806d0450c5 to handle symlink device path's in system.conf.

Potentially, one alternative could be to resolve this when loading the config already (or initializing the context) but this would mean you see only the resolved device paths in `rauc status` output but not the actual ones.

jluebbe, 2022-08-19:
We could add a cache for the path resolution and only print the warning once. Then `rauc status` could still show the original string.