rauc/rauc #908: rauc status: error retrieving slot status via D-Bus: error calling D-Bus method "GetSlotStatus": Timeout was reached

Lazibi, 2022-05-12:
Hi there,

I'm trying to integrate rauc into my yocto project. I've followed the documentation and the meta-rauc-community examples but still can't get it to work. `$ rauc status` gives:

```
(rauc:2747): rauc-WARNING **: 14:38:37.187: Timeout was reached
rauc-Message: 14:39:02.217: rauc status: error retrieving slot status via D-Bus: error calling D-Bus method "GetSlotStatus": Timeout was reached
```

`$ systemctl status rauc-mark-good.service` gives :
```
rauc-mark-good.service - Rauc Good-marking Service
     Loaded: loaded (/lib/systemd/system/rauc-mark-good.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2022-05-11 15:01:52 UTC; 40s ago
    Process: 293 ExecStart=/usr/bin/rauc status mark-good (code=exited, status=1/FAILURE)
   Main PID: 293 (code=exited, status=1/FAILURE)

May 11 15:01:01 AMR83FA51 systemd[1]: Starting Rauc Good-marking Service... May 11 15:01:27 AMR83FA51 rauc[293]: Timeout was reached May 11 15:01:52 AMR83FA51 rauc[293]: rauc status: error retrieving slot status via D-Bus: error calling D-Bus method "GetSlotStatus": Timeout wa> May 11 15:01:52 AMR83FA51 systemd[1]: rauc-mark-good.service: Main process exited, code=exited, status=1/FAILURE May 11 15:01:52 AMR83FA51 systemd[1]: rauc-mark-good.service: Failed with result 'exit-code'. May 11 15:01:52 AMR83FA51 systemd[1]: Failed to start Rauc Good-marking Service.
```

Below you could find the `strace fw_printenv` result :
```

execve("/usr/bin/fw_printenv", ["fw_printenv"], 0xbee8acf0 /* 20 vars */) = 0
brk(NULL)                               = 0xa95000
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6fd9000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_NO_AUTOMOUNT|AT_EMPTY_PATH, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=8984, ...}) = 0
mmap2(NULL, 8984, PROT_READ, MAP_PRIVATE, 3, 0) = 0xb6fd6000
close(3)                                = 0
openat(AT_FDCWD, "/usr/lib/libubootenv.so.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\334\f\0\0004\0\0\0"..., 512) = 512
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_NO_AUTOMOUNT|AT_EMPTY_PATH, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0755, stx_size=13552, ...}) = 0
mmap2(NULL, 77832, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6f9c000
mprotect(0xb6f9f000, 61440, PROT_NONE)  = 0
mmap2(0xb6fae000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2000) = 0xb6fae000
close(3)                                = 0
openat(AT_FDCWD, "/lib/libz.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\310\33\0\0004\0\0\0"..., 512) = 512
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_NO_AUTOMOUNT|AT_EMPTY_PATH, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0755, stx_size=62856, ...}) = 0
mmap2(NULL, 126984, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6f7c000
mprotect(0xb6f8b000, 61440, PROT_NONE)  = 0
mmap2(0xb6f9a000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xe000) = 0xb6f9a000
close(3)                                = 0
openat(AT_FDCWD, "/lib/libc.so.6", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\09\262\1\0004\0\0\0"..., 512) = 512
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_NO_AUTOMOUNT|AT_EMPTY_PATH, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0755, stx_size=935504, ...}) = 0
mmap2(NULL, 1025544, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6e81000
mprotect(0xb6f62000, 61440, PROT_NONE)  = 0
mmap2(0xb6f71000, 16384, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xe0000) = 0xb6f71000
mmap2(0xb6f75000, 26120, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xb6f75000
close(3)                                = 0
set_tls(0xb6fda4e0)                     = 0
mprotect(0xb6f71000, 8192, PROT_READ)   = 0
mprotect(0xb6f9a000, 4096, PROT_READ)   = 0
mprotect(0xb6fae000, 4096, PROT_READ)   = 0
mprotect(0x481000, 4096, PROT_READ)     = 0
mprotect(0xb6fdb000, 4096, PROT_READ)   = 0
munmap(0xb6fd6000, 8984)                = 0
brk(NULL)                               = 0xa95000
brk(0xab6000)                           = 0xab6000
openat(AT_FDCWD, "/etc/fw_env.config", O_RDONLY) = 3
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_NO_AUTOMOUNT|AT_EMPTY_PATH, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=70, ...}) = 0
read(3, "/dev/mmcblk3p2 0xF00000  0x20000"..., 4096) = 70
readlink("/dev", 0xbebcc2b0, 1023)      = -1 EINVAL (Invalid argument)
readlink("/dev/mmcblk3p2", 0xbebcc2b0, 1023) = -1 EINVAL (Invalid argument)
fstatat64(AT_FDCWD, "/dev/mmcblk3p2", {st_mode=S_IFBLK|0660, st_rdev=makedev(0xb3, 0xa), ...}, 0) = 0
openat(AT_FDCWD, "/dev/mmcblk3p2", O_RDONLY) = 4
close(4)                                = 0
readlink("/dev", 0xbebcc2b0, 1023)      = -1 EINVAL (Invalid argument)
readlink("/dev/mmcblk3p1", 0xbebcc2b0, 1023) = -1 EINVAL (Invalid argument)
fstatat64(AT_FDCWD, "/dev/mmcblk3p1", {st_mode=S_IFBLK|0660, st_rdev=makedev(0xb3, 0x9), ...}, 0) = 0
openat(AT_FDCWD, "/dev/mmcblk3p1", O_RDONLY) = 4
close(4)                                = 0
close(3)                                = 0
openat(AT_FDCWD, "/var/lock/fw_printenv.lock", O_WRONLY|O_CREAT|O_TRUNC, 0666) = 3
flock(3, LOCK_EX	
```

Anyone could help me to figure this out and make rauc works properly?!

jluebbe, 2022-05-12:
> openat(AT_FDCWD, "/var/lock/fw_printenv.lock", O_WRONLY|O_CREAT|O_TRUNC, 0666) = 3
> flock(3, LOCK_EX	
> ```

That looks like there is another fw_printenv process holding that lock. You should probably first tests that the fw_* tools work correctly. When they work fine, RAUC should have not problems using them.

Lazibi, 2022-05-12:
> That looks like there is another fw_printenv process holding that lock. You should probably first tests that the fw_* tools work correctly. When they work fine, RAUC should have not problems using them.

My knowledge is not very good in this part, how could I test the fw_* tools and check if it works properly?

MitchGaines, 2022-06-03:
Any updates on this issue? Any idea which process has been introduced that is acquiring this lock? Or are there any linked issues in the appropriate communities which address this issue?

This is a problem that I'm also experiencing.

jluebbe, 2022-06-03:
There's nothing we can do from RAUC's perspective here. It's just calling fw_printenv as a subprocess.

If you can reproduce it, you should be able to check which other process is holding the lock.

MitchGaines, 2022-06-06:
@jluebbe So, when I run `lslocks`, among other things, I get the following:
```
fw_printenv      1952 FLOCK      WRITE 0     0   0 /run/lock/fw_printenv.lock
```

When I `ps aux | grep 1952` (for the PID), I get the following:
```
root        1952 99.5  0.0   2280   520 ?        R    16:48   2:29 fw_printenv BOOT_ORDER:
```

Now, its reasonable to assume that other processes might be kicking off fw_printenv as a subprocess, but, when I run `pstree -p 1952 -s`, it shows that rauc is actually the parent process

```
root@mydevice:~# pstree -p 1952 -s
systemd(1)---rauc(795)---fw_printenv(1952)
```

Am I reading this correctly?

MitchGaines, 2022-06-10:
I suppose my issue is a bit different from the original issue that was brought up. My error is being induced by `fw_printenv` hanging. Below is my `strace` for `fw_printenv`. The condition in rauc is the same as a result of this error though. For what its worth, I cannot run `fw_printenv` successfully either. 

```
execve("/usr/bin/fw_printenv", ["fw_printenv"], 0x7fff66c1c0 /* 15 vars */) = 0
brk(NULL)                               = 0x5598644000
faccessat(AT_FDCWD, "/etc/ld.so.preload", R_OK) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=31940, ...}) = 0
mmap(NULL, 31940, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f8cf59000
close(3)                                = 0
openat(AT_FDCWD, "/usr/lib/libubootenv.so.0", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0 \23\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=18176, ...}) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f8cf57000
mmap(NULL, 81936, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f8cf1d000
mprotect(0x7f8cf21000, 61440, PROT_NONE) = 0
mmap(0x7f8cf30000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x3000) = 0x7f8cf30000
close(3)                                = 0
openat(AT_FDCWD, "/lib/libz.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\360%\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=92056, ...}) = 0
mmap(NULL, 155664, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f8cef6000
mprotect(0x7f8cf0c000, 61440, PROT_NONE) = 0
mmap(0x7f8cf1b000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x15000) = 0x7f8cf1b000
close(3)                                = 0
openat(AT_FDCWD, "/lib/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\200B\2\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=1421400, ...}) = 0
mmap(NULL, 1494016, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f8cd89000
mprotect(0x7f8cedd000, 65536, PROT_NONE) = 0
mmap(0x7f8ceed000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x154000) = 0x7f8ceed000
mmap(0x7f8cef3000, 11264, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f8cef3000
close(3)                                = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f8cf55000
mprotect(0x7f8ceed000, 12288, PROT_READ) = 0
mprotect(0x7f8cf1b000, 4096, PROT_READ) = 0
mprotect(0x7f8cf30000, 4096, PROT_READ) = 0
mprotect(0x556ab20000, 4096, PROT_READ) = 0
mprotect(0x7f8cf63000, 4096, PROT_READ) = 0
munmap(0x7f8cf59000, 31940)             = 0
brk(NULL)                               = 0x5598644000
brk(0x5598665000)                       = 0x5598665000
openat(AT_FDCWD, "/etc/fw_env.config", O_RDONLY) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=32, ...}) = 0
read(3, "/boot/uboot.env 0x0000  0x20000\n", 4096) = 32
newfstatat(AT_FDCWD, "/boot", {st_mode=S_IFDIR|0755, st_size=16384, ...}, AT_SYMLINK_NOFOLLOW) = 0
newfstatat(AT_FDCWD, "/boot/uboot.env", {st_mode=S_IFREG|0755, st_size=126976, ...}, AT_SYMLINK_NOFOLLOW) = 0
newfstatat(AT_FDCWD, "/boot/uboot.env", {st_mode=S_IFREG|0755, st_size=126976, ...}, 0) = 0
openat(AT_FDCWD, "/boot/uboot.env", O_RDONLY) = 4
close(4)                                = 0
read(3, "", 4096)                       = 0
close(3)                                = 0
openat(AT_FDCWD, "/var/lock/fw_printenv.lock", O_WRONLY|O_CREAT|O_TRUNC, 0666) = 3
flock(3, LOCK_EX)                       = 0
mmap(NULL, 135168, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f8cd68000
openat(AT_FDCWD, "/boot/uboot.env", O_RDONLY) = 4
read(4, "\315\270f\201BOOT_A_LEFT=2\0BOOT_B_LEFT=3\0"..., 131072) = 126976
read(4, "", 4096)                       = 0
read(4, "", 4096)                       = 0
read(4, "", 4096)                       = 0
...
read(4, "", 4096)                       = 0
read(4, "", 4096)                       = 0
read(4, "", 4096)                       = 0
```

Now, this obviously isn't an issue with Rauc, specifically, so I understand there's not a lot you can do. Mostly just updating this with more findings in case someone else has ideas. I will open an issue with [libubootenv](https://github.com/sbabic/libubootenv) as well, I suppose.

MitchGaines, 2022-06-10:
Actually, I just figured out the issue on my end. My `fw_env.config` was misconfigured :facepalm:

For the Raspberry Pi CM4, it should read as `/boot/uboot.env 0x0000  0x4000`.

ejoerns, 2022-06-13:
@MitchGaines Thanks for sharing your findings! Maybe this helps others running into similar issues to know where to have a look at (`fw_env.config`).

I would say we can close this now (and I'll just do so) as there has been nothing (also not in the original report) that suggests that there is an error within RAUC.

closed 2022-06-13 by ejoerns