rauc/rauc #913: RAUC with a dm-verity-protected partition failes

davidhuwyler, 2022-05-23:
With RAUC v1.6, dm-verity-protected partition are not updated correctly. The verity information is not copied (sits at the end of the partition). It remains 0!
With RAUC v1.5 there was no problem at all. The solution from issue #624 does not work with v1.6.

As in the FAQ stated, i used type=raw and statusfile=</path/to/global.status>

I tried updating the parititon with dd from the rauc bundle: This works fine!

It seems as the size for the slot is not correct.

I tried to make out the code which causes this but I didnt find the issue...

davidhuwyler, 2022-05-24:
> > With RAUC v1.6, dm-verity-protected partition are not updated correctly. The verity information is not copied (sits at the end of the partition). It remains 0! With RAUC v1.5 there was no problem at all. The solution from issue #624 does not work with v1.6.
> 
> You can see that the hash tree is in the fs image that is put into the bundle,but the image written to disk does not have the hash tree anymore?

Its the verity superblock which is not copied. The superblock lives after the end of the partition image.
I can see this block in the partition-file in the bundle (after loop / mounting the bundle). After rauc install, the bock is not in the destination partition.

> > As in the FAQ stated, i used type=raw and statusfile=</path/to/global.status>
> 
> Sounds right, yes.
> 
> > I tried updating the parititon with dd from the rauc bundle: This works fine!
> 
> So you mounted the bundle and dd'ed the image contained, right?

Yes i did losetup and mounted the bundle. Then i copied the rootfs-partition image with dd to the destination partition.
Worked fine...

> > It seems as the size for the slot is not correct.
> 
> The slot does not have any size information set.

The filesize? 

> > I tried to make out the code which causes this but I didnt find the issue...
> 
> The code responsible for writing should be https://github.com/rauc/rauc/blob/master/src/update_handler.c#L1924
> 
> Do you see anything in the logs that the write fails? Could you provide the output of the RAUC service runnig with debug enabled (`-d`)?

Ok The install with -d:

```
rauc install -d stick/update.raucb
rauc-Message: 06:20:02.427: Debug log domains: 'rauc'
(rauc:522): rauc-DEBUG: 06:20:02.434: install started
(rauc:522): rauc-DEBUG: 06:20:02.434: input bundle: /home/root/stick/update.raucb
(rauc:522): rauc-DEBUG: 06:20:02.501: Trying to contact rauc service
installing
  0% Installing
  0% Determining slot states
20% Determining slot states done.
20% Checking bundle
20% Verifying signature
40% Verifying signature done.
40% Checking bundle done.
[  136.202674] loop0: detected capacity change from 0 to 212708
[  136.242089] loop0: detected capacity change from 212708 to 212704
40% Checking manifest contents[
60% Checking manifest contents done.
  136.288947] device-mapper: verity: sha256 using implementation "sha256-generic"
60% Determining target install group
80% Determining target install group done.
80% Updating slots
80% Checking slot rootfs.1
85% Checking slot rootfs.1 done.
85% Copying image to rootfs.1
90% Copying image to rootfs.1 done.
90% Checking slot fitimage.1
95% Checking slot fitimage.1 done.
95% Copying image to fitimage.1
100% Copying image to fitimage.1 done.
100% Updating slots done.
100% Installing done.
Installing `/home/root/stick/update.raucb` succeeded

```

After that i did "rauc service -d":

```
rauc service -d
rauc-Message: 06:22:30.879: Debug log domains: 'rauc'
(rauc:552): rauc-DEBUG: 06:22:30.886: No value for key "max-bundle-download-size" in [system] defined - using default value of 8388608 bytes.
rauc-Message: 06:22:30.886: Using central status file /etc/rauc/central-status.raucs
(rauc:552): rauc-DEBUG: 06:22:30.887: service start
(rauc:552): rauc-DEBUG: 06:22:30.888: No value for key "max-bundle-download-size" in [system] defined - using default value of 8388608 bytes.
rauc-Message: 06:22:30.888: Using central status file /etc/rauc/central-status.raucs
Failed to obtain name de.pengutronix.rauc on system bus
```

ejoerns, 2022-05-24:
> Its the verity superblock which is not copied. The superblock lives after the end of the partition image. I can see this block in the partition-file in the bundle (after loop / mounting the bundle). [...]

I wonder how this can happen because at this level RAUC just takes an image and writes it. It cannot determine which is the actual FS and which is the verity information.

> Yes i did losetup and mounted the bundle. Then i copied the rootfs-partition image with dd to the destination partition. Worked fine...

Ok. Note that with 1.6 you should also be able to just to `rauc mount` for debugging purpose.

> > > It seems as the size for the slot is not correct.
> > 
> > The slot does not have any size information set.
> 
> The filesize?

So you mean the image size in the manifest? In our terminology the slot is the partition or whatever on the target.

So you can see in the manifest or rauc info that the size of the image is smaller than the actual image size in the bundle?
Could you check if the hash matches?

Is this a 'plan', 'verity' or 'crypt' bundle?

> Ok The install with -d:

This will not provide further details. What I was referring to is starting the *service* with debug output. The actual RAUC process runs in background and communicates with the command line interface client over D-Bus. 

> After that i did "rauc service -d":
> 

So the right order is to do this *before* running the installation and keeping this running in parallel

If you have systemd you can also handle this adding -d to the systemd unit file (don't forget to daemon-reload then).

The output (for systemd) is then in the journal:

   journalctl -u rauc

Also refer to https://rauc.readthedocs.io/en/latest/using.html#debugging-rauc for details on debugging.

> ```
> rauc service -d
> rauc-Message: 06:22:30.879: Debug log domains: 'rauc'
> (rauc:552): rauc-DEBUG: 06:22:30.886: No value for key "max-bundle-download-size" in [system] defined - using default value [...]

davidhuwyler, 2022-05-24:
Ah  rauc mount is a nice tool 👍 
Its a verity bundle.
Im not using systemd. Its a yocto build with sysVinit.

The size reported by  rauc info is the same as the file size of the slot file... but i saw this concerning error in the rauc info output (A0FFFDB6:error:....) :

 

```
rauc info stick/update.raucb
rauc-Message: 08:26:50.676: Using central status file /etc/rauc/central-status.raucs
rauc-Message: 08:26:50.677: valid /etc/rauc/system.conf found, using it
rauc-Message: 08:26:50.677: Reading bundle: /home/root/stick/update.raucb
rauc-Message: 08:26:50.694: Verifying bundle signature...
rauc-Message: 08:26:50.707: Verified inline signature by 'O = Test Org, CN = Test Org Development-1'
A0FFFDB6:error:1700006B:CMS routines:cms_get_enveloped_type:content type not enveloped data:../openssl-3.0.2/crypto/cms/cms_env.c:41:
A0FFFDB6:error:1700006B:CMS routines:cms_get_enveloped_type:content type not enveloped data:../openssl-3.0.2/crypto/cms/cms_env.c:41:
Compatible:     'CyOne_IPVPN_Module'
Version:        '1.0'
Description:    'ipvpn-module-bundle version 1.0-r0'
Build:          '20220523122426'
Hooks:          'install-check'
Bundle Format:  verity
  Verity Salt:  '266d1df5296bb35d15b0213bd9f614253dc646720e443829358e071babb81ab8'
  Verity Hash:  '0e3776dbe106cb342c98239bf94959e21c398148dbbad5293f4459b49d62db6c'
  Verity Size:  860160
 

2 Images:
  [rootfs]
        Filename:  rootfs.img
        Checksum:  439944a6db771bb6c7d57e93204c1726b59af44a3deb0e00f321c6ead7eecd2d
        Size:      290672640
        Hooks:     install post-install
  [fitimage]
        Filename:  fitImage-partition.img
        Checksum:  aaf49e4598d2206f99a660fbefc6973c71fc709361d9df6c2a9fde6706eb985b
        Size:      134217728
        Hooks:
 

Certificate Chain:
0 Subject: O = Test Org, CN = Test Org Development-1
   Issuer: O = Test Org, CN = Test Org rauc CA Development
   SPKI sha256: 7C:0D:C4:9C:D5:05:46:EF:93:84:94:83:61:76:EF:65:45:68:0D:E8:3C:94:82:96:59:4A:8B:E3:F1:39:B1:69
   Not Before: Jan  1 00:00:00 1970 GMT
   Not After:  Dec 31 23:59:59 9999 GMT
1 Subject: O = Test Org, CN = Test Org rauc CA Development
   Issuer: O = Test Org, CN = Test Org rauc CA Development
   SPKI sha256: 17:D8:B0:B4:DD:20:9F:5A:19:C6:8A:7B:5F:3A:79:C4:DE:19:B4:AC:4D:95:DE:0E:A4:09:0E:3A:F4:7E:C2:0B
   Not Before: Jan  1 00:00:00 1970 GMT
   Not After:  Dec 31 23:59:59 9999 GMT

```

Does RAUC work with openssl 3.0.2?

ejoerns, 2022-05-24:
> Ah rauc mount is a nice tool +1 Its a verity bundle. Im not using systemd. Its a yocto build with sysVinit.

Ok.

> The size reported by rauc info is the same as the file size of the slot file... but i saw this concerning error in the rauc info output (A0FFFDB6:error:....) :
>
> ```
> rauc info stick/update.raucb [...]

These error come from inside OpenSSL and are more or less known issues for us. OpenSSL checks the CMS type and the checks that internally fail propagate their error to the outer world, which in my impression is not how it should behave.

However, this is no functional problem.

> Compatible:     'CyOne_IPVPN_Module'
> Version:        '1.0'
> Description:    'ipvpn-module-bundle version 1.0-r0'
> Build:          '20220523122426'
> Hooks:          'install-check'
> Bundle Format: [...]

Yes, it does. At least as far as we have tested. Just a few modifications were required to make RAUC OpenSSL 3.0 compatible. Some more will probably be done in the future, but currently these only affect APIs that are still kept in OpenSSL for the sake of compatiblity.
As noted above, there are no relevant errors in OpenSSL processing.

ejoerns, 2022-05-24:
> I now tested starting the service with -d. But there was no different output and no errors for rauc install...

I haven't seen any bit of output from the service during installation, yet.

If the size and the checksum in the manifest are correct (you did not clarify tha latter one), I really have no clue what should go wrong.

davidhuwyler, 2022-05-24:
Ok here is the full output for rauc install with debug turned on:

 

```
rauc install -d stick/update.raucb
rauc-Message: 09:18:39.658: Debug log domains: 'rauc'
(rauc:686): rauc-DEBUG: 09:18:39.665: install started
(rauc:686): rauc-DEBUG: 09:18:39.666: input bundle: /home/root/stick/update.raucb
(rauc:686): rauc-DEBUG: 09:18:39.688: Trying to contact rauc service
input bundle: /home/root/stick/update.raucb
rauc-Message: 09:18:39.691: Active slot bootname: A
(rauc:682): rauc-DEBUG: 09:18:39.692: thread started for /home/root/stick/update.raucb
rauc-Message: 09:18:39.692: installing /home/root/stick/[update.raucb: started
(rauc:682): rauc-DEBUG: 09:18:39.694: Found booted slot: rootfs.0 on /dev/mmcblk0p2
  918.780924] loop0: detected capacity change from 0 to 212708
installing
rauc-Message: 09:18:39.697: installing /home/root/stick/update.raucb: Checking and mounting bundle...
  0% Installing
  0% Determining slot states
rauc-Message: 09:18:39.698: Reading bundle: /home/root/stick/update.raucb
20% Determining slot states done.
20% Checking bundle
rauc-Message: 09:18:39.715: Verifying bundle signature...
20% Verifying signature
rauc-Mes[sage: 09:18:39.729: Verified inline signature by 'O = Test Org, CN = Test Org Development-1'
40F33FB4:error:1700006B:CMS routines:cms_get_enveloped_type:content type not enveloped data:../openssl-3.0.2/crypto/cms/cms_env.c:41:
40F33FB4:error:1700006B:CMS routines:cms_get_enveloped_type:content type not enveloped data:../openssl-3.0.2/crypto/cms/cms_env.c:41:
(rauc:682): rauc-DEBUG: 09:18:39.731: Got 2 elements for trust chain
40% Verifying signature done.
rauc-Message: 09:18:39.733: Mounting bundle '/home/root/stick/update.raucb' to '/mnt/rauc/bundle'
40% Checking bundle done.
  918.852073] loop0: detected capacity change from 212708 to 212704
rauc-Message: 09:18:39.904: Configured loop device '/dev/loo[p0' for 108904448 bytes
40% Checking manifest contents
60% Checking manifest contents done.
  918.958310] device-mapper: verity: sha256 using implementation "sha256-generic"
rauc-Message: 09:18:39.932: Configured dm-verity device '/dev/dm-1'
(rauc:682): rauc-DEBUG: 09:18:39.943: Checking slot: rootfs.1
(rauc:682): rauc-DEBUG: 09:18:39.943:   Adding mapping: rootfs -> rootfs.1
(rauc:682): rauc-DEBUG: 09:18:39.943: Checking slot: fitimage.1
(rauc:682): rauc-DEBUG: 09:18:39.943:   Adding mapping: fitimage -> fitimage.1
(rauc:682): rauc-DEBUG: 09:18:39.944: Checking slot: rootfs.0
(rauc:682): rauc-DEBUG: 09:18:39.944:   No mapping found
(rauc:682): rauc-DEBUG: 09:18:39.944: Checking slot: fitimage.0
(rauc:682): rauc-DEBUG: 09:18:39.944:   No mapping found
60% Determining target install group
80% Determining target install group done.
(rauc:682): rauc-DEBUG: 09:18:39.945: Using default installation handler
(rauc:682): rauc-DEBUG: 09:18:39.948: Found image mapping: rootfs.img -> rootfs
(rauc:682): rauc-DEBUG: 09:18:39.948: Found image mapping: fitImage-partition.img -> fitimage
rauc-Message: 09:18:39.949: Running bundle hook install-check
rauc-Message: 09:18:39.964: Marking target slot rootfs.1 as non-bootable...
rauc-Message: 09:18:40.036: installing /home/root/stick/update.raucb: Updating slots...
rauc-Message: 09:18:40.037: installing /home/root/stick/update.raucb: Checking slot rootfs.1
80% Updating slots
(rauc:682): rauc-DEBUG: 09:18:40.039: Load status for slot rootfs.1.
(rauc:682): rauc-DEBUG: 09:18:40.039: Load status for slot fitimage.1.
(rauc:682): rauc-DEBUG: 09:18:40.039: Load status for slot rootfs.0.
(rauc:682): rauc-DEBUG: 09:18:40.040: Load status for slot fitimage.0.
rauc-Message: 09:18:40.040: Updating /dev/mmcblk0p4 with 'install' slot hook
rauc-Message: 09:18:40.041: installing /home/root/stick/update.raucb: Updating slot rootfs.1
rauc-Message: 09:18:40.041: Running slot hook 'slot-install' for rootfs.1
80% Checking slot rootfs.1
85% Checking slot rootfs.1 done.
85% Copying image to rootfs.1
(rauc:682): rauc-DEBUG: 09:18:40.057: Saving global slot status
rauc-Message: 09:18:40.058: installing /home/root/stick/update.raucb: Updating slot rootfs.1 status
90% Copying image to rootfs.1 done.
rauc-Message: 09:18:40.076: Checking image type for slot type: raw
rauc-Message: 09:18:40.076: installing /home/root/stick/update.raucb: Updating slot rootfs.1 done
rauc-Message: 09:18:40.077: installing /home/root/stick/update.raucb: Checking slot fitimage.1
90% Checking slot fitimage.1
95% Checking slot fitimage.1 done.
rauc-Message: 09:18:40.079: Updating /dev/mmcblk0p3 with /mnt/rauc/bundle/fitImage-partition.img
rauc-Message: 09:18:40.082: opening slot device /dev/mmcblk0p3
rauc-Message: 09:18:40.080: installing /home/root/stick/update.raucb: Updating slot fitimage.1
rauc-Message: 09:18:40.083: writing data to device /dev/mmcblk0p3
95% Copying image to fitimage.1
(rauc:682): rauc-DEBUG: 09:18:47.771: Saving global slot status
rauc-Message: 09:18:47.772: installing /home/root/stick/update.raucb: Updating slot fitimage.1 status
100% Copying image to fitimage.1 done.
rauc-Message: 09:18:47.778: Marking target slot rootfs.1 as bootable...
rauc-Message: 09:18:47.778: installing /home/root/stick/update.raucb: Updating slot fitimage.1 done
(rauc:682): rauc-DEBUG: 09:18:47.835: Saving global slot status
rauc-Message: 09:18:47.842: installing /home/root/stick/update.raucb: All slots updated
100% Updating slots done.
(rauc:682): rauc-DEBUG: 09:18:47.921: thread finished for /home/root/stick/update.raucb
rauc-Message: 09:18:47.921: installing /home/root/stick/update.raucb: finished
rauc-Message: 09:18:47.922: installing `/home/root/stick/up[date.raucb` succeeded
100% Installing done.
Installing `/home/root/stick/update.raucb` succeeded
  927.005021] device-mapper: ioctl: remove_all left 1 open device(s)
```

Can you spot any problems? Hmm im runing out of ideas :-(

ejoerns, 2022-05-24:
By having a look at your install log I just discovered that I missed something that was already visible in the `rauc info` output.

You are using a custom install hook.

a) why?
b) if used, it's in the responsibility of the hook to do image copying right, not in RAUC's

davidhuwyler, 2022-05-24:
Oh sh** Sure you are right! This was a test i inserted. Im sorry for wasting your time.

Thanks! Great support!

closed 2022-05-24 by davidhuwyler

ejoerns, 2022-05-24:
I guess we have all run into similar self-set traps before :smirk: 

I'm fine with knowing RAUC works as intended :grin: