rauc/rauc #930: Rauc status: poweroff after mark-active not syncing to disk?

taylor-winning, 2022-06-23:
It appears that after calling `rauc status mark-active rootfs.0` and then shortly after turning off the power of our device results in the call to `mark-active` not taking effect (e.g. mark-active slot 0 is rebooting into slot 1 after power is cut). 

It seems as though there is no sync call to flush changes to the file system. Calling sync manually mitigates this behavior. Do you think sync should be called by rauc or is there another solution.

ejoerns, 2022-06-23:
The command invokes communication with the bootloader. Which command is executed for this depends on which bootloader you have configured.

So which one do you use? And how/where does it store its state?

taylor-winning, 2022-06-23:
> The command invokes communication with the bootloader.

Ah, yes of course. Apologies for not including that info initially. We're using u-boot, and storing its environment in the /boot partition. 

Thanks

ejoerns, 2022-06-24:
All that RAUC does here is calling `fw_setenv <key> <value>`. Thus it should be reproducible without using RAUC, right?

I would have expected that U-Boot host tools care for syncing already. Which version are you using?

I've found this patch and discussion according to which an fsync should be called since 2017 U-Boot:

https://patchwork.ozlabs.org/project/uboot/patch/1369078482-5863-1-git-send-email-mhei@heimpold.de/

taylor-winning, 2022-06-27:
You are right, when I manually call `fw_setenv` the issue also still stands, so it seems the issue is not with RAUC at all. We're using 2021.01 u-boot. Thank you for your support.

closed 2022-06-27 by taylor-winning