rauc/rauc #933: OTAs without degrading system performance

om26er, 2022-07-06:
There are systems where it is desirable to automatically update the systems in the background. One important aspect in such system is to not degrade system performance while doing updates. It would be interesting to explore something like that with rauc as well.

For streaming mode I guess we could introduce some logic to take care of that. I do understand that this also means doing slow OTA updates.

@RyuzakiKK it may actually be relatively simpler with desync however we'd probably need to write custom code around desync to do updates slowly.

ejoerns, 2022-07-06:
How system performance is degraded actually depends on which resources are used and if they compete.

If your application is CPU-bound and your update is I/O-bound, it might not be that critical to run both in parallel.

Anyway, I would not see this to be something one should re-implement in RAUC as modern Linux systems already have good capabilities for controlling the resource consumption of applications.

With systemd and cgroups one should have the tools available for gaining what you suggested, see:
https://www.freedesktop.org/software/systemd/man/systemd.resource-control.html

om26er, 2022-07-06:
The inspiration comes from Google Pixel devices, which do have A/B partitioning scheme, in some cases they do super slow OTA updates in the background without any noticeable performance degradation for the user.

Think of a system with a camera that is recording video, that is both CPU-bound and I/O-bound operation. For such systems with frequent background updates the performance penalty of an OTA is quite high and I believe we can minimize that.

My plan is to extend desync (the library) and do a PoC to test this out.

jluebbe, 2022-07-07:
As @ejoerns said, cgroups are the correct mechanism to configure resource limits and sharing. They will then apply to all processes and accesses done by RAUC.

Take a look at https://www.kernel.org/doc/html/latest/admin-guide/cgroup-v2.html#io-interface-files, which can be configured using https://www.freedesktop.org/software/systemd/man/systemd.resource-control.html#IOReadBandwidthMax=device%20bytes (in the RAUC service file itself or in a snippet).

jluebbe, 2022-08-19:
We think this is better handled at the system level using cgroups. Please reopen if you have use-cases that can't be handled that way.

closed 2022-08-19 by jluebbe