rauc/rauc #953: add a way to pass config options via the command line

jluebbe, 2022-08-19:
This is the remaining open aspect of #729.

> > * [ ]   we should add a way to pass config options via the command line
> 
> Hi all, what is the status concerning the 2nd task "pass config options via the command line"? Will this be realized in near future? Thanks, Daniel

We don't have concrete plans for this, but would be happy to discuss concrete design and take a PR.

_Originally posted by @jluebbe in https://github.com/rauc/rauc/issues/729#issuecomment-1220659737_

ch-aurich, 2024-09-12:
Hi,
I'd like to contribute on this one... anything I need to be aware of before starting work?
Best Regards
Christian

jluebbe, 2024-09-12:
> I'd like to contribute on this one... anything I need to be aware of before starting work?

From the user's perspective, it could probably look like `rauc -c base-system.conf -C keyring:check-purpose=codesign -C keyring:check-crl=false`. I'm not sure if `:` is a good separator between section and key, but at least it's unused so far.

You'll need to collect all such options from the command line and add them to the `RaucContext`. The `load_config()` function can access them from there. Then, after parsing the config into a `GKeyFile`, you'd use [`g_key_file_set_value()`](https://docs.gtk.org/glib/method.KeyFile.set_value.html) to extend/override the loaded config file. This way, the existing loading of the individual options can stay as-is.

PR #1542 (2025-01-21): Add -C option to overwrite config file parameters
add -C option to set/overwrite config file parameters (currently only for keyring:check-crl=false and keyring:check-purpose=codesign)

Fixes: #953
closed 2025-01-21 by jluebbe