rauc/rauc #986: src/nbd: wrong dict name for tls_no_verify

stefanu21, 2022-10-03:
https://github.com/rauc/rauc/blob/bfbf94caf4e3deb3fcef69c8c14a3ac673707153/src/nbd.c#L955

I think the dict entry should be 'tls-no-verify' and not 'no-veirfy'

ejoerns, 2022-10-03:
It is consistent between all dict keys that the `tls` is left out (since the scope should be clear).

As this is only 'internal' API between RAUC and its helper it should also not be that important how it's named.

Or how did you came across this or is that an issue for a use case of yours?

stefanu21, 2022-10-03:
I was a little bit confused because I do a 
`rauc info --tls-no-verify <url>`
and I saw 
`rauc[1305]: sending: {'url': <'https://192.168.100.250:5000/update.raucb'>, 'no-verify': <true>, 'headers': <['Model:CB1', 'SerialNumber:00000099', 'Version:v1.56a-62-g45c2d5d0afc1']>}`
and we also have the option `rauc info --no-verify <url>`

jluebbe, 2022-10-04:
In the context of the internal protocol to the NBD server, all these entries apply to TLS, as the bundle signature is not handled there. We could change it later, if it becomes useful.

ejoerns, 2023-01-27:
Since this is neither broken nor important for 'external' usage, I guess we could close this first of all.

closed 2023-01-27 by ejoerns