rauc/rauc #989: LastError: Installation error: Destination device '/dev/mmcblk0p3' already mounted ？

gitgonewithwind, 2022-10-05:
Hi all,
I tried to run  rauc  install, but the error happened.before I always umount my redundancy rootfs:   mmcblk0p3  : slot.rootfs.1
LastError: Installation error: Destination device '/dev/mmcblk0p3' already mounted ？
how to fix it ?Maybe this is a linux question not rauc.
any help thanks?

my sh-4.4# cat /etc/rauc/system.conf

 [system]
 compatible=imx8mqevk
 bootloader=uboot
# 
[keyring]
path=/etc/rauc/ca.cert.pem
[slot.rootfs.0]
device=/dev/mmcblk0p2
type=ext4
bootname=system0

 [slot.rootfs.1]
device=/dev/mmcblk0p3
type=ext4
bootname=system1

umount /run/media/mmcblk0p3
-sh-4.4# lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sda           8:0    1 28.5G  0 disk 
`-sda1        8:1    1 28.5G  0 part /run/media/sda1
mmcblk0     179:0    0 29.7G  0 disk 
|-mmcblk0p1 179:1    0 41.7M  0 part /run/media/mmcblk0p1
|-mmcblk0p2 179:2    0  7.4G  0 part /
`-mmcblk0p3 179:3    0  7.4G  0 part 

rauc  install /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb 
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb: installing
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb:   0% Installing
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb:   0% Determining slot states
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb:  20% Determining slot states done.
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb:  20% Checking bundle
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb:  20% Verifying signature
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb:  40% Verifying signature done.
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb:  40% Checking bundle done.
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb:  40% Loading manifest file
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb:  60% Loading manifest file done.
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb:  60% Determining target install group
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb:  80% Determining target install group done.
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb:  80% Updating slots
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb: 100% Updating slots failed.
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb: 100% Installing failed.
rauc-Message: installing /run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb: LastError: Installation error: Destination device '/dev/mmcblk0p3' already mounted
Installing `/run/media/sda1/fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb` failed

ejoerns, 2022-10-05:
Which RAUC version do you test on?

This somehow looks like something that has been fixed with #485.

gitgonewithwind, 2022-10-05:
@ejoerns 
Thank you so much for reply so soon.
-sh-4.4# rauc --version
rauc 1.0

--
 my yocto is sumo, so  ported  https://github.com/rauc/meta-rauc/tree/sumo
Can I use  master branch ?   https://github.com/rauc/meta-rauc   for yocto sumo ??? 
let me try

ejoerns, 2022-10-05:
Don't use RAUC 1.0 anymore.

Don't use Yocto Sumo anymore.

Case closed.

closed 2022-10-05 by ejoerns

gitgonewithwind, 2022-10-06:
@ejoerns 
sorry  bother you again. 
does the rauc support  yocto zues and hardknott ? I find the current master branch at https://github.com/rauc/meta-rauc is yocto langdale.
Thank you very much.

Bastian-Krause, 2022-10-06:
All of these releases are EOL: https://wiki.yoctoproject.org/wiki/Releases

Please use a supported release!

gitgonewithwind, 2022-10-06:
@Bastian-Krause 
you posted info is for yocto. I mean meta-rauc  can support what yocto release? sumo ,zues .I found  https://github.com/rauc/meta-rauc  has sumo ,zues  branch. 
but after trying, now I know sumo do not works well. so I want to try zues or  hardknott branch ?Am I correct?

Bastian-Krause, 2022-10-06:
We don't support Yocto releases that reached EOL, so while you can use the meta-rauc branches with these release names, you will still get an outdated version of RAUC.

gitgonewithwind, 2022-10-06:
got it. 
I only can use yocto langdale(current master branch),right?
if that,give up.

Bastian-Krause, 2022-10-06:
We try to maintain branches for all non-EOL releases: https://wiki.yoctoproject.org/wiki/Releases

gitgonewithwind, 2022-10-06:
got it. thanks