rauc/rauc #995: rauc install  will make kernel crash? raucb file size is too big?

gitgonewithwind, 2022-10-25:
Hi all,
sorry to bother you again,
I am playing rauc in maaxboard (DDR 2G),but when I install .raucb file, the kernel crashed.
is it   file size is too big? my  fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb is about 1.2G ??

ls -l lite-image-bundle-rauc-maaxboard-ddr4-2g-sdcard-20221025154232.raucb
-rwxrwx--- 1 root vboxsf 1183774230 Oct 25 12:02 lite-image-bundle-rauc-maaxboard-ddr4-2g-sdcard-20221025154232.raucb

Thank you

rauc install /run/media/
mmcblk0p1/ mmcblk0p3/ sda1/      
-sh-5.0# rauc install /run/media/sda1/
ProdRec/
dump.sdcard
dump.sdcard_2
dump_188416.sdcard
dump_both.sdcard
dump_s.sdcard
fram.dat
fsl-image-validation-imx-bundle-rauc-imx8mqevk.raucb
json/
lite-image-bundle-rauc-maaxboard-ddr4-2g-sdcard-20221025154232.raucb
log/
lost+found/
srv/

rauc install media/sda1/lite-image-bundle-rauc-maaxboard-ddr4-2g-sdcard-20221025154232.raucb
installing
  0% Installing
  0% Determining slot states
 20% Determining slot states done.
 20% Checking bundle
 20% Verifying signature
[   88.562769] Unable to handle kernel paging request at virtual address ffff800011813060
[   88.564556] Unable to handle kernel paging request at virtual address ffff800011813070
[   88.570720] Mem abort info:
[   88.570724]   ESR = 0x96000046
[   88.576426] Unable to handle kernel paging request at virtual address ebccd8e5f5aee452
[   88.576428] Mem abort info:
[   88.576430]   ESR = 0x96000044
[   88.576433]   EC = 0x25: DABT (current EL), IL = 32 bits
[   88.576435]   SET = 0, FnV = 0
[   88.576437]   EA = 0, S1PTW = 0
[   88.576438] Data abort info:
[   88.576440]   ISV = 0, ISS = 0x00000044
[   88.576442]   CM = 0, WnR = 1
[   88.576445] [ebccd8e5f5aee452] address between user and kernel address ranges
[   88.576449] Internal error: Oops: 96000044 [#1] PREEMPT SMP
[   88.576452] Modules linked in: brcmfmac crct10dif_ce brcmutil ov5640_camera_mipi_v2 overlay
[   88.576468] CPU: 2 PID: 0 Comm: swapper/2 Not tainted 5.4.24-2.1.0+g491748038398 #1           8�1�F����?�~�X�HR� ʘ�A�bT(
[   88.576472] ��U�rv�Rg���T:�k0�"��������H�2�����Q�NݒN݇&b��E_k$�Xq�3�CL���|g�k�@N'�$Q�DL]�+�
[   88.576472] ��=�
[   88.576475] pstate: 20000085 (nzCv daIf -PAN -UAO)
[   88.576489] pc : sysctr_set_next_event+0x1c/0x88
[   88.576496] lr : clockevents_program_event+0xa4/0x118
[   88.576498] sp : ffff800011cebde0
[   88.576500] x29: ffff800011cebde0 x28: 0000000000000000 
[   88.576504] x27: ffff800011b9d010 x26: 0000000000000000 
[   88.576509] x25: 0000000000000002 x24: 0000000000000004 
[   88.576513] x23: ffff0000bd9bbf00 x22: ffff800011ae5900 
[   88.576517] x21: 0000000000000001 x20: 000081cf10cc1b10 
[   88.576521] x19: ffff800011ae5900 x18: 000000000000000e 
[   88.576525] x17: 0000000000000001 x16: 0000000000000019 
[   88.576529] x15: 00000000000000c1 x14: 0000000000000000 
[   88.576533] x13: 0000000000000003 x12: 00000000000000c1 
[   88.576537] x11: 071c71c71c71c71c x10: 0000000000000abb 
[   88.576541] x9 : ffff0000bd9b5f08 x8 : 00000000000008e7 
[   88.576545] x7 : 0000000000000f67 x6 : 000000002eb0828e 
[   88.576550] x5 : ffff800011c54de0 x4 : ebccd8e5f5aee452 
[   88.576554] x3 : ffff800011c54000 x2 : ebccd8e5f5ade426 
[   88.576558] x1 : 000000004db9c77b x0 : 00000000000081cf 
[   88.576562] Call trace:
[   88.576567]  sysctr_set_next_event+0x1c/0x88
[   88.576571]  tick_broadcast_set_event+0x40/0xe0
[   88.576576]  __tick_broadcast_oneshot_control+0x294/0x328
[   88.576579]  tick_broadcast_oneshot_control+0x30/0x38
[   88.576587]  cpuidle_enter_state+0x174/0x360
[   88.576590]  cpuidle_enter+0x34/0x48
[   88.576596]  call_cpuidle+0x18/0x38
[   88.576599]  do_idle+0x1e0/0x280
[   88.576603]  cpu_startup_entry+0x20/0x40
[   88.576611]  secondary_start_kernel+0x154/0x190
[   88.576619] Code: b94de061 f94004a2 91404044 9100b084 (b9000081) 
[   88.576629] ---[ end trace cf9d7d44aeb3635b ]---
[   88.576634] Kernel panic - not syncing: Attempted to kill the idle task!
[   88.576638] SMP: stopping secondary CPUs
[   88.578667] Kernel Offset: disabled
[   88.578671] CPU features: 0x0002,2000200c
[   88.578672] Memory Limit: none
[   88.842116] ---[ end Kernel panic - not syncing: Attempted to kill the idle task! ]---

ejoerns, 2022-10-25:
Which RAUC version is this?

Is it a 'plain' or a 'verity' bundle?

gitgonewithwind, 2022-10-26:
-sh-5.0# rauc --version                                                    
rauc 1.2  

I think it's plain bundle for rootfs
my file  /etc/rauc/system.conf:
cat /etc/rauc/system.conf                                         
## This is an example RAUC system configuration. This file will be installd
## into /etc/rauc/system.conf on your target and describes your system froe
## perspective of the RAUC update service.                                 
##                                                                         
## Adapt and extend the below configuration to your needs and place it in e
## BSP layer of you project. Create a rauc .bbappend file that adds this fe
## to your build:                                                          
##                                                                         
##   FILESEXTRAPATHS_prepend := "${THISDIR}/files:"                        
##                                                                         
##   SRC_URI_append := " file://system.conf"                               
##                                                                         
## ---                                                                     
##                                                                         
 [system]                                                                  
 compatible=maaxboard-zeus                                                 
 bootloader=uboot                                                          
#                                                                          
[keyring]                                                                  
path=/etc/rauc/ca.cert.pem                                                 
[slot.rootfs.0]                                                            
device=/dev/mmcblk0p2                                                      
type=ext4                                                                  
bootname=system0                                                           
                                                                           
 [slot.rootfs.1]                                                           
device=/dev/mmcblk0p3                                                      
type=ext4                                                                  
bootname=system1                                                           
#[slot.appfs.0]                                                            
# device=/home/myuser/STM_encrypted.out                                    
# type=ext4                                                                
# parent=rootfs.0                                                          
                                                                           
[handlers]                                                                 
#pre-install=/bin/mount --options remount,rw /

jluebbe, 2022-10-26:
The kernel oops above looks like memory corruption could be an issue. Do you see similar issues during other operation which do a lot of I/O? Perhaps try with stress-ng?

gitgonewithwind, 2022-10-26:
I did see similar issues before.
I worry about the file raucb is too big (1.2G), but the board DDR 2G is small, when the rauc install to unzip raucb to the rootfs(roofs is about 6.2G) , so the it  is out of memory?
I am going to build a smaller rootfs and update some small files to test.

gitgonewithwind, 2022-10-26:
@jluebbe 
I changed a new board and I reduced the size of .raucb zsize to 800M and rootfs to 2.6G, so error happened.
it seems the kernel has bug??

0% Installing
  0% Determining slot states
 20% Determining slot states done.
 20% Checking bundle
 20% Verifying signature
[   69.991377] Unable to handle kernel paging request at virtual address 30a920f
[   69.999315] Mem abort info:
[   70.002157]   ESR = 0x96000004
[   70.005238]   EC = 0x25: DABT (current EL), IL = 32 bits
[   70.010578]   SET = 0, FnV = 0
[   70.013651]   EA = 0, S1PTW = 0
[   70.016812] Data abort info:
[   70.019711]   ISV = 0, ISS = 0x00000004
[   70.023567]   CM = 0, WnR = 0
[   70.026569] [30a92027834b6ddf] address between user and kernel address ranges
[   70.033745] Internal error: Oops: 96000004 [#1] PREEMPT SMP
[   70.039316] Modules linked in: brcmfmac brcmutil crct10dif_ce ov5640_camera_y
[   70.047677] CPU: 0 PID: 32 Comm: kworker/0:1 Not tainted 5.4.24-2.1.0+g491741
[   70.072978] Workqueue: events dbs_work_handler��Y�T���CHIP�J��|r��q���     �
[   70.077421] pstate: 20000005 (nzCv daif -PAN -UAO)
[   70.082213] pc : __cpufreq_driver_target+0x60/0x5b0
[   70.087089] lr : od_dbs_update+0xb8/0x190
[   70.091096] sp : ffff800011e63cc0
[   70.094407] x29: ffff800011e63cc0 x28: ffff800011a06000 
[   70.099717] x27: 0000000000000000 x26: ffff800011cabce8 
[   70.105028] x25: 0000000000000000 x24: ffff0000b8d39200 
[   70.110338] x23: ffff800011c53000 x22: ffff0000b82e8e00 
[   70.115648] x21: 0000000000000000 x20: ffff0000b8beff00 
[   70.120959] x19: ffff0000b8d42800 x18: 000000000000000e 
[   70.126269] x17: 0000000000000001 x16: 0000000000000019 
[   70.131579] x15: 0000000000000033 x14: 000000000000b000 
[   70.136890] x13: 0000000000000000 x12: 0000000000000001 
[   70.142200] x11: 071c71c71c71c71c x10: 00000000000009c0 
[   70.147510] x9 : ffff0000b812546c x8 : 30a92027834b6da7 
[   70.152821] x7 : 0000000000000018 x6 : 000000000016e360 
[   70.158130] x5 : 000000000016e360 x4 : 0000000000000000 
[   70.163440] x3 : 00000000000f4240 x2 : 0000000000000002 
[   70.168750] x1 : 0000000000128a18 x0 : ffff0000b8d42800 
[   70.174061] Call trace:
[   70.176507]  __cpufreq_driver_target+0x60/0x5b0
[   70.181035]  od_dbs_update+0xb8/0x190
[   70.184696]  dbs_work_handler+0x3c/0x78
[   70.188533]  process_one_work+0x198/0x320
[   70.192540]  worker_thread+0x48/0x420
[   70.196203]  kthread+0xf0/0x120
[   70.199345]  ret_from_fork+0x10/0x18
[   70.202922] Code: a90363f7 d0008137 f946a2e8 b9008806 (f9401d07) 
[   70.209015] ---[ end trace eb468d3c47db53be ]---
[   70.300972] Unable to handle kernel paging request at virtual address ffff800
[   70.308929] Mem abort info:
[   70.311727]   ESR = 0x96000046
[   70.314833]   EC = 0x25: DABT (current EL), IL = 32 bits
[   70.316015] Unable to handle kerneress ffff800011813070
[   70.320173]   SET = 0, FnV = 0
[   70.320176]   EA = 0, S1PTW = 0
[   70.320177] Data abort info:
[   70.320179]   ISV = 0, ISS = 0x00000046
[   70.320181]   CM = 0, WnR = 1
[   70.320185] swapper pgtable: 4k pages, 48-bit VAs, pgdp=0000000041b13000
[   70.320188] [ffff800011813060] pgd=00000000fdfff003, pud=00000000fdffe003, p0
[   70.320195] Internal error: Oops: 96000046 [#2] PREEMPT SMP
[   70.320198] Modules linked in: brcmfmac brcmutil crct10dif_ce ov5640_camera_y
[   70.320411] Unable to handle kernel paging request at virtual address 00631f5
[   70.320414] Mem abort info:
[   70.320416]   ESR = 0x96000044
[   70.320418]   EC = 0x25: DABT (current EL), IL = 32 bits
[... 123 lines cut ...]
[   71.110498]  __handle_domain_irq+0x60/0xb8
[   71.114593]  gic_handle_irq+0x5c/0x148
[   71.118340]  el1_irq+0xb8/0x180
[   71.121479]  __do_softirq+0xa4/0x23c
[   71.125054]  irq_exit+0xb8/0xd8
[   71.128194]  __handle_domain_irq+0x64/0xb8
[   71.132287]  gic_handle_irq+0x5c/0x148
[   71.136034]  el1_irq+0xb8/0x180
[   71.139175]  mark_page_accessed+0xf0/0x1d8
[   71.143271]  unmap_single_vma+0x48/0xa0
[   71.147104]  unmap_vmas+0x6c/0xe0
[   71.150418]  exit_mmap+0xc4/0x180
[   71.153731]  mmput+0x74/0x148
[   71.156697]  do_exit+0x2a0/0x980
[   71.159923]  do_group_exit+0x40/0xa0
[   71.163500]  get_signal+0xd0/0x688
[   71.166901]  do_notify_resume+0x174/0x460
[   71.170909]  work_pending+0x8/0x10
[   71.174311] Code: b94de022 f9400461 91404021 9100b021 (b9000022) 
[   71.180401] ---[ end trace eb468d3c47db53c1 ]---
[   71.887600] SMP: failed to stop secondary CPUs 0-2
[   71.892389] Kernel Offset: disabled
[   71.895875] CPU features: 0x0002,2000200c
[   71.899882] Memory Limit: none
[   71.902939] ---[ end Kernel panic - not syncing: Attempted to kill the idle -

jluebbe, 2022-10-27:
These call traces point to multiple unrelated places in the kernel, so I'm relatively sure that this is not a RAUC-specific issue. You could try to reproduce it with https://github.com/ColinIanKing/stress-ng or a memory test (either in your bootloader or under linux).

gitgonewithwind, 2022-10-27:
@jluebbe 
got it.
thank you very much.
Avent vendor does not have newer kernel for maaxboard(maaxboard-mini). I am going to post the issue on nxp forum, I hope I can get patches.

closed 2022-10-27 by jluebbe