rauc/rauc #998: Automatically disable HTTP streaming when bundle is in plain format?

RyuzakiKK, 2022-10-26:
In RAUC v1.7 it has been added the ability to stream a bundle without the need to first download it.

However this functionality has several requirements. One of those is that bundles must use the `verity` format.

If you compile RAUC and you meet the pre-requisites (like having `libnl-genl-3.0`), the streaming support will be automatically enabled, unless you explicitly use `--disable-streaming`. This is also the case for the RAUC build that is available in the Arch Linux repositories https://archlinux.org/packages/community/x86_64/rauc/

I guess this is a fair default option, but the problem is that once the HTTP streaming is enabled, you can't install `plain` bundles anymore. All `plain` bundles will fail with:
`"Bundle format 'plain' not supported in streaming mode"`

Are you interested in having a fallback behavior where RAUC will automatically disable the streaming option when it notices that you are using a `plain` bundle? Or alternatively having a runtime configuration option to toggle between HTTP streaming and download?

Before starting to implement a solution I'd like to hear your opinion first.

ejoerns, 2022-10-26:
> I guess this is a fair default option, but the problem is that once the HTTP streaming is enabled, you can't install `plain` bundles anymore. All `plain` bundles will fail with: `"Bundle format 'plain' not supported in streaming mode"`

To clarify, you can't install plain bundles with using RAUC as the *downloader* anymore, yes.

Note that the download capability that RAUC had before streaming was actually only intended to be used together with *casync bundles* and thus has a default limitation to a few MiB of bundle size.

The 'plain' format is considered deprecated. We still support it, but only with the effort required to keep it usable and secure.

Thus I am not that convinced about taking such an option and would propose to switch to verity format instead.

Is there anything that prevents you from using the verity bundle format?

RyuzakiKK, 2022-10-26:
> To clarify, you can't install plain bundles with using RAUC as the downloader anymore, yes.

Right, so a short-term alternative would be to download the `.raucb` with something else, e.g. wget, curl etc..., and then launch `rauc install` with this local file.

> The 'plain' format is considered deprecated. We still support it, but only with the effort required to keep it usable and secure.
>
> Thus I am not that convinced about taking such an option and would [...]

Okay, that makes sense. I didn't realize that `plain` was basically deprecated.
We didn't switch to the verity bundle format yet because it requires some additional QA, especially to guarantee backwards compatibility. But given the `plain` status, I guess that the switch will likely be our medium-long term goal.

Thanks for your quick reply, I guess we can close this.

closed 2022-10-26 by RyuzakiKK

ejoerns, 2022-10-26:
> Okay, that makes sense. I didn't realize that `plain` was basically deprecated. We didn't switch to the verity bundle format yet because it requires some additional QA, especially to guarantee [...]

We are trying to make this more obvious from release to release. But we will surely still support 'plain' for some time.

A general good recommendation is to already enable everything required for verity in the system as quickly as possible (mainly the kernel options noted here: https://rauc.readthedocs.io/en/latest/integration.html#kernel-configuration)

The earlier this is done the higher the probability that all systems out in the field already support verity bundles.