rauc/rauc #1000: Failed mounting bundle: failed to resolve 'nbd' netlink family - BLK_DEV_NBD not enabled in kernel?

MitchGaines, 2022-10-28:
Hi there, 

I am attempting to set up HTTP streaming (for use with adaptive updates) of rauc bundles from a hawkbit instance to my end devices (RPi Compute Module 4). I am running a dunfell Yocto distribution on my system with rauc v1.8 and [linux-raspberrypi] kernel version 5.4.83. Historically, I have been installing rauc bundles via hawkbit without streaming with no issue.

When I attempt to stream a bundle, this is the error I get:
```
Oct 28 08:55:21 my-device-1234 rauc[254]: input bundle: http://hawkbit-instance:8080/DEFAULT/controller/v1/my-device-1234/softwaremodules/45/artifacts/update-bundle-my-device-20221027172750.raucb
Oct 28 08:55:21 my-device-1234 rauc[254]: Active slot bootname: B
Oct 28 08:55:22 my-device-1234 rauc[254]: Remote URI detected, streaming bundle...
Oct 28 08:55:22 my-device-1234 rauc[254]: installing http://hawkbit-instance:8080/DEFAULT/controller/v1/my-device-1234/softwaremodules/45/artifacts/update-bundle-my-device-20221027172750.raucb: started
Oct 28 08:55:22 my-device-1234 rauc[254]: installing http://hawkbit-instance:8080/DEFAULT/controller/v1/my-device-1234/softwaremodules/45/artifacts/update-bundle-my-device-20221027172750.raucb: Checking and mounting bundle...
Oct 28 08:55:22 my-device-1234 rauc[254]: sending: {'url': <'http://hawkbit-instance:8080/DEFAULT/controller/v1/my-device-1234/softwaremodules/45/artifacts/update-bundle-my-device-20221027172750.raucb'>}
Oct 28 08:55:22 my-device-1234 rauc-nbd[5615]: running as UID 65534, GID 65534
Oct 28 08:55:22 my-device-1234 rauc-nbd[5615]: received: {'url': <'http://hawkbit-instance:8080/DEFAULT/controller/v1/my-device-1234/softwaremodules/45/artifacts/update-bundle-my-device-20221027172750.raucb'>}
Oct 28 08:55:22 my-device-1234 rauc-nbd[5615]: configuring for URL: http://hawkbit-instance:8080/DEFAULT/controller/v1/my-device-1234/softwaremodules/45/artifacts/update-bundle-my-device-20221027172750.raucb
Oct 28 08:55:22 my-device-1234 rauc-nbd[5615]: file date 1666891933
Oct 28 08:55:22 my-device-1234 rauc-nbd[5615]: total size 413247409
Oct 28 08:55:22 my-device-1234 rauc[254]: received total size 413247409
Oct 28 08:55:22 my-device-1234 rauc[254]: received current time 0
Oct 28 08:55:22 my-device-1234 rauc[254]: received modified time 1666891933
Oct 28 08:55:22 my-device-1234 rauc[254]: nbd server started
Oct 28 08:55:22 my-device-1234 rauc[254]: Reading bundle: http://hawkbit-instance:8080/DEFAULT/controller/v1/my-device-1234/softwaremodules/45/artifacts/update-bundle-my-device-20221027172750.raucb
Oct 28 08:55:22 my-device-1234 rauc[254]: Verifying bundle signature... 
Oct 28 08:55:22 my-device-1234 rauc[254]: Verified inline signature by 'O = Test Org, CN = Test Org Development-1'
Oct 28 08:55:22 my-device-1234 rauc[254]: Mounting bundle 'http://hawkbit-instance:8080/DEFAULT/controller/v1/my-device-1234/softwaremodules/45/artifacts/update-bundle-my-device-20221027172750.raucb' to '/run/rauc/bundle'
Oct 28 08:55:22 my-device-1234 rauc[254]: configuring nbd device
Oct 28 08:55:22 my-device-1234 rauc[254]: nbd server stopping
Oct 28 08:55:22 my-device-1234 rauc-nbd[5615]: nbd dl_size: count=3 sum=1973.000 min=4.000 max=1961.000 avg=657.667 recent-avg=657.667
Oct 28 08:55:22 my-device-1234 rauc-nbd[5615]: nbd dl_speed: count=3 sum=46926.000 min=50.000 max=46690.000 avg=15642.000 recent-avg=15642.000
Oct 28 08:55:22 my-device-1234 rauc-nbd[5615]: nbd namelookup: count=3 sum=0.002 min=0.000 max=0.001 avg=0.001 recent-avg=0.001
Oct 28 08:55:22 my-device-1234 rauc-nbd[5615]: nbd connect: count=3 sum=0.036 min=0.000 max=0.036 avg=0.012 recent-avg=0.012
Oct 28 08:55:22 my-device-1234 rauc-nbd[5615]: nbd starttransfer: count=3 sum=0.166 min=0.042 max=0.080 avg=0.055 recent-avg=0.055
Oct 28 08:55:22 my-device-1234 rauc-nbd[5615]: nbd total: count=3 sum=0.167 min=0.043 max=0.080 avg=0.056 recent-avg=0.056
Oct 28 08:55:22 my-device-1234 rauc-nbd[5615]: exiting nbd server
Oct 28 08:55:22 my-device-1234 rauc[254]: nbd server stopped
Oct 28 08:55:22 my-device-1234 rauc[254]: Failed mounting bundle: failed to resolve 'nbd' netlink family - BLK_DEV_NBD not enabled in kernel?
Oct 28 08:55:22 my-device-1234 rauc[254]: installing http://hawkbit-instance:8080/DEFAULT/controller/v1/my-device-1234/softwaremodules/45/artifacts/update-bundle-my-device-20221027172750.raucb: Failed mounting bundle: failed to resolve 'nbd' netlink family - BLK_DEV_NBD not enabled in kernel?
Oct 28 08:55:22 my-device-1234 rauc[254]: installing http://hawkbit-instance:8080/DEFAULT/controller/v1/my-device-1234/softwaremodules/45/artifacts/update-bundle-my-device-20221027172750.raucb: finished
Oct 28 08:55:22 my-device-1234 rauc[254]: installing `http://hawkbit-instance:8080/DEFAULT/controller/v1/my-device-1234/softwaremodules/45/artifacts/update-bundle-my-device-20221027172750.raucb` failed: 1
```

Based on this error, I would think this has something to do with the proper kernel configs not being present.
However, these are the kernel configs I am adding: 
I have confirmed that they do make it into the kernel running on the system by inspecting the `/proc/configs.gz`
```
CONFIG_SQUASHFS=y
CONFIG_SQUASHFS_FILE_CACHE=y
CONFIG_SQUASHFS_DECOMP_SINGLE=y
CONFIG_SQUASHFS_ZLIB=y
CONFIG_SQUASHFS_FRAGMENT_CACHE_SIZE=3
CONFIG_CRYPTO_SHA256=y
CONFIG_MD=y
CONFIG_BLK_DEV_LOOP=y
CONFIG_BLK_DEV_DM=y
CONFIG_DM_VERITY=y
CONFIG_BLK_DEV_NBD=y
```

In the v1.8 documentation, a note in section 6.4 addresses a potential kernel bug in reference to mounting bundles. The commit it references (`7e81f99afd91c937f0e66dc135e26c1c4f78b003`) doesn't cleanly apply to the linux-raspberrypi kernel, so I crafted my own patch based on that patch. See below:
This does not appear to have had any impact on the outcome. 
```
diff --git a/drivers/block/loop.c b/drivers/block/loop.c
index ffbe792410d1..db734fb2af64 100644
--- a/drivers/block/loop.c
+++ b/drivers/block/loop.c
@@ -1559,16 +1559,16 @@ static int loop_set_block_size(struct loop_device *lo, unsigned long arg)
 	if (arg < 512 || arg > PAGE_SIZE || !is_power_of_2(arg))
 		return -EINVAL;
 
-	if (lo->lo_queue->limits.logical_block_size != arg) {
-		sync_blockdev(lo->lo_device);
-		invalidate_bdev(lo->lo_device);
-	}
+	if (lo->lo_queue->limits.logical_block_size != arg) 
+    return 0;
+
+	sync_blockdev(lo->lo_device);
+	invalidate_bdev(lo->lo_device);
 
 	blk_mq_freeze_queue(lo->lo_queue);
 
 	/* invalidate_bdev should have truncated all the pages */
-	if (lo->lo_queue->limits.logical_block_size != arg &&
-			lo->lo_device->bd_inode->i_mapping->nrpages) {
+  if (lo->lo_device->bd_inode->i_mapping->nrpages) {
 		err = -EAGAIN;
 		pr_warn("%s: loop%d (%s) has still dirty pages (nrpages=%lu)\n",
 			__func__, lo->lo_number, lo->lo_file_name,
```

Among other items the docs say to implement:
1. Configure RAUC to enable streaming
I have created a rauc bbappend with the following line:
`PACKAGECONFIG="service network streaming"`

2. Create bundles using the verity format
The bundle I am attempting to install was built with `RAUC_BUNDLE_FORMAT = "verity"` andded to my `update-bundle.bbappend`. Historically, I have not been using the verity format. By this, I mean that the image currently running was installed as a plain bundle (not sure this matters though). 

3. Host the bundle on a server which supports HTTP Range Requests
`curl -I http://hawkbit-instance:8080/DEFAULT/controller/v1/my-device-1234/softwaremodules/45/artifacts/update-bundle-my-device-20221027172750.raucb` among other things returns: `Accept-Ranges: bytes`

4. Enable NBD support in the kernel
Addressed at the beginning. 

I am unclear how to proceed. It seems to be able to run the nbd server which makes me think things are running as they should. Any suggestions on next steps for debugging? ]

Thanks for the help. Let me know if there is any additional information I can provide.

ejoerns, 2022-11-02:
> Oct 28 08:55:22 my-device-1234 rauc[254]: Failed mounting bundle: failed to resolve 'nbd' netlink family - BLK_DEV_NBD not enabled in kernel?

This means RAUC cannot configure NBD via its netlink API. The call that fails here is from libn the `genl_ctrl_resolve()` call for `nbd` family. This is normally the case if the family is not known or available which means that it is not compiled in.

However, if you are sure that `BLK_DEV_NBD` is enabled and supported by your kernel, I don't really have a clue what is going wrong, unfortunately. I have not tested with RPI kernel yet, but mainline supports this (the netlink API for nbd) since v4.12, thus this should not be the problem here.

MitchGaines, 2022-11-09:
Thanks for the insight @ejoerns. I was hoping it was something obvious I was missing.  

I ended up writing my updated image with NBD support enabled in the kernel directly to my emmc (as opposed to doing an update through rauc). That seems to work with the items outlined in my OP. I suspect this is going to be a result of the kernel not updating to the extent that I thought it was. Oddly, however, the configs seem to update. Probably a configuration problem related to trying to update the /boot partition (where my kernel image lives).

In any case, that is a different issue. So, I'll go ahead and close this out.

closed 2022-11-09 by MitchGaines

ejoerns, 2022-11-09:
> I ended up writing my updated image with NBD support enabled in the kernel directly to my emmc (as opposed to doing an update through rauc). That seems to work with the items outlined in my OP. I [...]

Couldn't you have used RAUC to update, but without using streaming capabilities? I.e. copy the bundle manually to the target and then install it using RAUC?

Ok, so having the config updated but not the kernel image itself sounds like a good explanation and let's me assume that there is nothing broken between rauc and the kernel that we were not aware of yet. Thanks for the update :+1:

MitchGaines, 2022-11-10:
> Couldn't you have used RAUC to update, but without using streaming capabilities? I.e. copy the bundle manually to the target and then install it using RAUC?

Right, this is what I was doing. My plan was to first update the devices to _support_ streaming. This is the update where it was not updating the kernel to the extent that I thought it was (updating the configs, but not the image). Then the second update is where the failure manifests itself in that the error is thrown when it tries to set up the NBD device.