rauc/rauc #1004: RAUC Long Term Support

mbessler, 2022-11-02:
I didn't find any discussions on the topic of RAUC long term support, so I wanted to open this discussion...

Background: 
I develop Linux systems that are usually deployed for long period (10-15 years, sometime longer) before being replaced.
As they are connected systems, they receive regular security updates, and sometime new features.

We chose RAUC because it matched our needs very closely. Because we track Yocto's LTS releases, new RAUC
releases will be pulled in with updates. 
My issue is that I'm seeing certain features being (apparently) deprecated that we depend on. 
Most recently the "plain" format bundles (which were the only supported format when we started). 

Another potential issue down the road is that RAUC is very pedantic (usually a good thing) about
configuration parameters: if unknown/unsupported parameters are found (in system.conf or in the bundle manifest)
it will error out. Esp the bundle manifest handling is problematic when an older version of RAUC running
on a deployed system is tasked with upgrading to a much newer version that may have additional settings in the .raucm.

That adds certain constraints to upgrades (version X has to first upgrade to version Y before it can upgrade to desired version Z).
(Yes, some of our customers sometimes do not upgrade for a long time, but then suddenly want a new feature requiring a relatively large version jump.) 

Is there any plan for a Long-Term Support series of versions of RAUC that we could track?
I'd rather not fork RAUC and maintain it for the next decade or so..., neither do I favor a rewrite
of our software upgrade process. 

I also wonder how others handle similar situations in real production environments...

ejoerns, 2022-11-02:
> My issue is that I'm seeing certain features being (apparently) deprecated that we depend on.
> Most recently the "plain" format bundles (which were the only supported format when we started).

Currently, our main goal is to prevent people from using RAUC `plain` bundles for new projects. This is why we slightly increase the warning lamps that go on when using it. This does not mean that we will quickly get rid of the `plain` format but there is the chance that at some point in the future such a decision might be made. And if, then mainly because of maintenance overhead and lacking test coverage.

This brings us to LTS. While I understand your points, I have to point out that stability and reliability of features or fixes is the best in the environment these were developed for and are mainly tested with. And this is normally not in an old LTS backport, but a recent release version. Also, the longer an LTS runs, the more effort it takes to backport bugfix commits and the more likely it is that they do not behave as the original commit.

This is why we currently mainly focus on keeping RAUC backwards compatible and thus to allow using recent releases instead.

I also have to point out that a significant aspect of RAUC's security concept lies withing the bundle verification and thus the bundle format. Thus, when we would drop support for `plain` in a regular releases, you can also not expect backports related to `plain` anymore.

This brings me back to your motivation for using LTS:

> Most recently the "plain" format bundles (which were the only supported format when we started).

As noted above, it should currently not require to maintain an LTS. BUT, I would highly recommend to prepare all systems in the field to potentially support `verity` bundles. I know it can be hard to determine which versions are in the field, but if you have shipped verity-compatible systems for 5 years there might be a high chance to not break too many non-compatible ones.

> My issue is that I'm seeing certain features being (apparently) deprecated that we depend on.

What other features are deprecated that you depend on?

> Another potential issue down the road is that RAUC is very pedantic (usually a good thing) about
configuration parameters: if unknown/unsupported parameters are found (in system.conf or in the bundle manifest)
it will error out.

It should only warn if existing configuration was invalid anyway. But since this can be solved (and tested) entirely within a shipped system, this should not be a migration issue.

> Esp the bundle manifest handling is problematic when an older version of RAUC running
on a deployed system is tasked with upgrading to a much newer version that may have additional settings in the .raucm.

New features in manifests should normally be optional due to this. Do you have examples where mandatory options were added that old versions of RAUC could not handle?

> That adds certain constraints to upgrades (version X has to first upgrade to version Y before it can upgrade to desired version Z).
(Yes, some of our customers sometimes do not upgrade for a long time, but then suddenly want a new feature requiring a relatively large version jump.)

I can at least confirm that yours are not the only projects with this constellation.

> Is there any plan for a Long-Term Support series of versions of RAUC that we could track?

So, given my intro above: No, there is none at the moment.

> I'd rather not fork RAUC and maintain it for the next decade or so..., neither do I favor a rewrite
of our software upgrade process.

It should also be the goal of RAUC not requiring you to do this.

> I also wonder how others handle similar situations in real production environments...

I would also be interested if others already face similar issues (or fear facing in the future).

jluebbe, 2022-11-03:
> That adds certain constraints to upgrades (version X has to first upgrade to version Y before it can upgrade to desired version Z). (Yes, some of our customers sometimes do not upgrade for a long time, but then suddenly want a new feature requiring a relatively large version jump.)

I'd argue that you'd want to prohibit large version jumps (perhaps skipping a major version) anyway, perhaps using an [install-check hook](https://rauc.readthedocs.io/en/latest/using.html#bundle-based-customization-hooks). Without this check, the number of migrations you'd need to test would keep growing with every release and become a bottleneck in your release process at some point.

With some constraints to the allowed upgrades, you only have a limited amount of (similar) versions that the migrations need to work and be tested for. When trying to install a too new bundle, it would return an error without modifying the system instead of possibly triggering an untested edge case.

Of course this means that a system that is running a very old version needs to be upgraded in multiple steps, but that is often a small price to pay for avoiding a large test matrix.

mbessler, 2022-11-04:
@ejoerns regarding LTS, given RAUC has one very specific purpose, that is to facilitate upgrades (and sometimes downgrades) of (typically embedded) Linux systems, my main concerns are:
- I can upgrade older software (running an older version of RAUC) to newer software (bundle generated by likely a newer RAUC version)
- Once running the newer software (with newer rauc binary) have to option to downgrade again to the older version (whose bundle was built by a much older RAUC version)
- Bugfixes
- Security fixes

Say RAUC version 1.7.0 was declared LTS, companies such as the one I work for can say "We want to use a RAUC 1.7 compatible version for the time being". We would not have to worry much (aka heavy testing) about incompatible changes until we decide to jump to a newer LTS version and then and only then do in-depth testing between the two LTS versions and how they interact in different scenarios.

When I need new features, I understand I will have to upgrade to a newer RAUC version (or wait).

Optimally, the RAUC LTS would align with a Yocto LTS. 

My problem is that when I upgrade to a newer Yocto LTS, I'm forced to update `meta-rauc` which may no longer support the RAUC version I intended to build upon. So my choices are: fork `meta-rauc`, copy recipes from `meta-rauc` to my own layers, or use whatever RAUC version the layer supports at that time.
Copying recipes or forking means that I take on maintenance and tracking bugfixes (or let it languish) and then deal with bigger deltas once I decide to upgrade RAUC. 

> New features in manifests should normally be optional due to this. Do you have examples where mandatory options were added that old versions of RAUC could not handle?

I'm pretty sure I ran across that once, but could not remember what options gave me trouble.

> Thus, when we would drop support for `plain`

Any idea what timeframe or version this is slated for?

> Without this check, the number of migrations you'd need to test would keep growing with every release and become a bottleneck in your release process at some point.

@jluebbe Well, this is less of a problem as we use a squashfs (ie read-only) rootfs and have a central configuration registry that controls all the configurable elements in our appliance. Thanks for the tip, I will keep the install-check hook in mind for the future.

jluebbe, 2022-11-04:
> @ejoerns regarding LTS, given RAUC has one very specific purpose, that is to facilitate upgrades (and sometimes downgrades) of (typically embedded) Linux systems, my main concerns are:
>
> * I can [...]

We currently don't have the resources to keep two releases (current and LTS) maintained at the same level, so we have to focus on where the most benefit is for community users (and for our customers). I'd like to avoid introducing a bug to an LTS version with a back-port, only because we weren't able to test it in all relevant scenarios.

> When I need new features, I understand I will have to upgrade to a newer RAUC version (or wait).
>
> Optimally, the RAUC LTS would align with a Yocto LTS.
>
> My problem is that when I upgrade to a newer [...]

As long as you don't use new features (by enabling them in the manifest), bundles generated by new RAUC versions are intended to be installable on old versions. If that doesn't work, we'd regard that as a bug.

Features that would break compatibility (such as verity bundles, encryption, adaptive updates, new hooks, ...) are always opt-in via the manifest.

We should probably make that more obvious in the documentation.

> So my choices are: fork `meta-rauc`, copy recipes from `meta-rauc` to my own layers, or use whatever RAUC version the layer supports at that time. Copying recipes or forking means that I take on [...]

So far, we've not removed any features from RAUC, so the current release can build bundles that are installable by very old releases. The decision of when (and how exactly) we'd do so would very much depend on feedback from users. Accordingly, we'd be interested in which (deprecated) features are still needed by users.

> > New features in manifests should normally be optional due to this. Do you have examples where mandatory options were added that old versions of RAUC could not handle?
> 
> I'm pretty sure I ran across that once, but could not remember what options gave me trouble.

Hmm, as above, we'd regard that as a bug and would like to fix it, so we'd be interested in specifics.

> > Thus, when we would drop support for `plain`
> 
> Any idea what timeframe or version this is slated for?

Not before 2024, probably. We've only just started to encourage using verity bundles more visibly now.

> @jluebbe Well, this is less of a problem as we use a squashfs (ie read-only) rootfs and have a central configuration registry that controls all the configurable elements in our appliance. Thanks for [...]

Ah, so you don't have any other state on the system that might change incompatibly?

mbessler, 2022-11-09:
>>> New features in manifests should normally be optional due to this. Do you have examples where mandatory options were added that old versions of RAUC could not handle?

>>  I'm pretty sure I ran across that once, but could not remember what options gave me trouble.

>Hmm, as above, we'd regard that as a bug and would like to fix it, so we'd be interested in specifics.

While not exactly the same, I'm currently pulling in an upgrade to meta-rauc (yes, different repo, thus my "not exactly the same") and it changes things out from under my feet by removing recipes-core/rauc/rauc_1.5.1.bb in favor of recipes-core/rauc/rauc_1.6.bb. This really bugs me. I run dunfell for its LTS-ish-ness. 
Thus I'm hoping that related layers (`meta-rauc` in this case) do not just remove versions that earlier dunfell versions of the layer included. I don't mind new versions getting added, but I do mind versions being removed. 

That leaves me two options, 1. fork `meta-rauc`, or 2. keeping `meta-rauc` at a specific commit and thus not get security related updates.

> Ah, so you don't have any other state on the system that might change incompatibly?

Correct. In addition to the configuration database and boot environment, I've only got specific directories under /var writable that are non-volatile, basically for logging.

ejoerns, 2022-11-09:
> Thus I'm hoping that related layers (`meta-rauc` in this case) do not just remove versions that earlier dunfell versions of the layer included. I don't mind new versions getting added, but I do mind versions being removed.

From my perspective, the main purpose of `meta-rauc` stable branches is to provide support for a recent RAUC version.
If you do not want to update RAUC, just stick to the version of meta-rauc you are currently using.

> That leaves me two options, 1. fork `meta-rauc`, or 2. keeping `meta-rauc` at a specific commit and thus not get security related updates.

What kind of security-related updates do you expect? The layer just provides some convenience for integrating RAUC into your project. I would say everything that is relevant in terms of security affects the `rauc` package itself, i.e. the RAUC source code. Thus even if you would stick to an old version's recipe, you would not get any security-related updates.

Thus what would be the motivation for forking meta-rauc? Do you want to maintain your own set of fix patches? Then how will you identify if RAUC bugfixes are relevant for you or not?

Even poky provides security updates only for CVEs. If there is any (new) CVE in RAUC, decisions about bugfix versions might be made then. But so far, as for all software, the best security is only provided by the recent release since most project's bugs are fixed without major attention or CVEs (since it is often not clear if a bugfix has some hidden security implications).

ejoerns, 2022-11-30:
FTR: #1018 has been written to clarify some of the above-noted points in the documentation

@mbessler Could we clear your concerns regarding bundle compatibility vs. LTS requirement?

mbessler, 2022-11-30:
@ejoerns Yes, I think that is a good step forward. 
I like the plan about the Yocto LTS compatibility.

ejoerns, 2023-01-27:
@mbessler Thanks for the feedback! I guess we can close this then.

closed 2023-01-27 by ejoerns