rauc/rauc #1008: Adaptive update fails without

landerweit-phytec, 2022-11-08:
I'm currently trying out the adaptive update feature with ext4 rootfs partitions.
The update works as expected when the ext4 partition in the bundle and the rootfs slot have the same size.
But when the size differs RAUC crashes with `g_propagate_error: assertion 'src != NULL' failed` instead of a meaningful error message. 
A check for a size mismatch could probably prevent this.

```
rauc install /tmp/phytec-headless-bundle-phyboard-segin-imx6ul-7.raucb
installing
 0% Installing
 0% Determining slot states
20% Determining slot states done.
20% Checking bundle
20% Verifying signature
40% Verifying signature done.
40% Checking bundle done.
40% Checking manifest contents
60% Checking manifest contents done.
60% Determining target install group
80% Determining target install group done.
80% Updating slots
80% Checking slot rootfs.0
83% Checking slot rootfs.0 done.
83% Copying image to rootfs.0
RAUC service disappeared
D-Bus error while installing `/tmp/phytec-headless-bundle-phyboard-segin-imx6ul-7.raucb`
```
Output of journalctl -u rauc
```
Apr 28 17:57:40 phyboard-segin-imx6ul-7 systemd[1]: Starting RAUC Update Service...
Apr 28 17:57:40 phyboard-segin-imx6ul-7 rauc[363]: Using central status file /mnt/config/rauc/central.raucs
Apr 28 17:57:40 phyboard-segin-imx6ul-7 rauc[363]: Using central status file /mnt/config/rauc/central.raucs
Apr 28 17:57:40 phyboard-segin-imx6ul-7 systemd[1]: Started RAUC Update Service.
Apr 28 17:57:40 phyboard-segin-imx6ul-7 rauc[363]: input bundle: /tmp/phytec-headless-bundle-phyboard-segin-imx6ul-7.raucb
Apr 28 17:57:40 phyboard-segin-imx6ul-7 rauc[363]: Active slot bootname: system1
Apr 28 17:57:40 phyboard-segin-imx6ul-7 rauc[363]: installing /tmp/phytec-headless-bundle-phyboard-segin-imx6ul-7.raucb: started
Apr 28 17:57:40 phyboard-segin-imx6ul-7 rauc[363]: installing /tmp/phytec-headless-bundle-phyboard-segin-imx6ul-7.raucb: Checking and mounting bundle...
Apr 28 17:57:40 phyboard-segin-imx6ul-7 rauc[363]: Reading bundle: /tmp/phytec-headless-bundle-phyboard-segin-imx6ul-7.raucb
Apr 28 17:57:40 phyboard-segin-imx6ul-7 rauc[363]: Verifying bundle signature... 
Apr 28 17:57:40 phyboard-segin-imx6ul-7 rauc[363]: Verified inline signature by 'O = SW, CN = PHYTEC Messtechnik GmbH Development-1'
Apr 28 17:57:40 phyboard-segin-imx6ul-7 rauc[363]: Mounting bundle '/tmp/phytec-headless-bundle-phyboard-segin-imx6ul-7.raucb' to '/run/rauc/bundle'
Apr 28 17:57:41 phyboard-segin-imx6ul-7 rauc[363]: Configured loop device '/dev/loop0' for 76886016 bytes
Apr 28 17:57:41 phyboard-segin-imx6ul-7 rauc[363]: Configured dm-verity device '/dev/dm-0'
Apr 28 17:57:41 phyboard-segin-imx6ul-7 rauc[363]: Starting pre install handler: /usr/lib/rauc/rauc-pre-install.sh
Apr 28 17:57:41 phyboard-segin-imx6ul-7 rauc[363]: Marking target slot rootfs.0 as non-bootable...
Apr 28 17:57:41 phyboard-segin-imx6ul-7 rauc[363]: installing /tmp/phytec-headless-bundle-phyboard-segin-imx6ul-7.raucb: Updating slots...
Apr 28 17:57:41 phyboard-segin-imx6ul-7 rauc[363]: Checking image type for slot type: ext4
Apr 28 17:57:41 phyboard-segin-imx6ul-7 rauc[363]: Image detected as type: *.ext4
Apr 28 17:57:41 phyboard-segin-imx6ul-7 rauc[363]: installing /tmp/phytec-headless-bundle-phyboard-segin-imx6ul-7.raucb: Checking slot rootfs.0
Apr 28 17:57:41 phyboard-segin-imx6ul-7 rauc[363]: installing /tmp/phytec-headless-bundle-phyboard-segin-imx6ul-7.raucb: Updating slot rootfs.0
Apr 28 17:57:41 phyboard-segin-imx6ul-7 rauc[363]: Updating /dev/mmcblk1p5 with /run/rauc/bundle/phytec-headless-image-phyboard-segin-imx6ul-7.ext4
Apr 28 17:58:01 phyboard-segin-imx6ul-7 rauc[363]: g_propagate_error: assertion 'src != NULL' failed
Apr 28 17:58:01 phyboard-segin-imx6ul-7 systemd[1]: rauc.service: Main process exited, code=killed, status=5/TRAP
Apr 28 17:58:01 phyboard-segin-imx6ul-7 systemd[1]: rauc.service: Failed with result 'signal'.```

jluebbe, 2022-11-08:
It should already work if the target slot is larger. Which exact sizes do your image and slots have?

landerweit-phytec, 2022-11-08:
The target slot is 205225984 bytes large and the partition in the bundle is 205565952 bytes in size.
I have tried the larger target with a smaller update which does indeed work.

jluebbe, 2022-11-08:
OK. So the issue here is that we should abort the installation earlier with a useful error message.

PR #1037 (2023-01-12): Check image size before starting installation
In some cases, we can abort early if the image in the bundle is too large for the slot. This is not possible when installing tar archives or when writing images to slot which point to a file.

Fixes: #1008
closed 2023-01-12 by ejoerns