rauc/rauc #1009: Image type of "boot-emmc" limited to "*.img" files

mschwan-phytec, 2022-11-15:
Currently, updating the bootloader in an eMMC boot partition is only allowed for files of type `.img`:
```c
RaucUpdatePair updatepairs[] = {
...
#if ENABLE_EMMC_BOOT_SUPPORT == 1
	{"*.img", "boot-emmc", img_to_boot_emmc_handler},
#endif
	{"*", "boot-emmc", NULL},
...
```
Is there a specific reason to limit it to that file type? Usual extensions for bootloaders also include `.bin` or even nothing at all. In general, I would assume the bootloader is almost always a binary file which needs to written as is to the eMMC boot partition.

ejoerns, 2022-11-15:
The idea was to be just as explicit as we are for the other slot types, too.

So the question is if we can be more relaxed for slot types that already clearly dictate the handling, but I understand your point of course.

This would drop some sanity checking of input types, but, to be honest, when e.g. using meta-rauc, the bundle class would also just take the given image and unconditionally append `*.img` to the filename if the slot type is set to `bootloader`.

@jluebbe Do you have any opinion on this?

jluebbe, 2022-11-25:
The selection of the update handler based on the file type is intended as a way to allow different handlers in the future, so just allowing any extension here would mean that the `boot-emmc` slot type can never gain a different handler.

I wouldn't like to have that limitation just to avoid the small inconvenience of renaming files while copying them into the bundle directory during creation.

mb-karo, 2023-07-20:
I also ran into this issue, and I also was unable to put any *.bin* file into a raw partition.

So I made this patch, and re-compiled RAUC and now it works as I need it. Any comments to this?
Maybe a bit "too" generic?

```
diff --git a/src/update_handler.c b/src/update_handler.c
index f800cf2..c783b0f 100644
--- a/src/update_handler.c
+++ b/src/update_handler.c
@@ -2778,7 +2778,7 @@ RaucUpdatePair updatepairs[] = {
 #if ENABLE_EMMC_BOOT_SUPPORT == 1
        {"*.img", "boot-emmc", img_to_boot_emmc_handler},
 #endif
-       {"*", "boot-emmc", NULL},
+       {"*", "boot-emmc", img_to_boot_emmc_handler},
        {"*.vfat", "boot-mbr-switch", img_to_boot_mbr_switch_handler},
        {"*.img", "boot-mbr-switch", img_to_boot_mbr_switch_handler},
        {"*", "boot-mbr-switch", NULL},
@@ -2792,6 +2792,7 @@ RaucUpdatePair updatepairs[] = {
        {"*", "boot-raw-fallback", NULL},
        {"*.img.caibx", "*", img_to_raw_handler}, /* fallback */
        {"*.img", "*", img_to_raw_handler}, /* fallback */
+       {"*.bin", "*", img_to_raw_handler}, /* fallback */
        {0}
 };
```

mschwan-phytec, 2023-07-20:
Handling `*.bin` files this way seems to be a good start. :+1:

However, allowing *any* type of file to be flashed using `img_to_boot_emmc_handler()` does not seem to be a good idea to me, as it breaks the handler selection mechanism, as described by @jluebbe.

jluebbe, 2023-07-20:
As mentioned in https://github.com/rauc/rauc/issues/1009#issuecomment-1327260133, we don't want to add more broad matches here. The easy fix is to name rename the images to `.img` in your build system.

mb-karo, 2023-07-20:
Ahh, I misunderstood that.
Thanks.

closed 2023-07-20 by jluebbe