rauc/rauc #1013: Problem with bundle updating boot and root partitions

Silcet, 2022-11-22:
Hi, I'm using rauc in a project and I'm facing an issue in version 1.8 that I managed to avoid in version 1.5.

My setup consists of 2 symmetric boot partitions and a GPT boot partition containing GRUB. When I install a bundle containing both the boot and the root partitions the system can never change from partition A to partition B. This happens because the grubenv variables get updated in the currently mounted boot partition and not in the newly updated boot partition. To prevent this I had a workaround in version 1.5 that would in a post-install hook of the esp partition unmount it, call partprobe to inform the kernel of the partition change and mount it again so when the root partition got updated and the grubenv changed accordingly, the correct partition would be updated.

However, in version 1.8 the installer changes the GPT partition after running the post-install hook, which brakes my old flow. An alternative would be to do those same steps in the pre or post install of the rootfs, but I'm finding that slightly more hacky.

Do you have any good alternative for this installation flow or am I already the closest possible to a solution?

jluebbe, 2022-11-22:
Why do you have symmetric boot partitions? I assume that you want to keep the kernel for each rootfs in the corresponding boot partition.

The clean approach is to keep the bootloader (the ESP with GRUB, in your case, with its state) separate from the redundant partitions. This is more natural, as the bootloader (which contains the A/B selection logic) cannot be part of the redundant fallback itself. The best you can normally do is *atomic* bootloader updates (https://rauc.readthedocs.io/en/latest/advanced.html#update-boot-partition-in-gpt).

With that setup, you'd have one slot for the bootloader, and two slots each for kernel (boot partition) and root FS. The bootloader's state should also be placed in a separate partition (which is never updated), so it's not overwritten on bootloader updates.

Silcet, 2022-11-22:
Thanks for your fast reply jluebbe.

I just realized I had a mistake in my text before. I meant 2 symmetric root partitions, not boot partitions.

I am already using atomic bootloader updates, but I never considered keeping the bootloader state in a different partition that is not updated. I'll try and see whether I can make that work on my setup since it sounds like the right solution for it.

If you don't have more suggestions than that I'd be very happy to close the issue :)

ejoerns, 2022-11-22:
@Silcet You might want to have a look at
https://github.com/rauc/meta-rauc-community/tree/master/meta-rauc-qemux86
for reference.

Silcet, 2022-11-22:
Thanks a lot for the help and the references. I believe I can work it out from here.

closed 2022-11-22 by Silcet