rauc/rauc #1014: Allow to create a bundle from a block device

dcd-arnold, 2022-11-23:
Hi Guys,

thank you for RAUC. I have a feature request.
I have provisioning tools in place to create a device as specified (master copy). After that I am using RAUC (and Hawkbit) to replicate this setup to other devices.

To obtain a bundle form my master copy I `dd` the root disk to an image. Build a RAUC bundle from the image and delete the image again. It would be nice if I would not have to create a block copy of the disk just to delete it after the next step. This costs time and places extra requirements on the media I build the image on, as it has to accommodate the block copy.

I would like to be able to pass a block device to RAUC directly instead of the image.

Thanks.

ejoerns, 2022-11-23:
Thanks for your requests.

Golden images are not state of the art in any way and will thus not be supported.

Please use a proper way to create your device image(s) offline. We recommend using a Linux build system as Yocto, Buildroot, or PTXdist, but there are several other valid approaches, too. Replicating an arbitrary disk state on a target device is none.

dcd-arnold, 2022-11-23:
I see. Thank you for clarification.

closed 2022-11-23 by dcd-arnold