rauc/rauc #1024: qemu-system-x86_64: -nic: invalid option

samannazz, 2022-12-20:
I have follow the all step and after running the sudo ./qemu-test. i am getting this error

qemu-system-x86_64: -nic: invalid option

[image]

where i have done wrong.

ejoerns, 2022-12-20:
On which host system do you test this?

The `-nic` option should be provided by your qemu implementation.

Also, note that the purpose of the qemu setup is to allow running test as non-root.

samannazz, 2022-12-20:
i am running ubuntu 18.04
and this is the some part of my qemu-test file

[image]

ejoerns, 2022-12-20:
Ubuntu 18.04 comes with QEmu v2.11

https://packages.ubuntu.com/bionic/qemu-system-x86

Please consider using a recent distro.

ejoerns, 2022-12-21:
For others coming accross this issue:

Alternatives are to use *Ubuntu* LTS 20.04 or 22.04, to use *Debian* buster or bullseye, or to use a container of these or, alternatively, the RAUC testing container http://ghcr.io/rauc/rauc/rauc-ci:latest

closed 2022-12-21 by ejoerns