rauc/rauc #1028: rauc extract is not extracting rauc bundle.

samannazz, 2022-12-27:
Creating a Bundle (Host):

step 1: `mkdir content-dir/`

step 2: adding a custom rootfs.ext4  → {rauc/content-dir}, because `cp $SOURCE/rootfs.ext4 content-dir/` is not working.
 

step 3: creating manifest.raucm

```
cat >> content-dir/manifest.raucm << EOF
[update]
compatible=FooCorp Super BarBazzer
version=2019.01-1
[image.rootfs]
filename=rootfs.ext4
EOF
```

step 4: creating key and cert. {/rauc}

`openssl req -x509 -newkey rsa:4096 -nodes -keyout demo.key.pem -out demo.cert.pem -subj "/O=rauc Inc./CN=rauc-demo"
`
[image]

step 5: creating a bundle from this: {/rauc}

`rauc --cert demo.cert.pem --key demo.key.pem bundle content-dir/ update-2019.01-1.raucb
`
output:
[image]

The output of the command indicates that the RAUC bundle is being created in the "plain" format, which is a simple format that includes all of the files and directories in the bundle without any additional metadata or compression.

`rauc-Message: 13:59:54.136: No keyring given, skipping signature verification` indicates that the RAUC was unable to find a keyring to use for verifying the signature of the RAUC bundle.

How can i create keyring?

Step 6: now i am extracting ruac bundle so that i can launch virtual machine i.e. qemu.

[image]

How can I solve this problem? Maybe it is because of the custom rootfs.ext4, or maybe I haven't created a keyring?

samannazz, 2022-12-28:
this error resloved by using this:
`sudo rauc extract -d update-2019.01.1.raucb --keyring deo.cert.pem /tmp/bundle`

ejoerns, 2023-01-02:
@samannazz The error message

> FATAL ERROR:dir_scan: failed to make directory output, because File exists

comes from the `unsquashfs` tool called by RAUC which does not allow extracting to existing directories by default.

I guess we could better handle this by pre-checking if the output path exists and emitting a more helpful error message then.

Note that using `sudo` should not be required for extracting (unless you intend to extract to a root-owned location, which I would not suggest).

PR #1123 (2023-05-11): src/bundle: ensure that the output directory does not exist before extracting
Otherwise, unsquashfs will just fail with a possibly confusing error message.

Fixes: #1028
commit b90197d7 (2023-05-11): src/bundle: ensure that the output directory does not exist before extracting

Otherwise, unsquashfs will just fail with a possibly confusing error message.

Fixes: #1028
closed 2023-05-11 by jluebbe