rauc/rauc #1045: Error during bundle installation with a manifest containing [meta.]

hatmaalla, 2023-01-13:
I'm using the master branch of rauc in my project
The idea is to add some projet specific values in the section [meta.] 
I followed the public documentation and the example in https://github.com/rauc/rauc/blob/2311e24a30480d54ae3322987df144fa17b685eb/test/manifest.c#L269 to test this feature

````
[meta.mycustomsection]
release-type=beta
````
During bundle installation the following error is raised

````
installing
  0% Installing
  0% Determining slot states
 20% Determining slot states done.
 20% Checking bundle
 20% Verifying signature
 40% Verifying signature done.
 40% Checking bundle failed.
100% Installing failed.
LastError: Failed to load manifest: Invalid group '[meta.mycustomsection]'
````

Is there a problem in my manifest file?

ejoerns, 2023-01-13:
Is the RAUC you are using for installation also from master?

hatmaalla, 2023-01-13:
Yes, rauc is the same as the one used during the installation.
However, a `systemctl daemon-reload` is needed after reinstalling of the new version of rauc to solve the problem.
Thanks for the quick feedback

closed 2023-01-13 by hatmaalla