rauc/rauc #1048: Signature verification fails with self-signed certificate

WillMuench, 2023-01-24:
**Describe the bug**
All, I'm at the end of my wits here. I'm new to rauc but can basically not get past a simple "hello world (nano)" update install. After working through various tutorials and issues, I enden up at  **_LastError: signature verification failed: Verify error: self-signed certificate_**.

I have created simple certificates using
`openssl req -x509 -newkey rsa:4096 -keyout key.pem -out cert.pem -days 365 -nodes`
and used them for bundle creation. The cert is also available in `/etc/rauc/` and configured in `system.conf` accordingly.

`rauc info <bundle>` shows the bundle information.

Looking at the certificate part of the info, it should be valid:
   Not Before: Jan 24 08:30:28 2023 GMT
   Not After:  Jan 24 08:30:28 2024 GMT

System time of the target is within the range 
  `date`
  Tue Jan 24 12:07:37 UTC 2023

I'm not entirely sure if that's a bug. But I have not idea what else I could d, to get past that hurdle.

Thanks a lot!

**Background information**
Describe which RAUC version and build system you are using.
* rauc 1.8
* build with yocto / kirkstone for raspberry pi4 (arm 64)

**To Reproduce**
Steps to reproduce the behavior:
* bitbake update-bundle
* copy .raucb to target system
* perform rauc install <bundle>

**Expected behavior**
The signature verification to pass.

**Additional context**

**Content of system.conf**
```
[system]
compatible=RaspberryPi4
bootloader=uboot
boot-attempts=3
data-directory=/data/

[keyring]
path=/etc/rauc/cert.pem
allow-partial-chain=true
 
[slot.rootfs.0]
device=/dev/mmcblk0p2
type=ext4
bootname=A

[slot.rootfs.1]
device=/dev/mmcblk0p3
type=ext4
```

**Output of rauc info**
```
rauc-Message: 12:06:51.505: Using central status file /data/central.raucs
rauc-Message: 12:06:51.506: valid /etc/rauc/system.conf found, using it
rauc-Message: 12:06:51.507: Reading bundle: /home/sb/sb-update-bundle-raspberrypi4-64-20230124083552.raucb
rauc-Message: 12:06:51.519: Verifying bundle signature... 
rauc-Message: 12:06:52.559: Verified detached signature by 'C = DE, ST = Schleswig-Holstein, L = Luebeck, O = SB GmbH, OU = Hub OS Development, CN = sb.com, emailAddress = info@sb.com'
Compatible: 	'RaspberryPi4'
Version:    	'v20200703'
Description:	'SB Nano add'
Build:      	'20230124083552'
Hooks:      	''
Bundle Format: 	plain

1 Image:
  [rootfs]
	Filename:  sb-image-base-raspberrypi4-64.ext4
	Checksum:  3ee5db3026ec8d332c7f35fcfa618525b84e0a1c5cf66ee85ba65b680a20e606
	Size:      197132288
	Hooks:     

Certificate Chain:
 0 Subject: C = DE, ST = Schleswig-Holstein, L = Luebeck, O = SB GmbH, OU = Hub OS Development, CN = sb.com, emailAddress = info@sb.com
   Issuer: C = DE, ST = Schleswig-Holstein, L = Luebeck, O = SB GmbH, OU = Hub OS Development, CN = sb.com, emailAddress = info@sb.com
   SPKI sha256: 46:5B:42:60:A4:38:99:5B:A1:B7:CA:9A:C0:AD:6B:99:2E:9D:B9:87:B3:B7:4F:B2:B9:4A:C3:DF:8C:6C:E7:78
   Not Before: Jan 24 08:30:28 2023 GMT
   Not After:  Jan 24 08:30:28 2024 GMT
```

ejoerns, 2023-01-25:
@WillMuench I am a bit confused that your 'rauc info' works (without providing `--keyring` arg manually), but `rauc install` doesn't, since both use the same config.

The validity should also not be relevant since the error message does not point into that direction.

Have you changed the keyring file after the rauc background service has started? 

Normally, using your setup and key.pem + cert.pem as signing keypair and cert.pem as verification keyring should work.

WillMuench, 2023-01-29:
@ejoerns thanks for your consideration! I'm still not sure what caused this but since you were puzzled as well, I did a full clean and rebuild and the issue is gone. I suggest to close the issue.

closed 2023-01-29 by WillMuench

rvdgracht, 2023-07-25:
Maybe this helps,..
I've seen this error when the keychain contains a self-signed certificate of a different PKI than the one used to sign the bundle. I.e. when you have a release and a development keychain (both self-signed), of which the release keychain is active, while the bundle is signed with the development key. In that case verification fails with this (misleading) error message.