rauc/rauc #1057: Error getting loop backing_file for /dev/loop0p1

paletteguy, 2023-02-08:
**Describe the bug**
If a loop0 device is already used for other purpose when Rauc installer  is unable mount image from rauc file.

**Background information**
RAUC version 1.8 - Yocto kirkstone - IMX8-MP

**To Reproduce**
Steps to reproduce the behavior:
1. Mount a loop0 devices
2. run rauc install image.raucb

**Expected behavior**
* Installation completed and RAUC used another loop-back device for mounting.

**Logs**
Feb 08 13:39:45 CCB18177 rauc[357]: input bundle: /tmp/tmp.aDGIesmks4/image-update-99.47.0.84.raucb
Feb 08 13:39:45 CCB18177 rauc[357]: Active slot bootname: system1
Feb 08 13:39:45 CCB18177 rauc[357]: installing /tmp/tmp.aDGIesmks4/image-update-99.47.0.84.raucb: started
Feb 08 13:39:45 CCB18177 updated[1367]: installing
Feb 08 13:39:45 CCB18177 rauc[357]: Error getting loop backing_file for /dev/loop0p1: Failed to open file “/sys/block/loop0p1/loop/backing_file”: No such file or directory
Feb 08 13:39:45 CCB18177 rauc[357]: installing /tmp/tmp.aDGIesmks4/image-update-99.47.0.84.raucb: Error getting loop backing_file for /dev/loop0p1: Failed to open file “/sys/block/loop0p1/loop/backing_file”: No such file or directory
Feb 08 13:39:45 CCB18177 rauc[357]: installing /tmp/tmp.aDGIesmks4/image-update-99.47.0.84.raucb: finished
Feb 08 13:39:45 CCB18177 rauc[357]: installing `/tmp/tmp.aDGIesmks4/image-update-99.47.0.84.raucb` failed: 1
Feb 08 13:39:45 CCB18177 updated[1367]: 0% Installing
Feb 08 13:39:45 CCB18177 updated[1367]: 0% Determining slot states
Feb 08 13:39:45 CCB18177 updated[1367]: 20% Determining slot states failed.
Feb 08 13:39:45 CCB18177 updated[1367]: 100% Installing failed.

**Additional context**
Reproducible 100 %

jluebbe, 2023-02-08:
Could you add the output of `losetup -l -a` and `cat /proc/mounts`?

It seems that RAUC gets confused while checking mounted filesystems for a partitioned loop device. The backing file is probably only set for the main loop device.

paletteguy, 2023-02-08:
cat /proc/mounts

/dev/mmcblk0p4 /service ext4 rw,relatime 0 0
tmpfs /run/user/1002 tmpfs rw,nosuid,nodev,relatime,size=174092k,nr_inodes=43523,mode=700,uid=1002,gid=1003 0 0
tmpfs /run/user/0 tmpfs rw,nosuid,nodev,relatime,size=174092k,nr_inodes=43523,mode=700 0 0
/dev/loop0p1 /run/media/root/SimPad vfat rw,nosuid,nodev,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,showexec,utf8,flush,errors=remount-ro 0 0

paletteguy, 2023-02-08:
root@CCB18176:~# losetup -l -a
NAME       SIZELIMIT OFFSET AUTOCLEAR RO BACK-FILE DIO LOG-SEC
/dev/loop0         0      0         0  0 /data.img   0     512

paletteguy, 2023-02-14:
Hi,

Nice news, will the Release V1.9 be supported on Yocto Kirkstone Release ?

Br,

Kasten

jluebbe, 2023-03-02:
This is the same underlying issue as https://github.com/rauc/rauc/issues/631.

PR #1082 (2023-03-02): src/install: support resolving partitioned loop devices
This allows obtaining the correct backing file for partitioned loop devices such as /dev/loop0p1 by using a regex match that cuts out the base loopN name.

Fixes: #631
Fixes: #1057
Closes: #1077

Reported-by: Uwe Kleine-König <u.kleine-koenig@pengutronix.de>
Reported-by: Karsten Sperling Opdal <karsten.sperling.opdal@laerdal.com>
commit 2000f91f (2023-03-02): src/install: support resolving partitioned loop devices

This allows obtaining the correct backing file for partitioned loop
devices such as /dev/loop0p1 by using a regex match that cuts out the
base loopN name.

Fixes: #631
Fixes: #1057
Closes: #1077

Reported-by: Uwe Kleine-König <u.kleine-koenig@pengutronix.de>
Reported-by: Karsten Sperling Opdal <karsten.sperling.opdal@laerdal.com>
closed 2023-03-02 by jluebbe