rauc/rauc #1090: "failed to recv nbd config reply header" when installing update from HTTP server

fredgreer, 2023-03-09:
Hello,

I am seeing the below error when attempting to install an adaptive update over HTTP:

```
> rauc install http://10.80.82.20:3000/update.raucb

installing
  0% Installing
  0% Determining slot states
 20% Determining slot states done.
 20% Checking bundle
RAUC service disappeared
D-Bus error while installing `http://10.80.82.20:3000/update.raucb

> journalctl -u rauc

Mar 09 03:19:22 rauc[195]: input bundle: http://10.80.82.20:3000/update.raucb
Mar 09 03:19:22 rauc[195]: Active slot bootname: B
Mar 09 03:19:22 rauc[195]: installing http://10.80.82.20:3000/update.raucb: started
Mar 09 03:19:22 rauc[195]: installing http://10.80.82.20:3000/update.raucb: Checking and mounting bundle...
Mar 09 03:19:22 rauc[195]: Remote URI detected, streaming bundle...
Mar 09 03:19:22 rauc[195]: sending: {'url': <'http://10.80.82.20:3000/update.raucb'>}
Mar 09 03:19:22 rauc-nbd[279]: running as UID 65534, GID 65534
Mar 09 03:19:22 rauc-nbd[279]: received: {'url': <'http://10.80.82.20:3000/update.raucb'>}
Mar 09 03:19:22 rauc-nbd[279]: configuring for URL: http://10.80.82.20:3000/update.raucb
Mar 09 03:19:22 rauc-nbd[279]: unexpected error from curl_easy_setopt in prepare_curl
Mar 09 03:28:58 rauc[306]: failed to recv nbd config reply header
Mar 09 03:29:00 systemd[1]: rauc.service: Main process exited, code=killed, status=5/TRAP
Mar 09 03:29:00 systemd[1]: rauc.service: Failed with result 'signal'.
```

**Environment**
- NXP iMX6 development board
- Buildroot distribution
- Kernel version 5.10.9
- RAUC version 1.8 on host and target, both installed / built through Buildroot

**Attempted troubleshooting**

Downloading the same file to the target with `curl http://10.80.82.20:3000/update.raucb -o update.raucb` and installing with `rauc install update.raucb` works as expected.

Following the HTTP streaming [guide](https://rauc.readthedocs.io/en/v1.8/advanced.html#http-streaming) I have confirmed:

- RAUC is being built with streaming support. The Buildroot config contains:
```
BR2_PACKAGE_RAUC=y
BR2_PACKAGE_RAUC_DBUS=y
BR2_PACKAGE_RAUC_GPT=y
BR2_PACKAGE_RAUC_NETWORK=y
BR2_PACKAGE_RAUC_STREAMING=y
BR2_PACKAGE_RAUC_JSON=y
```

- The bundle is being built using the verity format:
```
> rauc info update.raucb
...
Bundle Format: 	verity
...
```

- The server (npm [serve](https://www.npmjs.com/package/serve)) supports HTTP range requests:
```
> curl -I http://10.80.82.20:3000/update.raucb

HTTP/1.1 200 OK
Content-Length: 159910422
Content-Disposition: inline; filename="update.raucb"
Accept-Ranges: bytes
ETag: "691e06d21e6609bea7b02ddde36928760efb7bad"
Date: Thu, 09 Mar 2023 04:02:26 GMT
Connection: keep-alive
Keep-Alive: timeout=5
```

Additionally I have tried serving the file from AWS S3 and through a public URL (ngrok tunnel to the local server) with the same error.

- The kernel was built with NBD and the other required packages. Confirmed `/proc/config.gz` includes:
```
CONFIG_SQUASHFS=y
CONFIG_SQUASHFS_FILE_CACHE=y
CONFIG_SQUASHFS_DECOMP_SINGLE=y
CONFIG_SQUASHFS_ZLIB=y
CONFIG_SQUASHFS_FRAGMENT_CACHE_SIZE=3
CONFIG_CRYPTO_SHA256=y
CONFIG_MD=y
CONFIG_BLK_DEV_LOOP=y
CONFIG_BLK_DEV_DM=y
CONFIG_DM_VERITY=y
CONFIG_BLK_DEV_NBD=y
```

I am out of ideas on what else to investigate. I'd appreciate any suggestions on how to proceed.

Thanks!

ejoerns, 2023-03-09:
It may be the case that RAUC tries to configure unavailable options in libcurl.

Could you dump your configuration of libcurl? (`BR2_PACKAGE_LIBCURL_*` config items)

If `BR2_PACKAGE_LIBCURL_PROXY_SUPPORT` is disabled it could be that you run into #1039 which was fixed with v1.9.

fredgreer, 2023-03-09:
Thanks Enrico!

Sounds like that's likely the culprit. The config lines are:

```
BR2_PACKAGE_LIBCURL_CURL=y
# BR2_PACKAGE_LIBCURL_PROXY_SUPPORT is not set
# BR2_PACKAGE_LIBCURL_EXTRA_PROTOCOLS_FEATURES is not set
```

I'll try with `BR2_PACKAGE_LIBCURL_PROXY_SUPPORT` enabled tomorrow when I have the hardware at hand.

fredgreer, 2023-03-09:
@ejoerns I've confirmed this was the issue, thanks again for your help!

For anyone running into the same issue, the temporary fix (until RAUC 1.9 is the default in Buildroot) was to enable proxy support in libcurl in my Buildroot defconfig file. It now looks like:

```
BR2_PACKAGE_LIBCURL_CURL=y
BR2_PACKAGE_LIBCURL_PROXY_SUPPORT=y
# BR2_PACKAGE_LIBCURL_EXTRA_PROTOCOLS_FEATURES is not set
```

ejoerns, 2023-03-09:
@fredgreer Thanks for confirming and documenting!

closed 2023-03-09 by ejoerns