rauc/rauc #1114: Concept: Simple Update Server Support (and Life Cycle Handling)

ejoerns, 2023-04-28:
**Problem**

So far, the options for using RAUC in network-based OTA update scenarios are:

* set up a **hawkBit** deployment server and use it with e.g. `rauc-hawkbit-updater` as a client
* use a **simple HTTP web server**, place a bundle at a well-known location, and let RAUC download it
* have a **custom (proprietary) deployment server** with a custom client (perhaps integrated with the main application)

Typical features that can be required by an OTA update server are:

* Rollout scheduling to avoid updating all devices at once
* Deploy updates only to specific devices
* Monitoring of devices and basic device properties
* Notification after successful update + reboot (and collection of errors)

Excluding the custom server approach, the other two options are on opposite ends of the spectrum:

* *hawkBit* provides a rich feature set but is complex to set up and manage.
* A simple HTTP web server is easy to set up and manage but has hardly any capabilities for controlling update rollouts.

Now, since RAUC can directly download/stream bundles from an HTTP server, the idea is to equip RAUC with capabilities that support the above-mentioned features with a simple HTTP server (and optionally with little custom application logic).

Some of the required RAUC-side capabilities are currently missing for all scenarios. The most important one is the lack of update life-cycle state handling that would allow RAUC to more reliably detect and report a successful reboot of a just-updated system.

**Possible Solution**

The following generic features can be implemented in RAUC to support the usage of simple update servers.

* RAUC should optionally send some HTTP headers to the server. In the simplest case, this information can be logged by the server and introspected by the rollout operator. In more complex setups, the server can add some custom application logic and use the headers for filtering etc.

* The information reported to the server should be configurable and can be either standard information that is already available in RAUC or additional information retrieved by the `system-info` handler.

* System state information like a *boot id*, an *update transaction id*, and a general update cycle state should be generated and stored to allow RAUC to conclude if the system actually rebooted, if it rebooted to the expected system, etc.

* Automated polling for updates at a defined location (URL) can be implemented as part of the RAUC service.

* Simple rollout 'scheduling' can be realized by configuring an update time range for the devices. Each individual target chooses its point in time (in a reproducible manner) where it actually performs the download and update. This ensures not all platforms update at once to limit server load and allows operators to stop rollouts.

ejoerns, 2023-08-29:
Note that a notable number of features from the list above already made their way into RAUC.

With #1152 we can now actually configure RAUC to expose information like *boot id*, *transaction id* on the first HTTP request. See https://rauc.readthedocs.io/en/latest/advanced.html#additional-http-header-information.

More advanced features like polling or rollout scheduling are still to be done.

jluebbe, 2024-09-05:
In #1460, I promised @angeloc to write down our ideas for a simple client integrated into RAUC itself, so here it goes...

In the `system.conf`, we need to support configuration for polling. Perhaps:
```ini
[poll]
source=https://example.com/update.raucb
interval-sec=3600
max-interval-sec=86400
reboot-cmd=systemctl reboot
```

Also, we need to provide RAUC with a version that can be compared to the bundle version, which could be set by the `system-info` handler. Perhaps we can also derive a default from the status information of the booted slot, which might be enough for simple cases.

The RAUC service (in `src/service.c`) needs be extended by a polling timer, which:
* sets the service status to busy (to avoid concurrent install attempts)
* fetches, authenticates and parses the manifest from the source URL
* goes back to sleep with exponential back-off if fetching the bundle fails
* goes back to sleep if the system and bundle version are identical
* installs the bundle

The reboot could be done via a `post-install` handler, or a dedicated handler/config option could be added.

I think this could already work for simple scenarios. But especially with regards to error handling, it likely needs some extensions or server-side logic.

For cases where bundles don't update the rootfs (e.g. when using artifact updates #1301), the system version would be needed to be set via the `system-info` handler or we'd need to store the most recently installed bundle in the data directory. Otherwise it would just keep installing the same bundle over and over again...

Possible improvements:
* Announce the availability of a new version via D-Bus (manifest + manifest hash, perhaps via a property). The user can then confirm/trigger the installation of a specific bundle by passing this hash value back to RAUC. Using the manifest hash here protects against an attacker replacing the bundle while RAUC waits for the user's confirmation.
* Allow defining a time window in the `system.conf` in which to poll the server and attempt updates.
* Allow defining a "lock file" to temporarily disable update installation (e.g. during critical application-level actions)
* Add the current system version to the initial HTTP request as a header.
* Perhaps add an initial delay to avoid very quickly retrying installation on errors.
* Remember modification time and/or etag when polling the manifest, so that the server can respond with `304 Not Modified` if the bundle is unchanged. This would avoid the need to fetch the manifest every time.
* Add a maximum age to the manifest, so that the client can inform the user if a replay attach (using an old bunlde) may be happening.
* Add a maximum age limit to the `system.conf`, so that an attacker cannot use bundles with a higher maximum age.
* Add simple roll-out handling by giving each target a random (or fixed via config) value in the range 0.0-1.0 and defining a start time and duration in the manifest. The target can then calculate if it should install the update ((current time - start time)/duration < configured value).
* Add support for configuration snippets (e.g. `system.conf.d/*.conf`) which could be used to define different channels.

ashlin4010, 2024-11-29:
I mention all this just to share our use cause of Rauc and not to dismiss any of the above ideas.

I work with a fleet of devices that are extremely bandwidth-limited because of unreliable connections (it might take us a week or more to download an update). The existing HTTP download functionality does not have sufficient robustness/retry logic (at least last I tried), to make matters worse we don't have the cellular modem powered at all times.

For us, an ideal architecture would be an SDK/lib to build a Rauc D-Bus client and server with a production reference implementation.
This would allow the following:
- Custom transport layer (For example COAP)
- Fixed version of Rauc (We could qualify one version of Rauc for production use)
- Extensions / Modifications without needing to alter the Rauc code
- External control of the update life-cycle (only really want to trigger bundle install)
- Serialize all response data to save data
- Write it in a memory-safe language (rauc + client is critical software, it should be as safe as possible, especially if it's downloading and installing OS images)

We will not be able to utilize most of the suggested network-based OTA options and will build without it if possible.

Thanks

jluebbe, 2025-03-24:
PR #1672 implements significant parts of https://github.com/rauc/rauc/issues/1114#issuecomment-2332240262, while other are still only ideas.

My rough idea how a server-less rollout schedule could be configured in the manifest looks like this:
```ini
[rollout]
base=2025-04-01
#base=file-time
#base=signing-time
limit@1d=step 0.1
limit@2d=linear 0.25
limit@7d=cubic 1.0
limit@14d=linear 3.0
```
This would allow different phases, each with different slopes. The base time could be fixed in the bundle or use external information (e.g. modified time from the HTTP server/filesystem). The limits basically define a monotonic function which maps the (current) time to a numeric value.

To configure when each device should update, you'd have in the `system.conf`:
```ini
[polling]
rollout-phase=1
#rollout-point=0.2 # or bias? or offset?
#rollout-group=<seed value>
```
If the `rollout-point` (I'm not happy with the name yet) is not set, it will be derived (by hashing) from the machine ID. It's value between 0 and 1. If `rollout-group` is given, that value would be used as the seed (to update closely related devices roughly at the same time).

A given device would consider the update valid if the calculated limit value is higher `phase+point`.