rauc/rauc #1124: Certificate Key Usage Attributes - S/MIME signing is needed for bundle creation

Drotak, 2023-05-11:
**Describe the bug**
I use smallstep as CA and for creating my certificates. Like mentioned in the [documentation](https://rauc.readthedocs.io/en/stable/advanced.html#certificate-key-usage-attributes) I created my leaf certificate with the needed X509 keyUsage and extendedKeyUsage parameters. (keyUsage = "digitalSignature" and extendetKeyUsage = "codeSigning").
The bundle creation worked fine with the demo certificate mentioned in the documentation, but when I tried to use my self signed certificate it failed with the error message:
```
405A1CF6FB7F0000:error:17000064:CMS routines:cms_signerinfo_verify_cert:certificate verify error:../crypto/cms/cms_smime.c:289:Verify error: unsupported certificate purpose
Failed to resign bundle: failed to sign manifest: signature verification failed: error:1700006B:CMS routines::content type not enveloped data
```

**Background information**
RAUC version: rauc 1.9.96-973ae
System:
I'm running a ubuntu server 22.04.2 LTS version in a qemu virtualization 
```
root@secc04:~# hostnamectl
 Static hostname: my_comp
       Icon name: computer-vm
         Chassis: vm
      Machine ID: bae476e4d484402eae8536cd64b293ec
         Boot ID: cc401e676ee44947bced60480c38d611
  Virtualization: kvm
Operating System: Ubuntu 22.04.2 LTS              
          Kernel: Linux 5.19.0-41-generic
    Architecture: x86-64
 Hardware Vendor: QEMU
  Hardware Model: Standard PC _Q35 + ICH9, 2009_
root@secc04:~#
```

**To Reproduce**
Steps to reproduce the behavior:
1. Create leaf certificate 'my_leaf.pem' with smallstep (step ca certificate). Chain: "root -> intermediate -> leaf"
1.1 My provisioner config was updated with this template:
```
{
  "subject": {{ toJson .Subject }},
  "sans": {{ toJson .SANs }},
  "keyUsage": ["digitalSignature"],
  "extKeyUsage": ["codeSigning"],
  "basicConstraints": {
    "isCA": false
  }
}
``` 
2. Create keyring file `cat root.crt intermediate.crt > keyring.pem`
3. execute `rauc --cert=/data/update/certs/my_leaf.pem --key=/data/update/certs/my_leaf.key --keyring=/data/update/certs/keyring.pem bundle rauc-bundle/ update-2023.05-1.raucb`

**Expected behavior**
Bundle creation without error

**Logs**
```
402AEF40DF7F0000:error:17000064:CMS routines:cms_signerinfo_verify_cert:certificate verify error:../crypto/cms/cms_smime.c:289:Verify error: unsupported certificate purpose
Failed to resign bundle: failed to sign manifest: signature verification failed: error:1700006B:CMS routines::content type not enveloped data
```

**Additional context**
After a view hours of trying to find the problem and even changing my intermediate certificate to include the extendetKeyUsage = "codeSigning" parameter without luck, I found the problem!
The leaf certificate also needed the S/MIME signing parameter!
With the updated template, everything worked as expected:
```
{
  "subject": {{ toJson .Subject }},
  "sans": {{ toJson .SANs }},
  "keyUsage": ["digitalSignature"],
  "extKeyUsage": ["codeSigning", "emailProtection"],
  "basicConstraints": {
    "isCA": false
  }
}
```

Before including the "emailProtection" in the extKeyUsage Field, I also tried to include the "[check-purpose=codesign](https://rauc.readthedocs.io/en/stable/reference.html#check-purpose)" inside the config file, but it didn't change anything.

My config file now:
```
[system]
compatible=MyOwnDefinedName
bootloader=grub
bundle-formats=-plain
data-directory=/data/rauc-status
grubenv=/mnt/boot/grub/grubenv

[keyring]
path=/data/update/certs/keyring.pem
check-purpose=codesign

[slot.rootfs.0]
device=/dev/sda3
type=ext4
bootname=A

[slot.rootfs.1]
device=/dev/sda4
type=ext4
bootname=B

```

Therefore either the documentation is incomplete or the implementation is wrong.

Just wanted to let you know that I had huge problems with this. :)

Kind regards
Michael

jluebbe, 2023-10-24:
When you pass a keyring to `rauc bundle`, it will try to verify the generated signature against that keyring (which helps catching inconsistencies or expired certs). Currently, it's not possible to set the certificate purpose for that check without passing a config file. So for `bundle` either pass a config file or don't pass `--keyring`.

Your output also mentions `Failed to resign bundle`, is that part of your workflow? As resign currently requires `--keyring` even with `--no-verify`, you have to pass a config file to set the purpose **and** use the same purpose for both the old certificate (input signature check) and the new one (generated signature check).

Together this should be a workaround for your case. To fix this properly, we'd have to:

- [ ] split config file parsing from config consistency check (to require `[system]` only on the target)
- [ ] allow passing config file key/values from the command line (perhaps something like `-C keyring.check-purpose=codesign`)
- [ ] support different verification settings for the consistency check of the generated signature

jluebbe, 2023-10-24:
When #1268 merged, you should be able to use `rauc resign --no-verify` without `--keyring` to disable signature checks completely. You should probably automate those a different way, in that case. :)

jluebbe, 2023-10-24:
#1268 is merged now, so you should be able to use `rauc resign --no-verify`.

HiFiPhile, 2025-06-17:
I met this issue during an encrypted bundle creation with a simple PKI, 1st step `rauc bundle` is successful but `rauc encrypt` ends up with error `Verify error: unsuitable certificate purpose`.

Version: 1.14
OS: Yocto scarthgap

- Bare certificates (without `keyUsage`,`extendedKeyUsage`) generated by `openssl-ca-create.sh` works
- Our certificates with `keyUsage=critical,digitalSignature` and `extendedKeyUsage=critical,codeSigning` doesn't work

Some findings:

1. Add `-C keyring:check-purpose=codesign` doesn't work since the error happened in `CMS_Verify`
2. The error is reproducible with `openssl cms -sign -in test -out test.sig -signer release.crt -inkey release.key  -outform DER -nosmimecap -binary -certfile CA.cr` then `openssl cms -verify -in test.sig -content test -inform DER -binary -CAfile CA.crt` which returns
```
CMS Verification failure
808B6036CF700000:error:17000064:CMS routines:cms_signerinfo_verify_cert:certificate verify error:crypto/cms/cms_smime.c:291:Verify error: unsuitable certificate purpose
```
3. Looking into openssl code, `emailProtection` has to be included
https://github.com/openssl/openssl/blob/52dba1c098d9b0388afa4db2fbe3461df67461c6/crypto/cms/cms_smime.c#L284C38-L284C48

jluebbe, 2025-06-17:
Our current customer projects use a separate PKI for RAUC. This reduces complexity and avoids the need for certificate purposes. Accordingly, analyzing (and fixing) this issue has relatively low priority for us. If someone would provide funding, that would help. :)

jluebbe, 2025-06-18:
Do you have a suggestion on how to word it in a way that would have helped you?

HiFiPhile, 2025-06-18:
> Do you have a suggestion on how to word it in a way that would have helped you?

(sorry I'm not a native English speaker)

1. In `Certificate Key Usage Attributes` chapter add something like:
RAUC use OpenSSL CMS (Cryptographic Message Syntax, see RFC5652_) backend which requires `emailProtection` *extended key usage* attribute.

2. The given `v3_leaf_codesign` example is misleading which makes people think `codeSigning` is enough. It's better to rename it to `v3_leaf_bundlesign` and add `emailProtection`.

If somebody does `build/rauc bundle --cert=test/openssl-ca/dev/xku-codeSigning.cert.pem --key=test/openssl-ca/dev/private/xku-codeSigning.pem test/install-content/ out.raucb` it will end up with a unstable bundle.
Also applicable to `meta-rauc` since no keyring verification is done during bundle creation.