rauc/rauc #1132: Help misses information about the '--enable-streaming' option

martin-petzold, 2023-05-15:
**Describe the bug**

Help (`rauc --help`) misses information about the '--enable-streaming' option. Also from the documentation [1] it is not clear if this option should be used while creation of a bundle and/or when installing a bundle.

[1] https://rauc.readthedocs.io/en/latest/advanced.html#http-streaming

**Background information**

rauc v1.9-136-g49a9e26

**To Reproduce**

1. Invoke `rauc --help`

**Expected behavior**

Should contain the option '--enable-streaming' and description.

**Logs**

```
$ rauc --help
Usage:
  rauc [OPTION…] <COMMAND>

Options:
  -c, --conf=FILENAME     config file
  --keyring=PEMFILE       keyring file
  --mount=PATH            mount prefix
  -d, --debug             enable debug output
  --version               display version
  -h, --help              display help and exit

Command-specific help:
  rauc <COMMAND> --help

List of rauc bundle handling commands:
  bundle		Create a bundle
  resign		Resign an already signed bundle
  convert		Convert classic to casync bundle
  encrypt		Encrypt a crypt bundle
  replace-signature	Replaces the signature of an already signed bundle
  extract-signature	Extract the bundle signature
  extract		Extract the bundle content
  info			Show bundle information

List of rauc target commands:
  service		Start RAUC service
  install		Install a bundle
  status		Show status
  mount			Mount a bundle
  write-slot		Write image to slot and bypass all update logic

Environment variables:
  RAUC_KEY_PASSPHRASE Passphrase to use for accessing key files (signing only)
  RAUC_PKCS11_MODULE  Library filename for PKCS#11 module (signing only)
  RAUC_PKCS11_PIN     PIN to use for accessing PKCS#11 keys (signing only)
```

ejoerns, 2023-05-15:
With

> configure RAUC with --enable-streaming

a build-time configuration is meant. I guess this has been left unclear in the documentation?

Also, we should note the corresponding meson option here, too.

martin-petzold, 2023-05-15:
You mean it is this meson option: https://github.com/rauc/rauc/blob/master/meson_options.txt#L17

So it is only required when building RAUC? This option seems to have 'true' as default value.

It is not required for `rauc bundle` or `rauc install`?

martin-petzold, 2023-05-15:
Thanks, IMHO this line [1] should be removed because it is enabled by default.

[1] https://github.com/rauc/rauc/blob/master/docs/advanced.rst?plain=1#L418

ejoerns, 2023-05-16:
Maybe #1133 clarifies it. It might be disabled by package builder configurations, thus it might still be worth mentioning to look for having it enabled.

PR #1133 (2023-05-16): docs: clarify streaming build-time option
Meson was not mentioned and it might not have been clear that this is only a build-time option.

Fixes: #1132
commit df5b4191 (2023-05-16): docs: clarify streaming build-time option

Meson was not mentioned and it might not have been clear that this is
only a build-time option.

Fixes: #1132
closed 2023-05-16 by ejoerns