rauc/rauc #1136: RAUC fails running fw_printenv

martin-petzold, 2023-05-19:
**Describe the bug**

RAUC status fails to run fw_printenv, but this works on command-line.

**Background information**

Debian bullseye, custom 5.10 kernel, custom u-boot build, RAUC 1.9.

In this setup there was no RAUC script in u-boot in place.

**To Reproduce**

Steps to reproduce the behavior:

1. Install RAUC (and create 'system.conf' etc.)
2. Start RAUC service
3. Run RAUC status (`sudo rauc status`)

**Expected behavior**

No warning about using 'fw_printenv' or a more specific error about what actually went wrong with using 'fw_printenv'.

**Logs**

```
$ sudo rauc status

(rauc:553): rauc-WARNING **: 19:39:28.222: Failed getting primary slot: uboot backend: fw_printenv failed with exit code: 1

=== System Info ===
Compatible:  XYZ
Variant:     
Booted from: rootfs.0 (/dev/mmcblk2p1)

=== Bootloader ===
Activated: (null) ((null))

=== Slot States ===
○ [rootfs.1] (/dev/mmcblk2p2, ext4, inactive)
	bootname: rootfs1
	boot status: bad

○ [rootfs.0] (/dev/mmcblk2p1, ext4, booted)
	bootname: rootfs0
	mounted: /
	boot status: bad

$ sudo fw_printenv
Warning: Bad CRC, using default environment
bootcmd=run distro_bootcmd;run bsp_bootcmd
bootdelay=1
baudrate=115200
ethprime=eth0
loadaddr=0x40480000
arch=arm
cpu=armv8
[...]
```

**Additional context**

Maybe this only happens when the RAUC u-boot script is not included, however the warning just issues failed execution.

ejoerns, 2023-05-20:
Could you provide the output of the RAUC service?

```
journalctl -u rauc
```

 This should give more hints about why the `fw_printenv` command failed.

Also, using `sudo` for `rauc status` should not be required, did you make sure the service itself runs as root? 

> Maybe this only happens when the RAUC u-boot script is not included, however the warning just issues failed execution.

This is a u-boot script and will not be used from Linux userspace. Thus it should not have anything to do with the failure.

martin-petzold, 2023-05-21:
Okay, it is what I actually assumed (see message below). I have not added the bootloader script on this platform.

Of course RAUC is running as root user. I just don't use / have root user console for security reasons.

IMHO the error output of `rauc status` could be improved and give some hint. Nevertheless, I will close this issue.

```
Mai 21 13:10:17 tavla rauc[1757]: Warning: Bad CRC, using default environment
Mai 21 13:10:17 tavla rauc[1757]: ## Error: "BOOT_ORDER" not defined
Mai 21 13:10:17 tavla rauc[533]: Failed getting primary slot: uboot backend: fw_printenv failed with exit code: 1
Mai 21 13:10:17 tavla rauc[533]: Failed to get boot state of rootfs.1: uboot backend: fw_printenv failed with exit code: 1
Mai 21 13:10:17 tavla rauc[1759]: Warning: Bad CRC, using default environment
Mai 21 13:10:17 tavla rauc[1759]: ## Error: "BOOT_ORDER" not defined
Mai 21 13:10:17 tavla rauc[533]: Failed to get boot state of rootfs.0: uboot backend: fw_printenv failed with exit code: 1
Mai 21 13:10:17 tavla rauc[1760]: Warning: Bad CRC, using default environment
Mai 21 13:10:17 tavla rauc[1760]: ## Error: "BOOT_ORDER" not defined
Mai 21 13:10:17 tavla rauc[533]: Failed to determine boot states: Could not determine all boot states
```

ejoerns, 2023-05-21:
Ok, so the issue was that the variable `BOOT_ORDER` was not defined in the environment.
You are right, indirectly this is caused by the missing script that would have created the variables from within U-Boot upon the first boot.

However, as you see above, this is also not fatal to RAUC since it does not necessarily expect the variable to be set already. But for a consistent startup, it should be pre-defined during bring-up, yes.

The user output currently relies on the exit code of `fw_printenv` which does not give any hint why the tool failed. Otherwise, we would need to parse the error output of `fw_printenv` and guess which line holds the most valuable information. A general consideration we had was extending those journal entries with additional debugging hints, provided by systems catalogs.

martin-petzold, 2023-05-21:
Still, why does `fw_printenv` actually fail here with exit code `1`? It should work. If I call `sudo fw_printenv`, my exit code is `0`:

```
tavla@tavla:~$ echo $?
0
```
Also, how can RAUC know about `## Error: "BOOT_ORDER" not defined`, if `fw_printenv` returned with exit code `1`?

ejoerns, 2023-05-21:
The actual call that fails should be ``fw_printenv BOOT_ORDER``. Since this variable is not defined but attempted to be read, an error is returned.

The output
```
## Error: "BOOT_ORDER" not defined
```
comes from `fw_printenv` (not RAUC) for the same reason.

martin-petzold, 2023-05-21:
Okay, I understand. It is possible to read a specific property.

closed 2023-05-21 by martin-petzold